GreenWaves-Technologies/gap_sdk #4: Installation on Ubuntu 18.04

cjameshuff, 2018-07-25:
I'm aware it's not officially supported, but here's what I've come across under Ubuntu 18.04. The instructions largely worked, apart from some Git LFS errors:

    Error: Failed to call git rev-parse --git-dir --show-toplevel: "fatal: not a git repository (or any of the parent directories): .git\n"

These didn't prevent the toolchain from downloading. However, the toolchain does not work under Ubuntu 18.04, building the SDK failed with:

  /usr/lib/gap_riscv_toolchain/bin/../libexec/gcc/riscv32-unknown-elf/7.1.1/cc1: error while loading shared libraries: libmpfr.so.4: cannot open shared object file: No such file or directory

It appears libmpfr4 has been removed from the packages, only libmpfr6 is available.

After rebuilding/installing pulp-riscv-gnu-toolchain with:

    ./configure --prefix=/usr/lib/gap_riscv_toolchain --with-arch=rv32imc --with-cmodel=medlow --enable-multilib
    make

I was able to build the basic helloworld project, and several other projects, though I haven't tried actually running any of them yet.

Yaooooo, 2018-07-25:
Hi, Thank you very much for your feedback. 

As we are a small team, to avoid some support difficulties, we don't support other OS and other version of Ubuntu for now. We are working on a Cross-platform IDE, and this will resolve a lots of these problems.

In fact, if you want to use our SDK in Ubuntu 18.04, You can try this instead of rebuilt the toolchain:
Because the libmpfr4 was replaced by libmpfr6 in Ubuntu 18.04, which is compatible with libmpfr4, so you can create a link symbolical with a name of libmpfr.so.4, and link to the 6 one. 

Thank you again for your sharing, and if possible, please let us know your further feedbacks

Best
Yao

ivankravets, 2018-08-03:
Temporary (dirty) solution:
```shell
sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.6 /usr/lib/x86_64-linux-gnu/libmpfr.so.4
```

closed 2018-11-27 by Yaooooo