GreenWaves-Technologies/gap_sdk #14: cmake not found

ccravens, 2018-08-26:
When attempting to build the SDK for the first time on the VirtualBox Ubuntu image, I get the following error:

`
gap_sdk$ make all
mkdir -p -p /home/ccravens/dev/ml/gap8/gap_sdk/install/gap
mkdir -p /home/ccravens/dev/ml/gap8/gap_sdk/install/workstation/bin
cp -rf /home/ccravens/dev/ml/gap8/gap_sdk/tools/runner /home/ccravens/dev/ml/gap8/gap_sdk/install/workstation
cp -rf /home/ccravens/dev/ml/gap8/gap_sdk/tools/bin/* /home/ccravens/dev/ml/gap8/gap_sdk/install/workstation/bin
cp -rf /home/ccravens/dev/ml/gap8/gap_sdk/tools/ld /home/ccravens/dev/ml/gap8/gap_sdk/install/workstation
cp -rf /home/ccravens/dev/ml/gap8/gap_sdk/tools/rules /home/ccravens/dev/ml/gap8/gap_sdk/install/workstation
make -C /home/ccravens/dev/ml/gap8/gap_sdk/tools/pulp_tools all
make[1]: Entering directory '/home/ccravens/dev/ml/gap8/gap_sdk/tools/pulp_tools'
#### Building in /home/ccravens/dev/ml/gap8/gap_sdk/build
#### Release type is Debug
#### Installing to /home/ccravens/dev/ml/gap8/gap_sdk/install/workstation
mkdir -p /home/ccravens/dev/ml/gap8/gap_sdk/build
( cd /home/ccravens/dev/ml/gap8/gap_sdk/build ; \
  cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/home/ccravens/dev/ml/gap8/gap_sdk/install/workstation \
	 /home/ccravens/dev/ml/gap8/gap_sdk/tools/pulp_tools/ )
/bin/sh: 2: cmake: not found
Makefile:50: recipe for target '/home/ccravens/dev/ml/gap8/gap_sdk/build/CMakeCache.txt' failed
make[1]: *** [/home/ccravens/dev/ml/gap8/gap_sdk/build/CMakeCache.txt] Error 127
make[1]: Leaving directory '/home/ccravens/dev/ml/gap8/gap_sdk/tools/pulp_tools'
Makefile:67: recipe for target 'install_pulp_tools' failed
make: *** [install_pulp_tools] Error 2
`

Suggest adding `cmake` to the list of apt-get packages to install on the first step of the README

sousoux, 2018-08-26:
Hi, apologies for this. We started checking in portions of work that have been done to improve the debug bridge and that does have an effect on the needed packages. I will do this now.

closed 2018-08-26 by sousoux

ccravens, 2018-08-26:
No problem! Thanks! I'm excited to start working with your product