GreenWaves-Technologies/gap_sdk #18: gap_flasher broken

ccravens, 2018-09-01:
When trying to compile and run the pulp helloworld example program, I get the following error:

```
ccravens@osboxes:~/src/helloworld$ make all
mkdir -p /home/ccravens/src/helloworld/BUILD/GAP8/GCC_RISCV
riscv32-unknown-elf-gcc -O3 -g -fno-jump-tables -fno-tree-loop-distribute-patterns -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -march=rv32imcxgap8 -mPE=8 -mFC=1 -D__riscv__ -D__pulp__ -MMD -MP -c test.c  -I/home/ccravens/gap_sdk/install/gap/include/pulp-os  -I/home/ccravens/gap_sdk/install/gap/include/pulp-os/io  -I/home/ccravens/gap_sdk/install/workstation/include  -include /home/ccravens/gap_sdk/install/gap/include/pulp-os/gap_config.h -MD -MF /home/ccravens/src/helloworld/BUILD/GAP8/GCC_RISCV/test.d -o /home/ccravens/src/helloworld/BUILD/GAP8/GCC_RISCV/test.o
riscv32-unknown-elf-gcc -MMD -MP -Wl,--gc-sections -O3 -g -o /home/ccravens/src/helloworld/BUILD/GAP8/GCC_RISCV/test  /home/ccravens/src/helloworld/BUILD/GAP8/GCC_RISCV/test.o -L/home/ccravens/gap_sdk/install/gap/libs -lrt -lio -lrt -lgcc -T/home/ccravens/gap_sdk/install/workstation/ld/link.gap8.ld -T/home/ccravens/gap_sdk/install/workstation/ld/gapuino.conf.ld -nostartfiles -nostdlib -include /home/ccravens/gap_sdk/install/gap/include/pulp-os/gap_config.h
/home/ccravens/gap_sdk/tools/version/record_version.sh: line 7: cd: /home/ccravens/gap_sdk/tools/gap_flasher: No such file or directory
```

I'm wondering if the following commit may have broken this: https://github.com/GreenWaves-Technologies/gap_sdk/commit/d5d010a4e0be6f0539d87c4daf50af3d69d63e94

Yaooooo, 2018-09-01:
> I'm wondering if the following commit may have broken this: d5d010a

Hi, because the gap_flasher has been moved to pulp_tools as a submodule. It
seems the submodules haven’t been updated entirely. Please try this command
in gap_sdk:
git submodule update —init —recursive
Then rebuild the sdk:
make clean all

If this doesn’t work, please tell us with a file version.log in your test
folder.

Best
Yao

ccravens, 2018-09-01:
Still had the issue even after updating and re-building SDK. I submitted the following PR to address: https://github.com/GreenWaves-Technologies/gap_sdk/pull/19

However, I noticed that a `git status` seems to take a long time (10-30s) which is performed by https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/version/record_version.sh#L19

This stalls the build by 10-30s, so I had to comment out the `version` section of the Makefile: https://github.com/GreenWaves-Technologies/gap_sdk/blob/d5d010a4e0be6f0539d87c4daf50af3d69d63e94/tools/rules/pulp_rules.mk#L113

Yaooooo, 2018-09-01:
You are right. I have merged your request, thank you very much.
This script is using for record the commits of each submodule, which could help us to figure out the problem of our sdk user. 
I have just removed it from the each example/application build (xxx_rules.mk), and put it into the sdk build process. 
But why the "git status" takes so long, as I don't have this effect, I need some more time to reproduce your problem.

ccravens, 2018-09-01:
Hello Yao:

I believe my git status issue might be due to the fact that I am running the SDK inside of a VirtualBox Shared Folder. I will close this issue for now and continue investigating on my end.

Thank you for the quick PR approval / merge!

closed 2018-09-01 by ccravens