GreenWaves-Technologies/gap_sdk #34: Problems with running Cifar10 example application

adrielkuek, 2018-11-27:
Hi,

We are currently to run the Cifar10 example application shipped in the code, however there seem to be a truncated file that is not presented in the SDK. The /gap_sdk/tools/autotiler/lib/libtile.a file is not present. Is this the most updated version of the code? Should we be expecting any updates to this?

Thanks for the clarification.

Adriel

Yaooooo, 2018-11-27:
Hi, 

Thank you for asking.
As our documents indicated (https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html#section8), the library of autotiler need to be installed alternately.
If you are using our gap_sdk, you could just execute the command below:
make autotiler

This command will trigger the process of installation.

Best,
Yao

adrielkuek, 2018-11-27:
Hi Yao,

Thanks for the reply. I’m having problems installing autotiler. Running make autotiler runs into error. When I navigate the URL sent to my mail https://greenwaves-technologies.com/autotiler/ I’m led to a 404 page not found error. Is this normal?

Regards,

Adriel

adrielkuek, 2018-11-27:
> Hi Adriel,
>
> When you launch this command, and you will see:
> python3 get_tiler.py
> Registration is required to load the GAP8 AutoTiler library
>
> You will be prompted for your name, company and email address and the
> link for the AutoTiler libray will be sent to your email address.
> This information is used uniquely to keep track of AutoTiler users.
> Enter your first name: test
> Enter your last name: test
> Enter your company name: test
> Enter your country: test
> I don't know that country. Please enter a valid country name or ISO code.
> Enter your country: cn
> Country CN
> Enter your email address: support@greenwaves-technologies.com
> Triggering email ... please wait
> Please check your email and copy and paste the URL in the email below
> Enter URL from email:
>
> For the url we sent you, you should put it into the last question.
> Then the install processing will be executed.
>
> This install process is made for customer support needed, thank you for understanding.
> Now as you have already answer all the question, I won't let you to redo this. I will send you the liberary by mail.
>
> BTW, I have modified your comment above to mask the url :-)
>
> Best,
> YAO

Thanks Yao,

I’ll try and install the autotiler once again to generate the library files. 

Regards,

Adriel

adrielkuek, 2018-11-28:
Hi Yao,

I have installed the autotiler library as per your instructions and have
run the command -> make clean all run platform=gvsoc. But I'm still getting
the following error:

make clean all run platform=gvsoc
rm -f GenCifar10 Cifar10Kernels.* Cifar10KernelsInit.*
gcc -o GenCifar10 -DRT_HAS_HWCE=1
-I/home/adriel/gap_sdk/tools/autotiler/include
-I/home/adriel/gap_sdk/tools/autotiler/generators/CNN/generator/include
Cifar10Model.c
/home/adriel/gap_sdk/tools/autotiler/generators/CNN/generator/src/CNN_Generators.c
/home/adriel/gap_sdk/tools/autotiler/lib/libtile.a
/home/adriel/gap_sdk/tools/autotiler/lib/libtile.a: file not recognized:
File truncated
collect2: error: ld returned 1 exit status
Makefile:37: recipe for target 'GenTile' failed
make: *** [GenTile] Error 1

It still states that libtile.a is not recognised. I have checked inside
/gap_sdk/tools/autotiler/lib/ and confirmed that libtile.a is indeed
present. So I'm not too sure what went wrong that is hindering me from
running the sample application properly.

Thanks,

Adriel

adrielkuek, 2018-11-28:
> Hi,
>
> I think the lib hasn’t been downloaded successfully and completely.
> I have send you a mail yesterday, with the lib in it. And more details. You haven’t received it?
>
> Moreover, why you add « platform =gvsoc »? We haven’t released this platform .
>
> Best
> Yao

Hi Yao,

Yes I have received the lib file yesterday and have followed your steps to extract the libtile.a file.

I used the “make clean all run platform=gvsoc” command based on your documentation steps listed here “https://greeneaves-technologies.com/manuals/BUILD/APPLICATIONS/html/index.html#section14”

So to run the example application just need to run make clean all run?

Adriel

Yaooooo, 2018-11-28:
Sorry. This is a doc problem. Yes we use only make clean all run for the GAPuino.
For your problem of autotiler, have you tried both 2 methods in my mail?
Thanks

adrielkuek, 2018-11-28:
Hi Yao,

Yes I have tried both methods to unwrap without errors. So it’s quite puzzling as to why am I still getting the errors. 

Adriel

Yaooooo, 2018-11-28:
Hi Adrien,
Sorry I am in vacation without my PC. So I cannot retest this from my side. I recommend you to delete lib folder and the url file to retry the process.
And I will ask my college to follow your problem.
Best

adrielkuek, 2018-11-29:
> One thing worth trying:
>
> Normally lib/libtile.a is a symbolic link to the actual tiler lib with the
> version number. Something like lib/libtile.2.0.0.a.
>
> Do an ls -l in the lib directory and make sure that the link is pointing
> towards the correct file (the one that Yao sent you).

Hi Yao/Martin,

I have deleted the lib folder and url, and tried to re-install the library
by running the make autotiler command as per your earlier instructions. I
then run into error trying to resolve the url site:

adriel@adriel-VirtualBox:~/gap_sdk$ make autotiler
make -C /home/adriel/gap_sdk/tools/autotiler all
make[1]: Entering directory '/home/adriel/gap_sdk/tools/autotiler'
rm -rf lib/libtile.a
echo libtile.2.0.0.a | wget --no-use-server-timestamps --base=`cat
.tiler_url` --input-file=- -O lib/libtile.2.0.0.a
--2018-11-28 18:30:07--
ftp://https/://green-waves-technologies.com/autotiler/libtile.2.0.0.a
           => ‘lib/libtile.2.0.0.a’
Resolving https (https)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘https’
Makefile:18: recipe for target 'lib/libtile.2.0.0.a' failed
make[1]: *** [lib/libtile.2.0.0.a] Error 4
make[1]: Leaving directory '/home/adriel/gap_sdk/tools/autotiler'
Makefile:78: recipe for target 'autotiler' failed
make: *** [autotiler] Error 2

When I try to access the url "https://greenwaves-technologies.com/autotiler/",
I reached an 404 error accessing page. Is the link for the url correct?

Regards,

Adriel

adrielkuek, 2018-11-29:
Hi Martin,

I've checked the link in the lib folder and verified that it is indeed
pointing correctly.

adriel@adriel-VirtualBox:~/gap_sdk/tools/autotiler/lib$ ls -l
total 504
-rw-rw-r-- 1 adriel adriel 515188 Nov 28 18:36 libtile.2.0.0.a
lrwxrwxrwx 1 adriel adriel     19 Nov 28 18:39 libtile.a ->
lib/libtile.2.0.0.a

Regards,

Adriel

bacci85, 2018-11-29:
Hi Adriel,
 your are typing the wrong address. This should be the right one:
 https://greenwaves-technologies.com/autotiler/libtile.2.0.0.a

Without ftp:// and with the right positioning of https://

This url should stay inside the .tiler_url file

adrielkuek, 2018-11-29:
Hi all,

To date I have tried and re-tried multiple ways to fix the autotiler
library.

Following are the different methods used:

- Downloaded the libtile.2.0.0.a provided by Yao and placed into
gap_sdk/tools/autotiler/lib. Executed "ln -s lib/libtile.2.0.0.a
lib/libtile.a" under ../autotiler directory. Final libtile.a link showed
broken.

- Downloaded .tiler_url file provided by Yao and placed into
gap_sdk/tools/autotiler and run "make autotiler" command. Resulted in
webpage access error:
"rm -rf lib/libtile.a
echo libtile.2.0.0.a | wget --no-use-server-timestamps --base=`cat
.tiler_url` --input-file=- -O lib/libtile.2.0.0.a
--2018-11-28 18:30:07--
ftp://https/://green-waves-technologies.com/autotiler/libtile.2.0.0.a
           => ‘lib/libtile.2.0.0.a’
Resolving https (https)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘https’
Makefile:18: recipe for target 'lib/libtile.2.0.0.a' failed
make[1]: *** [lib/libtile.2.0.0.a] Error 4"

- Directly accessed webpage to download autotiler library as given by
Francesco in the eariler post. Final extracted libtile.a file still showed
link broken error.

Just to confirm, my libtile.2.0.0.a file size is 515.2 kB and is consistent
no matter where or how I pull it.

Is there any other way/method to properly extract the libtile.a library
file correctly?

Regards,

Adriel

sousoux, 2018-12-19:
Resolved offline

closed 2018-12-19 by sousoux

Narduzzi, 2019-02-19:
Hello,

After following the setup tutorial (https://greenwaves-technologies.com/setting-up-sdk/) and installing autotiler, I've tried to run the Cifar10 example, but I'm encountering the following error :

**command** : 
```bash
$ cd ~/gap_sdk/examples/pulp-examples/autotiler_examples/Cifar10 
$ make all
```
**error** : 

```bash
gcc -o GenCifar10 -DRT_HAS_HWCE=1 -I/home/sni/gap_sdk/tools/autotiler/include -I/home/sni/gap_sdk/tools/autotiler/generators/CNN/generator/include Cifar10Model.c /home/sni/gap_sdk/tools/autotiler/generators/CNN/generator/src/CNN_Generators.c /home/sni/gap_sdk/tools/autotiler/lib/libtile.a

/usr/bin/ld: /home/sni/gap_sdk/tools/autotiler/lib/libtile.a(GenTiling.c.o): relocation R_X86_64_32S against `.data' can not be used when making a PIE object; recompile with -fPIC

/usr/bin/ld: /home/sni/gap_sdk/tools/autotiler/lib/libtile.a(GenTilingDump.c.o): relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC

/usr/bin/ld: /home/sni/gap_sdk/tools/autotiler/lib/libtile.a(ModelCreate.c.o): relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC

/usr/bin/ld: /home/sni/gap_sdk/tools/autotiler/lib/libtile.a(TilingGenCode.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIC

/usr/bin/ld: /home/sni/gap_sdk/tools/autotiler/lib/libtile.a(TilingGenCodeHelper.c.o): relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC

/usr/bin/ld: /home/sni/gap_sdk/tools/autotiler/lib/libtile.a(ComputeTiling.c.o): relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC

/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
Makefile:37: recipe for target 'GenTile' failed
make: *** [GenTile] Error 1`

```

How can I solve this issue?
Thanks for the help!

Yaooooo, 2019-02-19:
Hi Narduzzi,

I am sorry for this issue. In fact, this issue comes from the incompatibility between the different version of gcc.
The liberary libtile.a is compiled with:
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11)
This is not upward compatible with the new one in Ubuntu 18.04, which you are using I suppose.

We will update the liberary recently and compile it with the new version of gcc. This will normally resolve your problem.

Best,
Yao

Yaooooo, 2019-02-19:
Hi,

I am sorry that you have met this problem.
This problem is because of the incompatibility between 2 different versions of gcc.
The liberary "libtile.a" has been compiled with: "gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11)"
This is not upward compatible with the one in Ubuntu 18.04.
We will recently update this liberary and compile it with the new version of gcc, which I think will resolve your problem.

Best,
Yao

Narduzzi, 2019-02-19:
Hello Yao,

Thanks for the information. I will wait for the update then!

Best,
Simon

Yaooooo, 2019-02-20:
Hi Simon,

We cannot keep you in waiting like this :)
Please find in the attachment the autotiler I have built in Ubuntu 18.04.
You can extract it and put it into gap_sdk/tools
We will keep you inform when we update the github officially.

BTW, I have just approve your request for gap8_developper group. Because
you haven't write your order number in the request, we don't know whether
you have a gapuino or not.
You could continue asking questions or discuss with us in this group.

Best regards,
Yao