GreenWaves-Technologies/gap_sdk #52: make run ignores flashImg.raw file build issues

asmorkalov, 2019-05-22:
To reproduce add file that does not exist to `PLPBRIDGE_FLAGS` in Makefile:
```
PLPBRIDGE_FLAGS += $(addprefix -f ,./quantized_model/*.bin) -jtag
```
`make run` command log:
```
Reading ELF file
  Segment : 1b000000 308
  Segment : 1b000308 50
  Segment : 1c000000 65af0
  Segment : 1c065bc8 44
  Segment : 1c065c0c 8
Traceback (most recent call last):
  File "/home/alexander/Projects/GAP8/gap_sdk/install/workstation/bin/flashImageBuilder", line 432, in <module>
    flashImage.appendComponent(os.path.dirname(comp), os.path.basename(comp))
  File "/home/alexander/Projects/GAP8/gap_sdk/install/workstation/bin/flashImageBuilder", line 133, in appendComponent
    self.compList.append(Comp(dirname, name))
  File "/home/alexander/Projects/GAP8/gap_sdk/install/workstation/bin/flashImageBuilder", line 69, in __init__
    self.size = os.path.getsize(self.path)
  File "/usr/lib/python3.5/genericpath.py", line 50, in getsize
    return os.stat(filename).st_size
FileNotFoundError: [Errno 2] No such file or directory: '/home/alexander/Projects/GAP8/gap8-reid-demo/tests/single_layer_test/../../ReID-Demo/quantized_model/*.bin'
Bridge init - log level 0
Cable state change connected
FATAL ERROR: [Errno 2] No such file or directory: 'flashImg.raw'
commands completed
Bridge init - log level 0
Cable state change connected
main call
pad config done
fs mounted
Loading layers to HyperRAM
```

The error is ignored and make is not stopped, as expected.

Yaooooo, 2019-05-22:
Hi,
The problem is the flash image "flashImg.raw" has not been created successfully.
Because of this error, showed in your log above:
"FileNotFoundError: [Errno 2] No such file or directory: '/home/alexander/Projects/GAP8/gap8-reid-demo/tests/single_layer_test/../../ReID-Demo/quantized_model/*.bin'"

Therefore, could you make sure your path is correct? 
$(addprefix -f ,./quantized_model/*.bin)

I have done a simple test by using: examples/pulp-examples/periph/file_system
I have created a folder "test", and add some files into it.  So my folder is like this:
.
├── hello.txt
├── Makefile
├── test
│   ├── hello.txt
│   ├── Makefile
│   └── test.c
└── test.c

Then I use your flag: PLPBRIDGE_FLAGS += $(addprefix -f , ./test/*) -jtag

When I do "make clean all run", we can see the last command has been executed is: (Just before the GAP8 logo)
/home/yao/Projects/SDK/gap_sdk_github/install/workstation/runner/run_gapuino.sh -f ./test/* -jtag

And in the log, we can see:
Reading ELF file
  Segment : 1b000004 324
  Segment : 1b000328 0
  Segment : 1c000000 5ff4
  Segment : 1c0068d8 44
Generating boot binary
  Nb areas: 4
  Area 0: offset: 0x400, base: 0x1b000004, size: 0x324, nbBlocks: 1
  Area 1: offset: 0x800, base: 0x1b000328, size: 0x0, nbBlocks: 0
  Area 2: offset: 0x800, base: 0x1c000000, size: 0x5ff4, nbBlocks: 24
  Area 3: offset: 0x6800, base: 0x1c0068d8, size: 0x44, nbBlocks: 1
Generating files (header offset: 0x6c00)
  Adding component (name: Makefile, flashOffset: 0x6c4c)
  Adding component (name: hello.txt, flashOffset: 0x6e04)
  Adding component (name: test.c, flashOffset: 0x6e28)

All the 3 files have been added successfully.

Therefore, I think the problem is the path of the file.
Please let me know if you got any updates.

Best,
Yao

closed 2019-06-13 by Yaooooo

asmorkalov, 2019-06-14:
@Yaooooo The path to files for flashing is not correct and it a point of the ticket. Build system ignores the error and starts application on board. I suggest to handle the script error and stop execution.