GreenWaves-Technologies/gap_sdk #54: Mnist Example Unable to connect to localhost port 9999 (connection refused)

peternomikos, 2019-06-04:
I try to run the example of Mnist from pulp-examples and i get this response :

Bridge init - log level 1
Program exited
Loading binary through jtag
Notifying to boot code that we are doing a JTAG boot
Cable state change connected
Loading binary from test
Loading section (base: 0x1b000004, size: 0x324)
Loading section (base: 0x1b000328, size: 0x0)
Init section to 0 (base: 0x1b000328, size: 0x808)
Loading section (base: 0x1c000000, size: 0x270f8)
Unable to connect to localhost port 9999 (Connection refused)
Init section to 0 (base: 0x1c0270f8, size: 0xcc)
Loading section (base: 0x1c0271c4, size: 0x44)
Loading section (base: 0x1c027208, size: 0x84)
debug address 0x1c000090 contents 0x1c008808
debug address 0x1c000090 contents 0x1c008808
Start execution on GAP8
Entering main controller
Error from ftdi_read_data() - usb bulk read failed (-7)
Cable state change disconnected
ft2232: Error from ftdi_read_data() - usb bulk read failed (-7)
Read failed
ft2232: fdti_stream_in has failed
ft2232: ftdi_stream_inout has failed
adv_dbg_itf: failed to read crc from device
*** Error in `python3': double free or corruption (out): 0x0000000003111cb0 ***

I am in a virtual box of ubuntu 16.04 Xenial and I can run other examples as cifar10.

Thank you for your time.

Yaooooo, 2019-06-04:
Hi,

This is a known issue of our new bridge. This application will upload a image from local to GAP8 through another tool. There are some issue of the synchronization between these 2 tools.
The FileIO tool need to be executed after the application is executed on GAP8.
Therefore, we have a flag in Makefile:
PLPBRIDGE_FLAGS += -fileIO 5

Here the "5" means the 2nd tool will wait for 5 second to start. But different PC, environment will change the latency. So you may increase this number. You can find the details here: https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html#section15

Otherwise, if you are using a VM, please change to USB 3.0 in VM settings. This will improve the performance of connection.

If this helps you, please close the issue.

Yao

peternomikos, 2019-06-04:
Thank you for your reply! I will begin testing.

peternomikos, 2019-06-04:
I increased the value and the localhost problem was fix but I get this message now:

Bridge init - log level 1
Program exited
Loading binary through jtag
Notifying to boot code that we are doing a JTAG boot
Cable state change connected
Loading binary from test
Loading section (base: 0x1b000004, size: 0x324)
Loading section (base: 0x1b000328, size: 0x0)
Init section to 0 (base: 0x1b000328, size: 0x808)
Loading section (base: 0x1c000000, size: 0x270f8)
Init section to 0 (base: 0x1c0270f8, size: 0xcc)
Loading section (base: 0x1c0271c4, size: 0x44)
Loading section (base: 0x1c027208, size: 0x84)
debug address 0x1c000090 contents 0x1c008808
debug address 0x1c000090 contents 0x1c008808
Start execution on GAP8
Entering main controller
ft2232: ftdi_write_data() failed - usb bulk write failed (-1)
Cable state change disconnected
ft2232: ftdi_stream_out has failed
ft2232: ftdi_stream_inout has failed
Loop exited
commands completed

Thank you for your time

Peter

Yaooooo, 2019-06-04:
Hi

It seems you have not only this problem.
Can you give a try on helloworld:  gap_sdk/examples/pulp-examples/helloworld
It seems you have some problem with lib usb.

Yao

peternomikos, 2019-06-04:
I tried the helloworld and it run correctly with no error messages.

peternomikos, 2019-06-04:
Hi 
I kept running the test of Mnist and one of them succeeded. But its not something stable as the next try failed again with Error from ftdi_read_data() - usb bulk read failed(-7).

Yaooooo, 2019-06-05:
Have you tried to set your VM to use USB 3.0 ?
And other examples?

peternomikos, 2019-06-05:
I set my VM to USB 3.0 and it fixes the problem. 
Thank you a lot for your time.

closed 2019-06-05 by peternomikos