GreenWaves-Technologies/gap_sdk #56: UART Connection

peternomikos, 2019-06-10:
Hi 

I want to transfer data from my Gap8 to the computer through a UART module but I have some issues so far. I am not sure if the pins I choose are correct because in the users manual are not connected anywhere . I connected them according to the Arduino Uno. Also I am not sure with what to fill the entry itf entry of the config structure and the device name for the uart_open function.

Thank you for your time

Panagiotis

Yaooooo, 2019-06-13:
Hi, 

Actually, on the GAPuino, the uart of GAP8 is connected to the FTDI chip. And the pinout has not been connected by default.
Therefore, for using the UART pin on GAPuino, some hardware modifications are necessary:
1, You need to disconnect the R9 and R36, and connect R37 and R39:  

  | UART | I2S
-- | -- | --
R9 | Unconnected | 0 ohm
R36 | Unconnected | 0 ohm
R37 | 0 ohm | Unconnected
R38 | Unconnected | Unconnected
R39 | 0 ohm | Unconnected

[image]

[image]

[image]

We are sincerely sorry for that inconvenience.

Yao

closed 2019-11-13 by sousoux

jinhuang01, 2019-12-19:
Hi, I am a little confused about the second and third images on this issue - is that we should also disconnect R41 and R40 to use UART module?

jinhuang01, 2019-12-19:
Also, if I want use GAP8 with Bluetooth Low Energy, should I also do the above hardware modifications, too? Thank you very much.

Pafrak, 2019-12-30:
Hi jinhuang01,
 on which board you want to use the BLE? do you want to connect a shield to Gapuino?