GreenWaves-Technologies/gap_sdk #70: Problems with accesing the performance counterS

eliasph88, 2019-08-17:
Hello,
I modified a bit the application you give in the following link:
https://github.com/GreenWaves-Technologies/benchmarks/tree/f12e26077a4f42409dcb597f36612cf7fe96dbdb/gap8/cnn
With the code you provide in the CIfar example to be able to read the performance counters. 

With the PCER configured like this:
    rt_perf_t *perf = cluster_perf;
    rt_perf_init(perf);
    rt_perf_conf(perf, (1<<RT_PERF_CYCLES));
                       ........
            rt_perf_reset(perf);
            gap8_resethwtimer();
            WriteGpio(GPIO, 1);
            rt_perf_start(perf);
            Ti = gap8_readhwtimer();
            for (int i=0; i<Iter; i++)      Counting();
            Ti = gap8_readhwtimer() - Ti;
            rt_perf_stop(perf);
            rt_perf_save(perf);
            WriteGpio(GPIO, 0);
            PERF_CYCLES[0] = rt_perf_get(perf, RT_PERF_CYCLES);
            *num_ops = Ti;
......
printf ("%10s Benchmark: (x%d iterations)  Time: %10lu uSec. Cycles1: %10lu / RT_PERF_CYCLES: %10lu\n",tests_names[i], ITERATIONS, tot_time, op_num, PERF_CYCLES[cur_test-1]);
......
and with the PCER configured like this:
    rt_perf_t *perf = cluster_perf;
    rt_perf_init(perf);
    rt_perf_conf(perf, (1<<RT_PERF_LD));
.......
    rt_perf_reset(perf);
     gap8_resethwtimer();
    WriteGpio(GPIO, 1);
    rt_perf_start(perf);
    Ti = gap8_readhwtimer();
    for (int i=0; i<Iter; i++)      Counting();
    Ti = gap8_readhwtimer() - Ti;
    rt_perf_stop(perf);
    rt_perf_save(perf);
    WriteGpio(GPIO, 0);
    PERF_CYCLES[0] = rt_perf_get(perf, RT_PERF_CYCLES);
    PERF_LD[0]  = rt_perf_get(perf, RT_PERF_LD);
    *num_ops = Ti;

printf ("%10s Benchmark: (x%d iterations)  Time: %10lu uSec. Cycles1: %10lu / PERF_LD: %10lu\n",tests_names[i], ITERATIONS, tot_time, op_num, PERF_LD[cur_test-1]);

I ALWAYS obtain the same data:
                      ---------------        Sequential   ---------------
Counting FC Benchmark: (x10000 iterations)  Time:    1528015 uSec. Cycles1:  263861947 / Cycles2:  469782528 Instructions:  263863029

                      ---------------          Parallel   ---------------
Counting FC Benchmark: (x10000 iterations)  Time:    1540527 uSec. Cycles1:  266204986 / PCER:  469782528
Counting 1C Benchmark: (x10000 iterations)  Time:    1548035 uSec. Cycles1:  267713181 / PCER:  469782528
Counting 2C Benchmark: (x10000 iterations)  Time:    1577729 uSec. Cycles1:  273205794 / PCER:  469782528
Counting 4C Benchmark: (x10000 iterations)  Time:    1669403 uSec. Cycles1:  289773049 / PCER:  469782528

Am I doing something wrong or there is something wrong with the function?
I need help with this ASAP
Kind regards

Pafrak, 2019-11-12:
Hi Eliasph88,
 sorry for the late answer, I guess that you are doing something wrong. Can you provide what you are doing in the Counting() function?

Pafrak, 2019-11-15:
Closed, due to inactivity

closed 2019-11-15 by Pafrak