GreenWaves-Technologies/gap_sdk #73: ftdi build problem

Yaooooo, 2019-10-21:
When building the sdk, got this error:

`[ 33%] Performing download step (git clone) for 'ftdi'
cd /home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/src && /usr/bin/cmake -P /home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/src/ftdi-stamp/ftdi-download-Debug.cmake
CMake Error at ftdi-stamp/ftdi-download-Debug.cmake:16 (message):
  Command failed: 1

   '/usr/bin/cmake' '-P' '/home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/tmp/ftdi-gitclone.cmake'

  See also

    /home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/src/ftdi-stamp/ftdi-download-*.log

CMakeFiles/ftdi.dir/build.make:89: recipe for target 'ftdi/src/ftdi-stamp/ftdi-download' failed
make[7]: *** [ftdi/src/ftdi-stamp/ftdi-download] Error 1
make[7]: Leaving directory '/home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build'
CMakeFiles/Makefile2:144: recipe for target 'CMakeFiles/ftdi.dir/all' failed
make[6]: *** [CMakeFiles/ftdi.dir/all] Error 2
make[6]: Leaving directory '/home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build'
Makefile:138: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build'
CMakeFiles/pulp-debug-bridge.dir/build.make:110: recipe for target 'pulp-debug-bridge-prefix/src/pulp-debug-bridge-stamp/pulp-debug-bridge-build' failed
make[4]: *** [pulp-debug-bridge-prefix/src/pulp-debug-bridge-stamp/pulp-debug-bridge-build] Error 2
make[4]: Leaving directory '/home/yao/Projects/SDK/gap_sdk_github/build'
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/pulp-debug-bridge.dir/all' failed
make[3]: *** [CMakeFiles/pulp-debug-bridge.dir/all] Error 2
make[3]: Leaving directory '/home/yao/Projects/SDK/gap_sdk_github/build'
Makefile:127: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/yao/Projects/SDK/gap_sdk_github/build'
Makefile:38: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/yao/Projects/SDK/gap_sdk_github/tools/pulp_tools'
Makefile:85: recipe for target 'install_pulp_tools' failed
make: *** [install_pulp_tools] Error 2`

Checked error logs: /home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/src/ftdi-stamp/ftdi-download-*.log

`yao@ubuntu:~/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/src$ cat /home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/src/ftdi-stamp/ftdi-download-*.log
Cloning into 'ftdi'...
error: cannot expand target domain name in SRV RR
fatal: Unable to look up developer.intra2net.com
Cloning into 'ftdi'...
error: cannot expand target domain name in SRV RR
fatal: Unable to look up developer.intra2net.com
Cloning into 'ftdi'...
error: cannot expand target domain name in SRV RR
fatal: Unable to look up developer.intra2net.com
CMake Error at /home/yao/Projects/SDK/gap_sdk_github/build/pulp-debug-bridge-prefix/src/pulp-debug-bridge-build/ftdi/tmp/ftdi-gitclone.cmake:40 (message):
  Failed to clone repository: 'git://developer.intra2net.com/libftdi'

-- Had to git clone more than once:
          3 times.`

Not sure "git://developer.intra2net.com/libftdi" is still reachable, please check.

Yao

haugoug, 2019-10-21:
Martin now that the sdk is provided as a single repo, I think we should include libftdi as well ?

Yaooooo, 2019-10-21:
Re-run with GWT's wifi, it works.

closed 2019-11-13 by sousoux