GreenWaves-Technologies/gap_sdk #74: Unable to adjust ssd_model.tflite with nntool - ReshapeParameters not implemented

tjmc, 2019-10-23:
In `gap_sdk/tools/nntool/tests/graph/ssd_model.tflite `
nntool fails 

I'm using `commit 504aff5f1305ce2605ce60c59085d5f396f89218 (HEAD -> master, tag: release-v3.0, origin/master, origin/HEAD)`

> Welcome to NNTOOL. Type help or ? to list commands.
> 
> (NNT) set debug true
> debug - was: False
> now: True
> (NNT) open ssd_model.tflite 
> nntool_shell - opening graph file ssd_model.tflite
> new_tflite_graph_all - Importing TFLITE model version 3
> nngraph - update graph dimensions
> nngraph - calculate liveness
> (NNT ssd_model.tflite 0) adjust
> adjust_order - step 0: input_1 adjust input 0 80x80x3 => c x h x w
> adjust_order - step 0: input_1 adjust output 0 80x80x3 => c x h x w
> adjust_order - step 1: CONV_2D_0_0 adjust weights 16x5x5x3 => out_c x in_c x h x w
> adjust_order - step 1: CONV_2D_0_0 adjust input 0 80x80x3 => c x h x w
> adjust_order - step 1: CONV_2D_0_0 adjust output 0 80x80x16 => c x h x w
> adjust_order - step 2: CONV_2D_0_0_activation adjust input 0 80x80x16 => c x h x w
> adjust_order - step 2: CONV_2D_0_0_activation adjust output 0 80x80x16 => c x h x w
> adjust_order - step 3: MAX_POOL_2D_0_1 adjust input 0 16x80x80 => c x h x w
> adjust_order - step 3: MAX_POOL_2D_0_1 adjust output 0 40x40x16 => c x h x w
> adjust_order - step 4: CONV_2D_0_2 adjust weights 16x5x5x16 => out_c x in_c x h x w
> adjust_order - step 4: CONV_2D_0_2 adjust input 0 16x40x40 => c x h x w
> adjust_order - step 4: CONV_2D_0_2 adjust output 0 40x40x16 => c x h x w
> adjust_order - step 5: CONV_2D_0_2_activation adjust input 0 40x40x16 => c x h x w
> adjust_order - step 5: CONV_2D_0_2_activation adjust output 0 40x40x16 => c x h x w
> adjust_order - step 6: CONV_2D_0_4 adjust weights 16x3x3x16 => out_c x in_c x h x w
> adjust_order - step 6: CONV_2D_0_4 adjust input 0 16x40x40 => c x h x w
> adjust_order - step 6: CONV_2D_0_4 adjust output 0 40x40x16 => c x h x w
> Traceback (most recent call last):
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 1753, in onecmd_plus_hooks
>     stop = self.onecmd(statement, add_to_history=add_to_history)
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 2148, in onecmd
>     stop = func(statement)
>   File "/home/tj/gap_sdk/tools/nntool/interpreter/nntool_shell.py", line 806, in do_adjust
>     self.execute_adjust_order()
>   File "/home/tj/gap_sdk/tools/nntool/interpreter/nntool_shell.py", line 797, in execute_adjust_order
>     self.G.adjust_order()
>   File "/home/tj/gap_sdk/tools/nntool/graph/nngraph.py", line 166, in adjust_order
>     adjust_order(self, reshape_weights)
>   File "/home/tj/gap_sdk/tools/nntool/graph/manipulations/adjust_order.py", line 49, in adjust_order
>     raise NotImplementedError("this needs to be implemented")
> NotImplementedError: this needs to be implemented
> EXCEPTION of type 'NotImplementedError' occurred with message: 'this needs to be implemented'

Pafrak, 2019-11-06:
Dear tjmc,
we does not yes support reshape in NNTool. This is at the top of our next features list.

Pafrak, 2020-06-08:
@sousoux Can you confirm that this has been added in version 3.5 of the sdk?

sousoux, 2020-06-08:
Yes. It has been added.