GreenWaves-Technologies/gap_sdk #75: nntool quantization fails with 3x3 kernel

tjmc, 2019-10-24:
aquant with a 3x3 kernel fails. 

[nntool_kernel_3x3_5x5.tar.gz](https://github.com/GreenWaves-Technologies/gap_sdk/files/3768716/nntool_kernel_3x3_5x5.tar.gz)

Call stack:
> Traceback (most recent call last):
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 1753, in onecmd_plus_hooks
>     stop = self.onecmd(statement, add_to_history=add_to_history)
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 2148, in onecmd
>     stop = func(statement)
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 272, in cmd_wrapper
>     return func(cmd2_app, args)
>   File "/home/tj/gap_sdk/tools/nntool/interpreter/nntool_shell.py", line 877, in do_aquant
>     stats_collector.collect_stats(self.G, [data])
>   File "/home/tj/gap_sdk/tools/nntool/stats/stats_collector.py", line 53, in collect_stats
>     stat = self._collect(G, input_tensors)
>   File "/home/tj/gap_sdk/tools/nntool/stats/activation_stats_collector.py", line 27, in _collect
>     execute_iterator(G, input_tensors, disable_cache=True):
>   File "/home/tj/gap_sdk/tools/nntool/execution/execute_graph.py", line 292, in execute_uncached_iterator
>     output, details = Executer.execute(fusion_node, output, qrec=fqrec)
>   File "/home/tj/gap_sdk/tools/nntool/execution/executer.py", line 41, in execute
>     return func(node, in_tensors, qrec)
>   File "/home/tj/gap_sdk/tools/nntool/execution/executer.py", line 156, in execute_conv2d
>     details=details)
>   File "/home/tj/gap_sdk/tools/nntool/execution/kernels/conv2d.py", line 284, in conv2d
>     qrec, details)
>   File "/home/tj/gap_sdk/tools/nntool/execution/kernels/conv2d.py", line 327, in process_input
>     res = np.dot(in_slice, w_slice)
> ValueError: shapes (6,) and (9,) not aligned: 6 (dim 0) != 9 (dim 0)
> EXCEPTION of type 'ValueError' occurred with message: 'shapes (6,) and (9,) not aligned: 6 (dim 0) != 9 (dim 0)'

3x3 log:

> (NNT 28_5x5.tflite 0) open 28_3x3.tflite
> nntool_shell - opening graph file 28_3x3.tflite                                                                                                                                                               
> new_tflite_graph_all - Importing TFLITE model version 3
> nngraph - update graph dimensions
> nngraph - calculate liveness
> (NNT 28_3x3.tflite 0) adjust
> adjust_order - step 0: input_1 adjust input 0 28x28x1 => c x h x w
> adjust_order - step 0: input_1 adjust output 0 28x28x1 => c x h x w
> adjust_order - step 1: DEPTHWISE_CONV_2D_0_0 adjust weights 16x3x3x1 => out_c x in_c x h x w
> adjust_order - step 1: DEPTHWISE_CONV_2D_0_0 adjust input 0 28x28x1 => c x h x w
> adjust_order - step 1: DEPTHWISE_CONV_2D_0_0 adjust output 0 14x14x16 => c x h x w
> adjust_order - step 2: DEPTHWISE_CONV_2D_0_0_activation adjust input 0 14x14x16 => c x h x w
> adjust_order - step 2: DEPTHWISE_CONV_2D_0_0_activation adjust output 0 14x14x16 => c x h x w
> adjust_order - step 3: DEPTHWISE_CONV_2D_0_1 adjust weights 1x3x3x16 => out_c x in_c x h x w
> adjust_order - step 3: DEPTHWISE_CONV_2D_0_1 adjust input 0 16x14x14 => c x h x w
> adjust_order - step 3: DEPTHWISE_CONV_2D_0_1 adjust output 0 14x14x16 => c x h x w
> adjust_order - step 4: DEPTHWISE_CONV_2D_0_1_activation adjust input 0 14x14x16 => c x h x w
> adjust_order - step 4: DEPTHWISE_CONV_2D_0_1_activation adjust output 0 14x14x16 => c x h x w
> adjust_order - step 5: output_1 adjust input 0 14x14x16 => c x h x w
> adjust_order - step 5: output_1 adjust output 0 14x14x16 => c x h x w
> nngraph - adjusted order
> nngraph - update graph dimensions
> nngraph - calculate liveness
> (NNT 28_3x3.tflite 0) fusions
> nngraph - update graph dimensions
> nngraph - calculate liveness
> (NNT 28_3x3.tflite 0) aquant -q 1 image/* -D 255
> nntool_shell - input file image/tmp.pgm
> execute_graph - execute uncached: quantization mode none
> Traceback (most recent call last):
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 1753, in onecmd_plus_hooks
>     stop = self.onecmd(statement, add_to_history=add_to_history)
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 2148, in onecmd                                                                                                                       
>     stop = func(statement)
>   File "/home/tj/.local/lib/python3.6/site-packages/cmd2/cmd2.py", line 272, in cmd_wrapper
>     return func(cmd2_app, args)
>   File "/home/tj/gap_sdk/tools/nntool/interpreter/nntool_shell.py", line 877, in do_aquant
[... 49 lines cut ...]
> nngraph - update graph dimensions
> nngraph - calculate liveness
> (NNT 28_5x5.tflite 0) aquant -q 1 image/* -D 255
> nntool_shell - input file image/tmp.pgm
> execute_graph - execute uncached: quantization mode none
> +------+----------------------------------+------+------+---------+------+-------+------+
> | Step |               Name               |  In  | Out  | Weights | Bias | Calc  | Acc  |
> +======+==================================+======+======+=========+======+=======+======+
> |  0   | input_1                          |      | Q1.7 |         |      |       |      |
> +------+----------------------------------+------+------+---------+------+-------+------+
> |  1   | DEPTHWISE_CONV_2D_0_0            | Q1.7 | Q2.6 | Q1.7    | Q2.6 | Q2.14 | Q2.6 |
> +------+----------------------------------+------+------+---------+------+-------+------+
> |  1   | DEPTHWISE_CONV_2D_0_0_activation | Q2.6 | Q2.6 |         |      |       |      |
> +------+----------------------------------+------+------+---------+------+-------+------+
> |  1   | DEPTHWISE_CONV_2D_0_0_fusion     | Q1.7 | Q2.6 |         |      |       |      |
> +------+----------------------------------+------+------+---------+------+-------+------+
> |  2   | DEPTHWISE_CONV_2D_0_1            | Q2.6 | Q1.7 | Q1.7    | Q1.7 | Q3.13 | Q1.7 |
> +------+----------------------------------+------+------+---------+------+-------+------+
> |  2   | DEPTHWISE_CONV_2D_0_1_activation | Q1.7 | Q1.7 |         |      |       |      |
> +------+----------------------------------+------+------+---------+------+-------+------+
> |  2   | DEPTHWISE_CONV_2D_0_1_fusion     | Q2.6 | Q1.7 |         |      |       |      |
> +------+----------------------------------+------+------+---------+------+-------+------+
> |  3   | output_1                         | Q1.7 | Q1.7 |         |      |       |      |
> +------+----------------------------------+------+------+---------+------+-------+------+
>

tjmc, 2019-10-24:
When I use VALID padding instead of SAME padding nntool quantization works.

[3x3_VALID_padding.tflite.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/3769187/3x3_VALID_padding.tflite.zip)

sousoux, 2019-11-13:
This is fixed by https://github.com/GreenWaves-Technologies/gap_sdk/commit/0833b4306bd9a052ad42d73256d598ff57734f2c
#

closed 2019-11-13 by sousoux