GreenWaves-Technologies/gap_sdk #77: Missing FlatBuffers Builder module

mariusz-r, 2019-11-08:
Builder module seems to be missing in `gap_sdk/tools/nntool/flatbuffers/` when compiling `gap_sdk/examples/nntool/mnist`:
```
Traceback (most recent call last):
  File "/home/greenwaves/gap_sdk/tools/nntool/nntool", line 86, in <module>
    main()
  File "/home/greenwaves/gap_sdk/tools/nntool/nntool", line 76, in main
    mod = importlib.import_module('interpreter.nntool_shell')
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/greenwaves/gap_sdk/tools/nntool/interpreter/nntool_shell.py", line 27, in <module>
    from importer.importer import create_graph
  File "/home/greenwaves/gap_sdk/tools/nntool/importer/importer.py", line 9, in <module>
    from .tflite.new_tflite_graph_all import create_graph as create_tflite_graph
  File "/home/greenwaves/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 41, in <module>
    from .tflite_schema_head import\
  File "/home/greenwaves/gap_sdk/tools/nntool/importer/tflite/tflite_schema_head/ConcatenationOptions.py", line 5, in <module>
    import flatbuffers
  File "/home/greenwaves/gap_sdk/tools/nntool/flatbuffers/__init__.py", line 15, in <module>
    from .builder import Builder
ModuleNotFoundError: No module named 'flatbuffers.builder'
model_rules.mk:57: recipe for target 'BUILD_MODEL_8BIT/Mnist.json' failed
make: *** [BUILD_MODEL_8BIT/Mnist.json] Error 1
```

Pafrak, 2019-11-12:
Hi mariusz-r,
did you install the requirements listed here:
https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/requirements.txt

mariusz-r, 2019-11-13:
Yes, I've followed the README.md file. Copying the missing file into the mentioned location fixes this for me.

Pafrak, 2019-11-13:
Ok, Great!

closed 2020-03-12 by Pafrak