GreenWaves-Technologies/gap_sdk #78: How to run helloworld

dookerpark, 2019-11-15:
Dear GreenWaves,

I have GAPUINO Combo Kit and install GAP_SDK. But when I run the example, there is error as followings. Please review it and would you let know how I can solve it?

dooker@dooker-vm:~/gap_sdk/examples/pmsis/helloworld$ make clean all run
mkdir -p /home/dooker/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV
riscv32-unknown-elf-gcc  -DCONFIG_GAPUINO -fno-jump-tables -fno-tree-loop-distribute-patterns -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -mchip=gap8 -mPE=8 -mFC=1 -D__riscv__ -D__pulp__ -DCONFIG_GAP -D__PULP_OS__ -MMD -MP -c helloworld.c  -I/home/dooker/gap_sdk/install/GAP8/include  -I/home/dooker/gap_sdk/install/GAP8/include/io  -I/home/dooker/gap_sdk/install/workstation/include  -include /home/dooker/gap_sdk/install/GAP8/include/rt/chips/gap/config.h  -MD -MF /home/dooker/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV/helloworld.d -o /home/dooker/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV/helloworld.o
cc1: fatal error: /home/dooker/gap_sdk/install/GAP8/include/rt/chips/gap/config.h: No such file or directory
compilation terminated.
/home/dooker/gap_sdk/tools/rules/pulp_rules.mk:108: recipe for target '/home/dooker/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV/helloworld.o' failed
make: *** [/home/dooker/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV/helloworld.o] Error 1

Rgds,
Dooker

dookerpark, 2019-11-15:
Hi GreenWaves,

I solved it. There was error of python version.
Let's close issue.

Rgds,
Dooker

ryanjtr, 2024-04-11:
i have the same problem, which python version you installed