GreenWaves-Technologies/gap_sdk #80: I2S drivers on FreeRTOS/PMSIS

Satish-Chandra-Mindtree, 2019-11-15:
Hello,

I see that the SDK doesnt have drivers for I2S on FreeRTOS/PMSIS. Would it be available on future versions ? .

Thanks,
Satish.

Satish-Chandra-Mindtree, 2019-11-18:
Any help in this regards will be helpful.

haugoug, 2019-11-19:
Hello,
indeed, the last release does not have any I2S driver for FreeRTOS/PMSIS. We will deliver it in the next release, end of next week, on the 29th. We can also provide you an early release with this driver on the 26th. Would that be suitable for you ?
Kind regards,
Germain

Satish-Chandra-Mindtree, 2019-11-19:
Hi Germain,

Yes that would be suitable . It would be good if you could give the early release on 26.

Thanks,
Satish.

haugoug, 2019-11-19:
Perfect we'll let you know when it is ready.

haugoug, 2019-11-21:
Could you also tell us which features you need (i2s format, word size, etc) ?

Satish-Chandra-Mindtree, 2019-11-21:
Hello,

Sorry for delayed reply. Didn’t notice your response on github.

Our application uses a data size of 32 bits, MSB first configuration.
By format are you referring to mode ? If yes then we are using in both master and slave mode.

Thanks,
Satish

Pafrak, 2020-03-12:
This is done in release 3.1

closed 2020-03-12 by Pafrak