GreenWaves-Technologies/gap_sdk #82: nntool example fails

GregorBader, 2019-11-20:
Making the NNTool Example fails with
`Error: CNN_LinearReLU Kernel: S3_Linear_10x64x4x4, Can't find a matching  basic kernel`

sousoux, 2019-11-20:
Thanks for the report. We'll take a look.

GregorBader, 2019-11-20:
The default configuration for 8 bit model fails, the 16Bit cofiguration succeeds
`make all MNIST_BITS=16`

sousoux, 2019-11-21:
Hi. Have you made any changes anywhere? I'm unable to reproduce this.

GregorBader, 2019-11-22:
Hi,
Of course I made changes in the examples & applications folder.
But I tried today and got a complete different error in "GENERATING NNTOOL STATE FILE":
```
nntool_shell - input file images/9208_4.pgm
execute_graph - execute uncached: quantization mode none
/home/osboxes/gap_sdk/tools/nntool/execution/kernels/misc.py:33: RuntimeWarning: overflow encountered in exp
  return np.exp(in_tensors)/np.sum(np.exp(in_tensors))
/home/osboxes/gap_sdk/tools/nntool/execution/kernels/misc.py:33: RuntimeWarning: invalid value encountered in true_divide
  return np.exp(in_tensors)/np.sum(np.exp(in_tensors))
/home/osboxes/anaconda3/lib/python3.7/site-packages/numpy/lib/function_base.py:3826: RuntimeWarning: Invalid value encountered in percentile
  interpolation=interpolation)
/home/osboxes/gap_sdk/tools/nntool/utils/stats_funcs.py:30: RuntimeWarning: invalid value encountered in less
  weak_min = (npa < quant1_3[0] - 1.5 * iqr)
/home/osboxes/gap_sdk/tools/nntool/utils/stats_funcs.py:31: RuntimeWarning: invalid value encountered in greater
  weak_max = (npa > quant1_3[1] + 1.5 * iqr)
/home/osboxes/gap_sdk/tools/nntool/utils/stats_funcs.py:32: RuntimeWarning: invalid value encountered in less
  strong_min = (npa < quant1_3[0] - 3 * iqr)
/home/osboxes/gap_sdk/tools/nntool/utils/stats_funcs.py:33: RuntimeWarning: invalid value encountered in greater
  strong_max = (npa > quant1_3[1] + 3 * iqr)
EXCEPTION of type 'ValueError' occurred with message: 'cannot convert float NaN to integer'
```

GregorBader, 2019-11-22:
Update:
A clean setup brought back the original error.

I think I have found the problem.
The autotiler model generated in my make does the 8 bit to 16 bit conversion in the CNN_LinearReLU() step. There does not exist a kernel for this -> so running the generator fails.
```
    // FULLY_CONNECTED_0_4
    CNN_LinearReLU("S3_Linear_10x64x4x4", 0, 1, 1, 2, 2, 1, 1, 1, 1, 1024, 10,
        KOP_LINEAR, KOP_NONE);
    // SOFTMAX_0_5
    CNN_SoftMax("S4_SoftMax", 0, 2, 2, 1, 1, 10, KOP_SOFTMAX);
    // output_1
```
After manually modifying the code, so that the 8 bit -> 16 bit conversion is done by CNN_SoftMax() it is working. I suppose my recent versions of the tools generate a slightly different tflite model than yours.

GregorBader, 2019-11-22:
Maybe this change introduced the error:
https://github.com/GreenWaves-Technologies/gap_sdk/commit/0833b4306bd9a052ad42d73256d598ff57734f2c#diff-4a4a3be81ad8b40890ab113781f2e66dR85

anujdutt9, 2019-11-22:
Even I got this error. Looks like it's due to a missing file "libtile.a"

```
error: /home/anujd9/gap_sdk/tools/autotiler_v3/lib/libtile.a: No such file or directory
model_rules.mk:71: recipe for target 'BUILD_MODEL_8BIT/GenTile' failed
```

**Command used:**  make all run

Yaooooo, 2019-11-22:
Hi Gregor,

Indeed, I think there's a version problem here, I tried both 8 bits and 16 bits from my side, last commit of master, and both 8&16 works. Here's the code generated:

(base) yao@ubuntu:~/Projects/SDK/gap_sdk_github/examples/nntool/mnist$ grep -rnI "CNN_LinearReLU"
BUILD_MODEL_16BIT/MnistModel.c:58:    CNN_LinearReLU("S3_Linear_10x64x4x4", 0, 2, 2, 2, 2, 1, 1, 1, 1, 1024, 10,
BUILD_MODEL_8BIT/MnistModel.c:58:    CNN_LinearReLU("S3_Linear_10x64x4x4", 0, 1, 1, 1, 1, 1, 1, 1, 1, 1024, 10,

And it should be found in this file:
gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Generators.c 
Line 1537

Here's the doc for this generator api: https://greenwaves-technologies.com/manuals/BUILD/AUTOTILER/html/group__groupCNN.html#ga83f9923a7b31566cb1ead503da9ee64f

Best,
Yao

Yaooooo, 2019-11-22:
Please do the following command in your sdk to trigger the download of
autotiler lib:
cd gap_sdk
make autotiler

sousoux, 2019-11-23:
I confirm. This is a mismatch between autotiler version and nntool. I added an autotiler generator for an 8bit input to 16 bit output with 8 bit weights linear layer and nntool wants to select that because the accuracy will be better for no overhead (10 16 bit outputs fit in L1 anyway). Updating the autotiler generators should fix this.

GregorBader, 2019-11-25:
Hi Martin, Yao,
I'm on latest commit, too. But nntool generates this:
```
(base) osboxes@osboxes:~/gap_sdk/examples/nntool/mnist$ grep -rnI "CNN_LinearReLU"
BUILD_MODEL_8BIT/MnistModel.c:58:    CNN_LinearReLU("S3_Linear_10x64x4x4", 0, 1, 1, 2, 2, 1, 1, 1, 1, 1024, 10,
```
So I'm unsure wheter I still made some mistake, or what to do to fix it?!?

sousoux, 2019-11-25:
Hi Gregor

There is a mismatch between nntool and the autotiler generators. Classic. I
fixed a bug but that introduced changes that passed all regression tests on
our master because we already have all the new generators. I'll see if I
can find a temporary workaround tomorrow.

Sorry

Martin

isakim, 2020-01-07:
Hello, 

I'm trying to run the same nntool example. And I'm getting the same error:

`Error: CNN_LinearReLU Kernel: S3_Linear_10x64x4x4, Can't find a matching basic kernel
`
Could you find a solution to this error?

Thank you very much,
Isabella

Pafrak, 2020-06-08:
This has been solved with recent versions of sdk

closed 2020-06-08 by Pafrak