GreenWaves-Technologies/gap_sdk #86: nntool example problems

GregorBader, 2019-11-27:
I'm using gap_sdk [f23a8be](https://github.com/GreenWaves-Technologies/gap_sdk/commit/f23a8be695446a570f3a08fb8c4db64d3b34f9c5) with `configs/gapoc_a_v2.sh`.
I'm building `examples/nntool/mnist` with `make clean all run MNIST_BITS=16`.

Following problems occur:
- Hyperflash fails with error in function `load_jtag_hyper()`. This can be fixed with:
```
diff --git a/tools/pulp_tools/pulp-debug-bridge/python/bridge/chips/gap_rev1.py b/tools/pulp_tools/pulp-debug-bridge/python/bridge/chips/gap_rev1.py
index e22272c..c79ddd4 100644
--- a/tools/pulp_tools/pulp-debug-bridge/python/bridge/chips/gap_rev1.py
+++ b/tools/pulp_tools/pulp-debug-bridge/python/bridge/chips/gap_rev1.py
@@ -409,7 +409,7 @@ class gap_debug_bridge(debug_bridge):
         return True
 
 
-    def load_jtag_hyper(self, binaries):
+    def load_jtag_hyper(self):
 
         if self.verbose:
             self.log(0, 'Loading binary through jtag_hyper')
```
- Cluster fails with memory exception. In the generated kernel constructor `MnistCNN_Construct()` the pointers are not checked for `NULL` so the cluster is started as usual. I suggest this should be done in the constructor as it is done in the `applications/WaterMeter/Mnist.c`. The problem here is, that the stacks eat up too much heap, so the fix is:
```
diff --git a/examples/nntool/mnist/Mnist.c b/examples/nntool/mnist/Mnist.c
index 6bbdd2f..5cc3c41 100644
--- a/examples/nntool/mnist/Mnist.c
+++ b/examples/nntool/mnist/Mnist.c
@@ -20,7 +20,7 @@
 #endif
 
 
-#define STACK_SIZE      2048
+#define STACK_SIZE      1024
 #define MOUNT           1
 #define UNMOUNT         0
 #define CID             0
@@ -187,6 +187,7 @@ int main()
     printf("Graph constructor exited with an error\n");
     return 1;
   }
+  printf("Mem: %p %p\n", L1_Memory, Mnist_L2_Memory);
 
   printf("Call cluster\n");
   // Execute the function "RunCifar10" on the cluster.
```
- Running on GVSOC fails (no output at all, on gapuino config it is working!). This can be fixed by adding the runner config to 'install/workstation/configs/config/gapoc_a.json':
```
  "runner": {
    "config": {
      "boot-mode": "rom_hyper",
      "flash_type": "hyper",
      "default_profile": "hyper"
    }
  },
```
- Different results for same binary running on target vs. GVSOC!
GVSOC recognizes correctly a '5' on image 'samples/5223_5.pgm', the gap8 recognizes '0'!
I havent' found a fix for this.

Pafrak, 2019-11-28:
Hi GregorBader,
 thanks for the comments, we will release an sdk update by the end on this week, with these fixes as well.

closed 2019-11-28 by Pafrak

roshantha9, 2019-12-10:
Hi Pafrak and GregorBader,
I applied GregorBader's patch. Then the "load_jtag_hyper" error goes away. But, not sure if the board is flashed correctly. Please see log below. I do not get anything on the UART pins (P4.7, 4.8). Tested on Gapuino Model 1.1. UART works when I load and run from jtag. 

I'm testing the following example: examples\pmsis\test_features\uart_delegate
in the makefile I added this : 
PLPBRIDGE_FLAGS += -f -hyper

---------------------------------GAPUINO------------------------------------

Reading ELF file
  Segment : 1b000004 440
  Segment : 1b000444 83c
  Segment : 1c000000 4018
  Segment : 1c004170 18
Generating boot binary
  Nb areas: 4
  Area 0: offset: 0x400, base: 0x1b000004, size: 0x440, nbBlocks: 2
  Area 1: offset: 0xc00, base: 0x1b000444, size: 0x83c, nbBlocks: 3
  Area 2: offset: 0x1800, base: 0x1c000000, size: 0x4018, nbBlocks: 17
  Area 3: offset: 0x5c00, base: 0x1c004170, size: 0x18, nbBlocks: 1
Generating files (header offset: 0x6000)
Bridge init - log level 2
Program exited
Program exited
Loading binary through jtag
Notifying to boot code new boot mode (mode: 1)
ftdi2232 libftdi version 1.4 (major: 1, minor: 4, micro: 0, snapshot ver: unknown)
Found ftdi device i:0x403:0x6010:0
Connecting to ftdi device i:0x403:0x6010:0
Cable state change connected
Waiting for notification from boot code
Received for notification from boot code
Loading binary from /home/osboxes/Documents/gap_sdk/install/workstation/bin/flasher
Loading section (base: 0x1b000004, size: 0x324)
Loading section (base: 0x1b000328, size: 0x908)
Loading section (base: 0x1c000000, size: 0x650c)
Init section to 0 (base: 0x1c00650c, size: 0xec)
Loading section (base: 0x1c0065f8, size: 0x44)
debug address 0x1c000090 contents 0x1c0062d8
Starting execution
[Flasher]: Flasher Version 1.0 start
[Flasher]: pad config done
[Flasher]: conf finished.
Initializing bridge flasher
Flash address buffer 0x1c028730
Bridge flasher completed
[Flasher]: Flasher Finish
Detected end of application, exiting with status: 0
Loop exited
commands completed
Bridge init - log level 2
Program exited
Loading binary through jtag_hyper
Notifying to boot code that we are doing a JTAG boot from hyperflash
ftdi2232 libftdi version 1.4 (major: 1, minor: 4, micro: 0, snapshot ver: unknown)
Found ftdi device i:0x403:0x6010:0
Connecting to ftdi device i:0x403:0x6010:0
Cable state change connected
Loop exited
commands completed