GreenWaves-Technologies/gap_sdk #90: GAPuino Rel.B breaks Sensor Board compatibility

GregorBader, 2019-12-09:
GAPuino Rel. B has removed I2S1_SDI from P4 connector. Therefor half of the microphones are not usable any more!

A test with `examples/pmsis/test_periph/i2s/pdm_4mic` even shows, that only 1 out of the 4 mic's are working:
´´´
Starting execution
Entering main controller
Interface 0 / channel 0
  Sample 0: -1640
  Sample 1: 5393
  Sample 2: -2442
  Sample 3: 5407
  Sample 4: -3329
  Sample 5: 3911
  Sample 6: 1399
  Sample 7: 8035
  Sample 8: 12571
  Sample 9: -5118
  Sample 10: -4563
  Sample 11: 1690
  Sample 12: 954
  Sample 13: 16759
  Sample 14: 11322
  Sample 15: -1367
Interface 0 / channel 1
  Sample 0: -32768
  Sample 1: -32768
  Sample 2: -32768
  Sample 3: -32768
  Sample 4: -32768
  Sample 5: -32768
  Sample 6: -32768
  Sample 7: -32768
  Sample 8: -32768
  Sample 9: -32768
  Sample 10: -32768
  Sample 11: -32768
  Sample 12: -32768
  Sample 13: -32768
  Sample 14: -32768
  Sample 15: -32768
Interface 1 / channel 0
  Sample 0: -32768
  Sample 1: -32768
  Sample 2: -32768
  Sample 3: -32768
  Sample 4: -32768
  Sample 5: -32768
  Sample 6: -32768
  Sample 7: -32768
  Sample 8: -32768
  Sample 9: -32768
  Sample 10: -32768
  Sample 11: -32768
  Sample 12: -32768
  Sample 13: -32768
  Sample 14: -32768
  Sample 15: -32768
Interface 1 / channel 1
  Sample 0: -32768
  Sample 1: -32768
  Sample 2: -32768
  Sample 3: -32768
  Sample 4: -32768
  Sample 5: -32768
  Sample 6: -32768
  Sample 7: -32768
  Sample 8: -32768
  Sample 9: -32768
  Sample 10: -32768
  Sample 11: -32768
  Sample 12: -32768
  Sample 13: -32768
  Sample 14: -32768
  Sample 15: -32768
Detected end of application, exiting with status: 0
´´´

Satish-Chandra-Mindtree, 2019-12-12:
I too am seeing similar issue.

haugoug, 2019-12-13:
There is indeed an issue, thanks for reporting, we are fixing it.

haugoug, 2019-12-18:
We have fixed the issue, you can get an intermediate release of the SDK in branch 3.1.1_dev and try this example: examples/pmsis/test_periph/i2s/pdm_4mic/
This example needs OpenOCD to transfer the output samples to the workstation.
Let us know if this fixes the issue or not for you.

closed 2020-03-12 by Pafrak