GreenWaves-Technologies/gap_sdk #91: Cannot boot from HyperFlash

roshantha9, 2019-12-13:
Hi, 
I'm sorry for posting this issue here, as I don't know completely if it's related to the SDK or HDK. However, the forum seems quite inactive and no one has replied to this issue yet on the forum in the last 3-4 days. We have purchased multiple gapuino boards in the hope of using them for energy harvesting applications (powered externally- e.g. from solar power), so would like to get a minimum example working (booting) from Flash not from USB. 

I am using a **Gapuino 1.1** board (purchased after Oct. 2019). Using **SDK 3.1**, Ubuntu 16.04 VM, USB 3.0.  have sourced configs/gapuino_v2.sh. Have also applied the patch/fix for load_jtag_hyper() as mentioned here: https://github.com/GreenWaves-Technologies/gap_sdk/issues/86

Booting from JTAG works fine, I can see uart, LEDs blink, no errors, bridge works fine. But, I have tried to boot several simple applications from FLASH, but the process described in the docs do not work. For the blink LED example : examples\pmsis\test_periph\blink_led

Added 
**PLPBRIDGE_FLAGS  += -f -hyper**
to the makefile

According to the log (high verbose) below, it appears as if the binary has been flashed, but the board does not boot and execute from the Flash. I have tried power cycling, reseting while connected to USB, external power sources etc. nothing seems to boot the board from the Flash.

make clean all run

+ echo ———————————GAPUINO————————————
———————————GAPUINO————————————
+ echo

+ pwd
+ here=/home/osboxes/Documents/gap8_sandpit/blink_led
+ cd /home/osboxes/Documents/gap8_sandpit/blink_led/BUILD/GAP8_V2/GCC_RISCV
+ FLAG_FLASH=0
+ FLAG_SCRIPT=0
+ FLAG_BOOT=0
+ FLAG_NORUN=0
+ FLAG_NOFLASH=0
+ verbose=4
+ fileio=0
+ [ gapuino = gapoc_a ]
+ cable=ftdi@digilent
+ boot_mode=jtag
+ gdb=
+ ioloop=ioloop
+ reqloop=reqloop
+ binary=–binary=test
+ buildFlashImage=
+ [ 2 -eq 0 ]
+ [ 2 -gt 0 ]
+ FLAG_FLASH=1
+ buildFlashImage=flashImageBuilder –verbose –flash-boot-binary=test –raw=flashImg.raw
+ shift
+ [ 1 -gt 0 ]
+ ioloop=
+ reqloop=
+ binary=
+ boot_mode=jtag_hyper
+ shift
+ [ 0 -gt 0 ]
+ [ -n ]
+ [ 1 -eq 1 ]
+ eval flashImageBuilder –verbose –flash-boot-binary=test –raw=flashImg.raw
+ flashImageBuilder –verbose –flash-boot-binary=test –raw=flashImg.raw
Reading ELF file
Segment : 1b000004 3f0
Segment : 1b0003f4 83c
Segment : 1c000000 3aac
Segment : 1c003be8 18
Generating boot binary
Nb areas: 4
Area 0: offset: 0x400, base: 0x1b000004, size: 0x3f0, nbBlocks: 1
Area 1: offset: 0x800, base: 0x1b0003f4, size: 0x83c, nbBlocks: 3
Area 2: offset: 0x1400, base: 0x1c000000, size: 0x3aac, nbBlocks: 15
Area 3: offset: 0x5000, base: 0x1c003be8, size: 0x18, nbBlocks: 1
Generating files (header offset: 0x5400)
+ plpbridge –verbose=4 –cable=ftdi@digilent –boot-mode=jtag –flash-image=flashImg.raw –chip=gap_rev1 flash wait
Bridge init – log level 4
Program exited
Waiting termination
Program exited
Loading ELF binaries (path: /home/osboxes/Documents/gap_sdk/install/workstation/bin/flasher)
Loading binary through jtag
Notifying to boot code new boot mode (mode: 1)
Using debug IR: 0x4
Using retry count: 0
Checking errors: 0
Using access timeout: 1000000 us
ftdi2232 libftdi version 1.4 (major: 1, minor: 4, micro: 0, snapshot ver: unknown)
iterating 1 devices
try opening i:0x403:0x6010:0
Found ftdi device i:0x403:0x6010:0
iterating 1 devices
try opening i:0x1D6B:0x2:0
Connecting to ftdi device i:0x403:0x6010:0
Connected to libftdi driver.
Cable state change connected
JTAG chain len is 1
adv_dbg_itf: jtag_ir_chainlen = 4
JTAG IR len is 4, DR len is 32
Device 0 ID: 149511C3
ft2232: chip reset (active: 1)
ft2232: chip reset (active: 0)
Waiting for notification from boot code
Received for notification from boot code
adv_dbg_itf: select ir 4
Loading binary from /home/osboxes/Documents/gap_sdk/install/workstation/bin/flasher
Loading section (base: 0x1b000004, size: 0x324)
Loading section (base: 0x1b000328, size: 0x908)
Loading section (base: 0x1c000000, size: 0x650c)
Init section to 0 (base: 0x1c00650c, size: 0xec)
Loading section (base: 0x1c0065f8, size: 0x44)
Lauching IO loop
debug address 0x1c000090 contents 0x1c0062d8
ReqLoop started
Starting execution
Starting execution
Flashing the flash image
adv_dbg_itf: select ir 4
[Flasher]: Flasher Version 1.0 start
[Flasher]: pad config done
[Flasher]: conf finished.
Initializing bridge flasher
Flash address buffer 0x1c028730
Bridge flasher completed
adv_dbg_itf: select ir 4
[Flasher]: Flasher Finish
adv_dbg_itf: select ir 4
Detected end of application, exiting with status: 0
Loop exited
commands completed
+ [ 0 -eq 0 ]
+ pid_bridge=3183
+ [ -z ]
+ [ 0 -gt 0 ]
+ [ 0 -eq 0 ]
+ wait 3183
+ plpbridge –verbose=4 –cable=ftdi@digilent –boot-mode=jtag_hyper –chip=gap_rev1 load start wait
Bridge init – log level 4
Waiting termination
Program exited
Loading ELF binaries (path: )
Loading binary through jtag_hyper
Notifying to boot code that we are doing a JTAG boot from hyperflash
Using debug IR: 0x4
Using retry count: 0
Checking errors: 0
Using access timeout: 1000000 us
ftdi2232 libftdi version 1.4 (major: 1, minor: 4, micro: 0, snapshot ver: unknown)
iterating 1 devices
try opening i:0x403:0x6010:0
Found ftdi device i:0x403:0x6010:0
iterating 1 devices
try opening i:0x1D6B:0x2:0
Connecting to ftdi device i:0x403:0x6010:0
Connected to libftdi driver.
Cable state change connected
JTAG chain len is 1
adv_dbg_itf: jtag_ir_chainlen = 4
JTAG IR len is 4, DR len is 32
Device 0 ID: 149511C3
ft2232: chip reset (active: 1)
ft2232: chip reset (active: 0)
Starting execution
Loop exited
commands completed

haugoug, 2019-12-13:
The procedure is different with the new gapuino, I'm adding information about it, and let you know today once it's ready.

roshantha9, 2019-12-13:
Hi haugoug, I really appreciate your reply. I look forward to the new information.

roshantha9, 2019-12-16:
Hi, just want to check if the documentation has been updated with the Flash-boot information for the new gapuino (v1.1) board ? I can help test different examples on my board if you wish. thanks

roshantha9, 2019-12-18:
@haugoug - Please can you give some indication on the procedure to boot the new gapuino from Flash. sorry for troubling, but we have halted development on the board until this issue is resolved. Am I doing something incorrect in the above process ?  Thank you

haugoug, 2019-12-18:
This was not only a lack of documentation, you will have to get an intermediate release of the SDK in branch 3.1.1_dev. 
The procedure for the new gapuino board is different, you will first need to burn an efuse to setup the right flash for the boot. You will need openocd for that. You can find some information in the doc, but you just need to execute this script and confirm:
openocd-fuser-hyperflash
Then you can use the examples as before. Let us know if it is fine.

roshantha9, 2019-12-18:
@haugoug - thank you. But, if I burn the efuse, is it a "one-time" operation ? or can I continue to reprogram the flash and boot into a different binary each time I run openocd ? can I interchange between booting from flash and USB (jtag) as I please, after I have burned the efuse ?

haugoug, 2019-12-18:
Indeed it's a one-time operation, after that, the chip will only be able to boot from hyperflash. Booting from SPI flash will for example not be possible anymore. Booting from JTAG will still be possible. Even after you execute this procedure, you can still use openocd to upload a new binary or flash a new one. Just to be sure, you are having the Gapuino board with USB micro B, that you use with configs/gapuino_v2.sh, right ?

roshantha9, 2019-12-18:
Yes correct - I have the board with USB micro B and use the configs/gapuino_v2.sh. We do not intend to use SPI flash, but definitely hyperflash (working over a super-capacitor / energy harvesting unit) and USB (for debugging/dev)

Actually, I have previously tried to use openocd with the sdk 3.1 release, but had trouble compiling it. I will try again and see. I will report any issues.

wei-1006, 2019-12-19:
Hi, I have the same question, but I got the old version (V1.0).
I also add PLPBRIDGE_FLAGS += -f -hyper to Makefile.
What should I do to make the board act like Arduino board, which means that after I load the binary file, then unplug USB, then put it back, the board can just do what I loaded .
Sorry for posting here.

roshantha9, 2019-12-20:
@wei-1006 - you can try using openocd as described above (by haugoug). I haven't tried it myself yet, due to other tasks, but there is brief guideline here : https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/gap8-openocd-tools/README 
Please share your progress. Thanks

roshantha9, 2019-12-23:
@haugoug - I can confirm that I can now boot from Flash using the openocd-fuser-hyperflash script first, and then using the "PLPBRIDGE_FLAGS += -f -hyper" flags to compile the examples. 

However, there are still a few issues - 
1) I have to first press RESET every time I boot before the program can start. I thought the board would automatically boot as soon as it receives power, but it seems I have to press RESET first, then it starts. Is this the expected behavior ?

2) Boot from Flash does not seem to work when powering the board from VIN (P1.8) /Aurduino P1.5 (5V) . I have tested using the Blink LED example. The gapuino Green LED light comes on, but not the blink blue light. However, powering from USB works - blink LED program runs from flash. Is there a reason why Flash bootup from external power does not work ? Note - I have set the J21/J22/J23 jumpers correctly, it seems like the board is getting power, but bot the Gap8/HyperFlash.  I have tried supplying power (using a external DC power supply) at both 3V and 5V on the above P1.5/P1.8 pins. both do not work.  Any idea why this might be ?

Yaooooo, 2020-09-29:
Hi all, This bug has been fixed, please read this: https://gwt-website-files.s3.eu-central-1.amazonaws.com/AN004_BootFromFlash.pdf
to have a reliable boot.

closed 2020-09-29 by Yaooooo