GreenWaves-Technologies/gap_sdk #93: Problem using PLPBRIDGE

GuillaumeDEVIC, 2019-12-18:
Hi,

I acquired the GAPuino board recently (end of November 2019) and I am in the process of familiarizing myself with the board and the applications within the SDK.(using Ubuntu 16.04 VM, USB 3.0)
I sourced /configs/gapuino_v2.sh and sourceme.sh.
The installing of the toolchain is successfully passed, however, by manipulating the Mnist application (/examples/autotiler/Mnist) I cannot change the inference input image :

**origin declaration**
#define IMG_DIR test_img
#define NUM_DIR 6
#define NUM_PIC 1578
**my modifications**
#define IMG_DIR test_img
#define NUM_DIR 2
#define NUM_PIC 1574

Then, I tested the Bridge connection with the program "test_bridge.c" ("examples/native/pulpos/bridge"). The program does not run, there is no display after "GAP8" of the script tools/runner/run_gapuino.sh. Even with the plpbridge command executed on another terminal, nothing happens after "Connecting to bridge". However, I get the following error:

$ plpbridge-rt --binary=~/gap_sdk/examples/native/p lpos/bridge/BUILD/GAP8_V2/GCC_RISCV --chip=gap

/home/osboxes/gap_sdk/install/workstation/bin/plpbridge-rt:25: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Traceback (most recent call last):
  File "/home/osboxes/gap_sdk/install/workstation/bin/plpbridge-rt", line 96, in <module>
    bridge.reqloop()
  File "/home/osboxes/gap_sdk/install/workstation/python/bridge/rt_bridge.py", line 60, in reqloop
    addr = self._get_binary_symbol_addr('__rt_debug_struct_ptr')
  File "/home/osboxes/gap_sdk/install/workstation/python/bridge/rt_bridge.py", line 44, in _get_binary_symbol_addr
    with open(binary, 'rb') as file:
IsADirectoryError: [Errno 21] Is a directory: '/home/osboxes/gap_sdk/examples/native/pulpos/bridge/BUILD/GAP8_V2/GCC_RISCV'

Does anyone have a solution to this problem?

Additional question: When running applications such as Hello_World and Mnist, I observed the following warning sometimes:
Warn: Haven't made progress in mpsse_flush () for 2014ms.
What does it mean?

Thanks in advance.

Yaooooo, 2020-01-10:
Hi,
Sorry for the late reply. Have you resloved your problem?
I tried from my side with the last version of SDK, I cannot reproduct your problem. 
The result is like this:
`/home/yao/Projects/SDK/gap_sdk_github/install/workstation/runner/run_gapuino.sh -fileIO 2

          ,ad8888ba,        db        88888888ba   ad88888ba   
         d8"'    `"8b      d88b       88      "8b d8"     "8b  
        d8'               d8'`8b      88      ,8P Y8a     a8P  
        88               d8'  `8b     88aaaaaa8P'  "Y8aaa8P"   
        88      88888   d8YaaaaY8b    88""""""'    ,d8"""8b,   
        Y8,        88  d8""""""""8b   88          d8"     "8b  
         Y8a.    .a88 d8'        `8b  88          Y8a     a8P  
          `"Y88888P" d8'          `8b 88           "Y88888P"   

---------------------------------GAPUINO------------------------------------

/home/yao/Projects/SDK/gap_sdk_github/install/workstation/bin/plpbridge:25: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Bridge init - log level 2
Program exited
Loading binary through jtag
Notifying to boot code new boot mode (mode: 1)
ftdi2232 libftdi version 1.4 (major: 1, minor: 4, micro: 0, snapshot ver: unknown)
Found ftdi device i:0x403:0x6010:0
Connecting to ftdi device i:0x403:0x6010:0
Cable state change connected
Waiting for notification from boot code
Received for notification from boot code
Loading binary from test
Loading section (base: 0x1b000004, size: 0x3f0)
Loading section (base: 0x1b0003f4, size: 0x83c)
Loading section (base: 0x1c000000, size: 0x6b1c)
Init section to 0 (base: 0x1c006b1c, size: 0x21bc)
Loading section (base: 0x1c008cd8, size: 0x18)
debug address 0x1c000090 contents 0x1c0068d4
debug address 0x1c000090 contents 0x1c0068d4
RSP server setup on port 9999
Starting execution

	 *** PMSIS Bridge Test ***

Entering main controller
Connecting to bridge
/home/yao/Projects/SDK/gap_sdk_github/install/workstation/bin/plpbridge-rt:25: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Tcp_listener: client connected (fd 14)
client connected
Connection done

Test success with 0 error(s) !
Detected end of application, exiting with status: 0
Loop exited
commands completed
`
FYI, in the last 2 releases, we have released our openOCD implementation. Now you can use openOCD instead of plpbridge. In the future, the bridge tool will be removed from our SDK.

Yao

GuillaumeDEVIC, 2020-01-24:
Hi,

Thank you for your reply.

1. To begin, I have resolved my additional question. I forgot to select the USB-3 controller in the virtual machine settings in VirtualBox.

2. Even when launching directly the script "gap_sdk/install/workstation/runner/run_gapuino.sh -fileIO 2" , it doesn't work.
The best I could get is by removing "-fileIO 2" from the Makefile (examples/native/pulpos/bridge), but the execution is blocked after the connection is done.

          ,ad8888ba,        db        88888888ba   ad88888ba   
         d8"'    `"8b      d88b       88      "8b d8"     "8b  
        d8'               d8'`8b      88      ,8P Y8a     a8P  
        88               d8'  `8b     88aaaaaa8P'  "Y8aaa8P"   
        88      88888   d8YaaaaY8b    88""""""'    ,d8"""8b,   
        Y8,        88  d8""""""""8b   88          d8"     "8b  
         Y8a.    .a88 d8'        `8b  88          Y8a     a8P  
          `"Y88888P" d8'          `8b 88           "Y88888P"   

---------------------------------GAPUINO------------------------------------

Open On-Chip Debugger 0.10.0+dev-g823615c (2019-12-18-02:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : Listening on port 3333 for gdb connections
Loading binary through JTAG
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Burst read timed out

	 *** PMSIS Bridge Test ***

Entering main controller
Connecting to bridge
Connection done

Otherwise, can I easily use OpenOCD for the autotiler/Mnist example? And especially do I have to download an updated version of the sdk to use OpenOCD?

Thanks in advance.

sivasivarajah, 2020-01-24:
Hi Guillaume,

The program is hanging because it is trying to reach the plpbridge while you are using OpenOCD. Both can not be used at the same time.

> Otherwise, can I easily use OpenOCD for the autotiler/Mnist example? And especially do I have to download an updated version of the sdk to use OpenOCD?

Yes you can, using either plpbridge or OpenOCD.
To run with OpenOCD, execute _make clean all run io=host_
With plpbridge, you don't need _io_ part.
Also as said previously, plpbridge will be deprecated soon. It is best to use OpenOCD.

As for the updates, yes you should. You can either download latest sources on **master** branch, or use the current dev branch **3.1.1_dev**.

Siva.

GuillaumeDEVIC, 2020-01-24:
Thank you for your quick reply.

So, I'm not sure I understood how we defined if the program will use plpbridge or OpenOCD.
Currently from what I understood, if I want to use plpbridge I define USE_BRIDGE = 1 in the Makefile and export GAP_USE_OPENOCD = 0. And conversely, if I want to use OpenOCD.

But if I set USE_BRIDGE = 0 in the Makefile, export GAP_USE_OPENOCD = 1 and I run the program Mnist.c (example/autotiler/Mnist) with "make clean all run io = host" the example ends with "Test success with 0 errro (s)! "(at first seen it's ok). However, I cannot modify the image which must be classified. Whatever the image I define in Mnist.c, the program always predicts the number 6.
 
Below are the modifications to define the image to be classified.
origin declaration
#define IMG_DIR test_img
#define NUM_DIR 6
#define NUM_PIC 1578
=> Recognized number : 6

my modifications
#define IMG_DIR test_img
#define NUM_DIR 0
#define NUM_PIC 3
=> Recognized number : 6

I had assumed that the problem was with plpbridge, but apparently I may have a problem with OpenOCD instead.

sivasivarajah, 2020-01-24:
Hi Guillaume,

I think there is some misunderstanding on the Makefile.
The flag **USE_BRIDGE** is used to flash a picture in memory flash in order to find the number of the picture. As said in the Makefile, for the moment the tool used to flash the picture in memory is plpbridge. But soon, it will be deprecated and OpenOCD wil be used to flash those pictures.

And setting **USE_BRIDGE** to 0 will not put any picture in memory, it will use a golden model, cf. file golden.h, to take as an example to use with Mnist. The golden model is an array representing a "6". That is why you get a 6 even if you change the name file.

If you want to try other picture file do as follow :
- USE_BRIDGE=1 in Makefile
- Edit NUM_DIR and NUM_PIC
- then make [clean] all run

Since plpbridge is used to flash the image file, OpenOCD can not be used at the same time.

Siva.

GuillaumeDEVIC, 2020-02-03:
Hi,

In seeking the source of the problem, I forgot that define USE_BRIDGE = 0 allow the use of golden.h. Thanks for the reminder.

I followed your previous indications:
1. USE_BRIDGE=1 in Makefile
2. Edit NUM_DIR=0 and NUM_PIC=3
3. make clean all run

No results, still blocked after displaying the large GAP8.

To make sure that the problem does not come from my installation of the SDK, I reinstalled the SDK on VM Ubuntu 18.04 and tested the installation with some example, such as HelloWorld, nntool/mnist.

But, I still cannot modify the image to classify in /examples/autotiler/Mnist. I also tried using OpenOCD by adding the following lines in the Makefile according to https://github.com/GreenWaves-Technologies/gap_sdk/issues/101:
export GAP_USE_OPENOCD=1
io=host

However, the image change works with the nntool/mnist example but not for /examples/autotiler/Mnist.

sivasivarajah, 2020-02-04:
Hi,

Can you first try this example? examples/native/pulpos/bridge
This will copy the example file. It will load the c test file using plpbridge then reproduce into out_bridge.c.

If this works, then the Mnist example should also work.
We tried on native Ubuntu 18.04, and a VM Ubuntu 16.04, all work well.
Could you send a full trace when you do make run ?

GuillaumeDEVIC, 2020-02-05:
Hi,

Well, I think I have identified the source of the problem, it is the export variable GAP_USE_OPENOCD which stops the execution (whatever its value 0 or 1)

I was able to correctly run the example "examples/native/pulpos/bridge" and the one that gave me problem "example/autotiler/Mnist".
Only if the variable GAP_USE_OPENOCD(0 or 1) was not defined in the environment variables of the terminal.

Can this come from my installation of OpenOCD?
And is it reproducible on your side? (define GAP_USE_OPENOCD(0 or 1) then execute examples/native/pulpos/bridge)

sivasivarajah, 2020-02-05:
Hi,

For this Mnist example, you should not use OpenOCD when trying to load an image file.
So you can not export this variable(GAP_USE_OPENOCD), because then the flash part will use OpenOCD while the image loading will use plpbridge, but those two are not compatible. Only one can be used.

By the way, if you want to use OpenOCD(for other applications), you can just set io=host in the Makefile or on command line.

closed 2020-02-10 by GuillaumeDEVIC