GreenWaves-Technologies/gap_sdk #95: Documentation for NNTools

s-shashidhar, 2019-12-20:
Hi,

I plan to use a tensorflow model to deploy on gap8. I was hoping to use nntools for model conversion from tensorflow to gap8 format, but I could not find any documentation. Any help regarding the model conversion or nntools documentation is appreciated.

Thanks

Yaooooo, 2019-12-22:
Hi,

We are sorry that for you inconvenience.
We are actually preparing such a manual and doc, which would be available
soon.
If you want, please send me your email address, and company name to:
support@greenwaves-technologis.com
We will send you a beta version, and a presentation of our NNTools.

Best regards
Yao

isakim, 2020-01-15:
Hello, 

I'm also trying to deploy a tensorflow model on GAP8 through NNtool. Could you please send me this beta version of the documentation? It'd help me a lot.

Thank you very much,
Isabella Kim

sandeep-iitr, 2020-05-14:
Hi,
Is there any documentation of using NNtool available now, that can guide the process of deploying a pretrained neural network model (whichever framework is supported: Maybe TensorFlow is fine as well) on the GAP8. The readme file of NNtool doesn't explain in detail the process and steps to follow. It just outlines the process at a very high level, with many code files that are left for the users to go through to figure it out. I can see many parameters and configurations that may require a lot of time to reverse engineer and seems they are hardcoded to the given example.

thanks

Yaooooo, 2020-05-20:
Hi Sandeep,

Sorry for late reply, the doc has been sent to your email.

Thank you for asking,

closed 2020-05-20 by Yaooooo

KrishnaShalini0518-arch, 2021-07-21:
Hi @Yaooooo , It would be really helpful if you can share the beta version docs of NN tools. Thank you

Yaooooo, 2021-07-22:
HI, the doc is already in the NNTool repo since long time. You can also visit our website, there is some blogs and videos show you how to use it.

KrishnaShalini0518-arch, 2021-07-22:
@Yaooooo thank you.