GreenWaves-Technologies/gap_sdk #97: FaceDetection failed

DonQuixote-lab, 2020-01-07:
I have tested the CannyEdgeDetection application successfuly.But the FaceDetection failed.

[image]
There is also a problem when I run the FaceDetection on the gvsoc platform,
[image]
What should I do?

Toxic-Scofield, 2020-01-07:
You should think more，don't just ask when you meet problems.

DonQuixote-lab, 2020-01-07:
> You should think more，don't just ask when you meet problems.

DonQuixote-lab, 2020-01-07:
When openocd uesed by application,no problems appear. But when I use plpbridge,files can not loaded from pc.

Yaooooo, 2020-01-07:
Hi,

If you are using the last version of SDK, we have updated this application to use filesystem api (in ImgIO.c), which provided by PMSIS. These api supports only openOCD.  
In the future, we will remove plpbridge step by step, and maintain only openOCD.

Best,
Yao

DonQuixote-lab, 2020-01-09:
Hi
I am using the last version of SDK.I have executed the WaterMeter and CannyEdgeDetection with no problem.But the BilinearResize failed 
[image]
PLPBRIDGE_FLAGS += -fileIO 2

Yaooooo, 2020-01-09:
Thank you for report, you are right, there's a regression on this example because of the update. 
We will fix it asap.

Yaooooo, 2020-01-10:
I have fixed it and pushed. 
Please pull the last commit and retry.

closed 2020-01-10 by Yaooooo