GreenWaves-Technologies/gap_sdk #98: Question about nntool

GuillaumeDEVIC, 2020-01-09:
Hi,

I would like to execute on the board a CNN architecture as MobileNet or ShuffleNet with Tensorflow. To do this, I ask myself some question:

**1.** Does the operation of NNTOOL require a python script with Keras or a script with only Tensorflow is enough ?
**2.** The example gap_sdk/examples/nntool/mnist/ is dense, what are the steps to follow to make my own example ?
**3.** Does the example gap_sdk/examples/nntool/mnist/ use all the cores in the cluster to make the inference? (it seems to me that the inference is executed only on the master core of the cluster,am I right?)

Regards,

sousoux, 2020-01-09:
1. Does the operation of NNTOOL require a python script with Keras or a script with only Tensorflow is enough ?
NNTOOL takes a tflite graph as input. You would need to change the conversion script to convert a frozen TF graph to TFLITE format. Google has a lot of documentation on this.

2. The example gap_sdk/examples/nntool/mnist/ is dense, what are the steps to follow to make my own example ?
Copy the project. Modify the scipts to convert your graph to TFLITE format. Load it with nntool on the command line and use the various nntool commands (see the script files) to:

1) Run the graph in float and validate the output
2) Quantize the graph using sample data and check the quantization error
3) Once you have something that works save the nntool commands into a script

3. Does the example gap_sdk/examples/nntool/mnist/ use all the cores in the cluster to make the inference? (it seems to me that the inference is executed only on the master core of the cluster,am I right?)

NNTOOL produces an autotiler model.
The AutoTiler produces  API calls that loads your graph, execute it and unload your graph. The graph execution is carried out on the cluster and fully exploits it.

sousoux, 2020-01-09:
We are in the process of producing more significant networks than MNIST that use this development flow. Please contact support@greenwaves-technologies.com for more information.

closed 2020-01-09 by sousoux