GreenWaves-Technologies/gap_sdk #99: NNtool example issue

isakim, 2020-01-15:
Hello,

I'm trying to run the same nntool example. And I'm getting the same error:

> Error: CNN_LinearReLU Kernel: S3_Linear_10x64x4x4, Can't find a matching basic kernel

Do you have a clue of what could be the issue for this error?

Thank you very much,
Isabella

Yaooooo, 2020-01-15:
Hi Isabella,

It seems you still have the problem of mismatching between nntool and
autotiler.
Can you please tell us which version of SDK you are using?
I just tested from the last commit on github, I cannot reproduce your
problem.
Can you pull the last version of SDK, then retry?

Yao

Sent with Gmelius <https://gmelius.com/?ref=mail>

isakim, 2020-01-15:
Hello Yao,

Thank you very much for your answer. I updated the SDK as you said and it's better, but I'm still getting some errors in the end:

> Protocol version mismatch between bridge and runtime (bridge: 4, runtime: -1091576147)
terminate called after throwing an instance of 'std::logic_error'
  what():  Unable to connect to runtime
/home/osboxes/gap_sdk/tools/rules/pulp_rules.mk:152: recipe for target 'run' failed
make: *** [run] Error 134

Do you know what that might be?

Best regards,
Isabella Kim

Pafrak, 2020-01-16:
Hello Isabella,
 I had the same problem. Please re clone a fresh SDK because we changed the structure of the sub modules. Can you report if you then still experience the same problem?

isakim, 2020-01-16:
Hello, 

I already did it and I still get this same message. Until generating and running the autotiler it works perfectly, it's only on the final step (building and running the code on the board) that this runtime message error appears.

But apparently, even this message appearing, it still generates the model.

I tried evaluating some images, running:

`(envpython36) osboxes@osboxes:~/gap_sdk/examples/nntool/mnist$ ./mnist_emul images/189_1.pgm
`

 But I keep getting the same number (7) as recognized number:

```
	 *** NNTOOL Mnist Example ***

Entering main controller
Reading image
File: images/6843_0.pgm open: Ok
Image images/6843_0.pgm:  [W: 28, H: 28] Gray, HeaderSize: 13
P5
28 28
255

Image images/6843_0.pgm, [W: 28, H: 28], Gray, Size: 784 bytes, Loaded sucessfully
Finished reading image
Preprocess 8 bit
Constructor
Call cluster
Running on cluster
Writing tensor step 1 type 2 len 800 addr 0x2244f90
Wrote 800 bytes
Writing tensor step 1 type 3 len 32 addr 0x22452f0
Wrote 32 bytes
Writing tensor step 0 type 0 len 784 addr 0x2234420
Wrote 784 bytes
Writing tensor step 1 type 1 len 4608 addr 0x2234d90
Wrote 4608 bytes
Writing tensor step 2 type 2 len 51200 addr 0x2235f90
Wrote 51200 bytes
Writing tensor step 2 type 3 len 64 addr 0x22452b0
Wrote 64 bytes
Writing tensor step 2 type 1 len 1024 addr 0x2234990
Wrote 1024 bytes
Writing tensor step 3 type 2 len 10240 addr 0x2242790
Wrote 10240 bytes
Writing tensor step 3 type 3 len 20 addr 0x2245310
Wrote 20 bytes
Writing tensor step 3 type 1 len 20 addr 0x2234d90
Wrote 20 bytes
Writing tensor step 4 type 1 len 20 addr 0x2234740
Wrote 20 bytes
Runner completed

Recognized: 7
Ended
```

Do you know what might be the problem? Do you get this same result?

Thank you,
Isabella Kim

Yaooooo, 2020-01-16:
Hi,

This is because of the bridge version, which we will remove it completely.
Now we are moving all of our examples, applications, and SDK to use
openOCD.
I just changed this example to openOCD, which resolved the problem.

Please pull the last commit on master, and retry.
If it doesn't work for you, please let me know.

Yao

Sent with Gmelius <https://gmelius.com/?ref=mail>

closed 2020-03-12 by Pafrak