GreenWaves-Technologies/gap_sdk #101: MNIST example error: Protocol version mismatch between bridge and runtime

danieldennett, 2020-01-23:
Dear GWT,

At the moment I am working with the AI-deck from BitCraze and plan on making a simple program in TF lite for it. For this I want to check if the gap_sdk with its nntool and autotiler is working properly for the AI-deck. I am working from the `3.1.1_dev_001 ` branch and I was able to run the helloworld example from gap-sdk/examples/native/pulpos. Consequently, I was testing the mnist example on the AI-deck and I was able to flash it partially. It stopped after loading the test binary. The error message below did not gave us any leads where to start looking for. Except that `debug address 0x00000001 contents 0x-10215242` could be a clue, but I am not sure how to harness this information and where to start debugging. Thanks in advance.

```
Loading section (base: 0x1c009d8c, size: 0x48)
debug address 0x00000001 contents 0x-10215242
Protocol version mismatch between bridge and runtime (bridge: 4, runtime: -1091576147)
terminate called after throwing an instance of 'std::logic_error'
  what():  Unable to connect to runtime
/home/user/gap_sdk/tools/rules/pulp_rules.mk:162: recipe for target 'run' failed
make: *** [run] Error 134

```

Yaooooo, 2020-01-23:
Hi,

Thank you for the report, you are right, there's a bug between the bridge
and runtime. We will fix it.
In the meantime, you can use openOCD instead of the bridge. (We will remove
this bridge in the future).
To use openOCD:
1, install the openOCD: run "make openOCD " in your sdk folder.
2, add this 2 flags in the makefile:
  export GAP_USE_OPENOCD=1
  io=host

Here's my changes in this makefile, and tested:
gap_sdk_github/examples/nntool/mnist$ git diff
diff --git a/examples/nntool/mnist/Makefile b/examples/nntool/mnist/Makefile
index 9f99363..7c724f7 100644
--- a/examples/nntool/mnist/Makefile
+++ b/examples/nntool/mnist/Makefile
@@ -71,6 +71,9 @@ pulpChip = GAP
 APP = mnist2
 USE_PMSIS_BSP=1

+export GAP_USE_OPENOCD=1
+io=host
+
 APP_SRCS += Mnist.c ImgIO.c $(MODEL_SRCS)

Best,
Yao

Sent with Gmelius <https://gmelius.com/?ref=mail>

danieldennett, 2020-01-24:
Hey, 

Thank you for your response. I have tried to use the openOCD as a bridge. I started with your instructions first, but that gave me the following error message 

```
Generating files (header offset: 0xc400)
  Adding component (name: Mnist_L3_Flash_Const.dat, flashOffset: 0xc434)
112576
Open On-Chip Debugger 0.10.0+dev-gc0bf3edc (2020-01-08-16:56)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex OpenOCD JTAG ARM-USB-OCD-H', serial '*' at bus location '*'

Open On-Chip Debugger 0.10.0+dev-gc0bf3edc (2020-01-08-16:56)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex OpenOCD JTAG ARM-USB-OCD-H', serial '*' at bus location '*'
```
All the while I only have 1 Olimex connected and it is being recognizedby `lsusb` and also it was correctly used by the plpbridge with the helloworld example. So my next step was trying to solve the openocd issue with the tools the gap_sdk is providing this led to these commands

```
cd ~/gap_sdk/tools/gap-openocd
./configure && make && make install
cd ~/gap_sdk/tools/gap-openocd- tools
openocd -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/gap8revb.tcl -f .tcl/jtag_boot.tcl

``` 
this gave me the same error message but now singular

```
Open On-Chip Debugger 0.10.0+dev-gc0bf3edc (2020-01-08-16:56)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex OpenOCD JTAG ARM-USB-OCD-H', serial '*' at bus location '*'
``` 
After this the helloworld example also didn't work anymore. Also the lights on the ai-deck are not blinking anymore when I attach the JTAG connection. While working with the plpbridge this was not the case. This is how far I got and will have another look at it on monday with a colleague of mine.

sivasivarajah, 2020-01-25:
Hi Daniel,

What is the name you see on the Olimex ? There are two types of Olimex : [ARM-USB-OCD](https://www.olimex.com/Products/ARM/JTAG/ARM-USB-OCD/) and [ARM-USB-OCD-H](https://www.olimex.com/Products/ARM/JTAG/ARM-USB-OCD-H/). The configs are different depending on the version.
This might be the reason why the device is not found.

For Mnist, you have two errors message because the first one concerns the picture flashing in memory, the second one is for running the app/example.
For the helloworld example, you flash nothing in memory, that is why the error appears only once.

Siva.

danieldennett, 2020-01-27:
Hey Siva, 

Thank you for your response! The name of my Olimex is Olimex Ltd. ARM-USB-TINY-H JTAG interface. 

Ah that makes a lot of sense! Thanks for the clarification. 

Best regards,

Daniel

sivasivarajah, 2020-01-27:
Hi David,

Glad I could help.
If that is all, I am closing this thread, feel free to open a new one if you have issues.

Siva.

closed 2020-01-27 by sivasivarajah