GreenWaves-Technologies/gap_sdk #109: openocd cannot be compiled on ubuntu 19.10

LorenzoLamberti94, 2020-03-02:
I am trying to compile openocd on ubuntu 19.10
I tried with both gcc versions 9.2.1 and 7.4.0

I compiled the latest gap sdk successfully (v3.1), but when I run the command make openocd I get the following warning/error:

`/usr/include/x86_64-linux-gnu/sys/sysctl.h:21:2: error: #warning "The <sys/sysctl.h> header is deprecated and will be removed." [-Werror=cpp]
#warning "The <sys/sysctl.h> header is deprecated and will be removed."
^~~~~~~`

I tried to ignore sysctl setting  the `HAVE_SYS_SYSCTL_H` flag equal to 0 in the following config files:
tools/gap8-openocd/config.h
tools/gap8-openocd/configure

But no luck yet

Yaooooo, 2020-03-05:
Dear Lorenzo,
It seems the native openocd hasn't support ubuntu 19.10. 
The error is not on GAP8 side, but the openocd itself.
I will suggest you firstly try to compile the native openocd: https://github.com/ntfreak/openocd
To check whether it works for you.
If yes, please tell us, maybe we need to update our version.
If no, please report this issue to openOCD. We can update ours once they fix your issues.

Best,
Yao

LorenzoLamberti94, 2020-03-12:
Thank you Yao for your kind answer!
I compiled openocd and it goes through without issues. I think it should work on ubuntu 19
let me know if I can make more tests
Lorenzo

Yaooooo, 2020-03-12:
Hi Lorenzo,
We are updating the openocd repo soon (rebased on the official one).
This will help us to resolve the problem with Ubuntu19.
We will keep you informed once it's done.
Yao

Pafrak, 2020-06-08:
closing due to inactivity

closed 2020-06-08 by Pafrak