GreenWaves-Technologies/gap_sdk #110: uart_read hangs indefinately after hyperflash device is opened

benmarlin, 2020-03-02:
Hi. We are having a problem trying to read from the uart after opening the hyperflash device. Below is a minimal modification to test_hyperflash.c that shows the issue. If you send a single byte to the Gap8 (for example, using screen /dev/ttyUSB1 9600) the code will pass the first uart_read() test after entering test_hyper_flash(). It will then hang at the second uart_read() test at the end of test_hyper_flash() no matter how many bytes are sent to the Gap8. Although not shown in this example, uart_write calls still work. We are using the version 2 Gapuino board. We have several version 2 boards and all have the same problem. Is there some way to resolve this? 

Thanks.

/* PMSIS includes */
#include "pmsis.h"
#include "bsp/flash.h"
#include "bsp/flash/hyperflash.h"
#include "Gap8.h"

/* Variables used. */
#define BUFFER_SIZE      ( 2048 )

//RT_L2_DATA uint8_t rx_buffer[1];
static uint8_t *rx_buffer;
static uint8_t *buff, *rcv_buff;
static uint32_t hyper_buff;
static struct pi_device flash;
static struct pi_hyperflash_conf conf;

struct pi_device uart;
struct pi_uart_conf uart_conf;
#ifdef ASYNC
static volatile uint8_t done = 0;

/* Callback for end of RX transfer. */
void __end_of_rx(void *arg)
{
    printf("Read async done.\n");
    done = 1;
}

/* Callback for end of TX transfer. */
void __end_of_tx(void *arg)
{
    pi_task_t *cb = (pi_task_t *) arg;
    printf("Write async done.\n");
    pi_flash_read_async(&flash, hyper_buff, rcv_buff, (uint32_t) BUFFER_SIZE, cb);
}

/* Callback for end of erase. */
void __end_of_erase(void *arg)
{
    pi_task_t *cb = (pi_task_t *) arg;
    printf("Erase async done.\n");
    pi_flash_program_async(&flash, hyper_buff, buff, (uint32_t) BUFFER_SIZE, cb);
}
#endif  /* ASYNC */
void uart_read() {
    pi_uart_conf_init(&uart_conf);
    uart_conf.baudrate_bps = 9600;
    uart_conf.uart_id = 0;
    uart_conf.enable_tx = 1;
    uart_conf.enable_rx = 1;

    pi_open_from_conf(&uart, &uart_conf);
    if (pi_uart_open(&uart)) {
      printf("uart open failed\n");
    }

    rx_buffer = (uint8_t *) pmsis_l2_malloc((uint32_t) 1);
    pi_task_t wait_task = {0};
    pi_task_block(&wait_task);
    printf("tasK_block passed\n");
    pi_uart_read_async(&uart, rx_buffer, 1, &wait_task);
    printf("pi_uart_async_read called, waiting\n");
    pi_task_wait_on(&wait_task);
    printf("waiting over, got 1 byte with value %d\n",  rx_buffer[0]);
}

void test_hyper_flash(void)
{
    printf("Entering main controller\n\n");
    uart_read();

    uint32_t errors = 0;
    struct pi_flash_info flash_info;

    buff = (uint8_t *) pmsis_l2_malloc((uint32_t) BUFFER_SIZE);
    if (buff == NULL)
    {
        printf("buff alloc failed !\n");
        pmsis_exit(-1);
    }

    rcv_buff = (uint8_t *) pmsis_l2_malloc((uint32_t) BUFFER_SIZE);
    if (rcv_buff == NULL)
    {
        printf("rcv_buff alloc failed !\n");
        pmsis_exit(-2);
    }

    for (uint32_t i=0; i<(uint32_t) BUFFER_SIZE; i++)
    {
        buff[i] = i & 0xFF;
        rcv_buff[i] = 0;
    }

    /* Init & open flash. */
    pi_hyperflash_conf_init(&conf);
    pi_open_from_conf(&flash, &conf);
    if (pi_flash_open(&flash))
    {
        printf("Error flash open !\n");
        pmsis_exit(-3);
    }

    /* Retrieve info on flash. */
    pi_flash_ioctl(&flash, PI_FLASH_IOCTL_INFO, (void *) &flash_info);
    hyper_buff = ((flash_info.flash_start + flash_info.sector_size - 1) &
                  ~(flash_info.sector_size - 1)) + 128;
    /* Erase a sector in flash, write a buffer then read back from flash. */
    #ifdef ASYNC
    pi_task_t cb_erase, cb_tx, cb_rx;
    pi_task_callback(&cb_erase, __end_of_erase, &cb_tx);
    pi_task_callback(&cb_tx, __end_of_tx, &cb_rx);
    pi_task_callback(&cb_rx, __end_of_rx, NULL);
    pi_flash_erase_async(&flash, hyper_buff, (uint32_t) BUFFER_SIZE, &cb_erase);
    while (!done)
    {
        pi_yield();
    }
    #else
    pi_flash_erase(&flash, hyper_buff, (uint32_t) BUFFER_SIZE);
    printf("Erase sync done.\n");
    pi_flash_program(&flash, hyper_buff, buff, (uint32_t) BUFFER_SIZE);
    printf("Write sync done.\n");
    pi_flash_read(&flash, hyper_buff, rcv_buff, (uint32_t) BUFFER_SIZE);
    printf("Read sync done.\n");
    #endif  /* ASYNC */

    /* Verification. */
    for (uint32_t i=0; i<(uint32_t) BUFFER_SIZE; i++)
    {
        if (buff[i] != rcv_buff[i])
        {
            errors++;
            printf("%2x-%2x ", buff[i], rcv_buff[i]);
        }
    }

    pmsis_l2_malloc_free(buff, (uint32_t) BUFFER_SIZE);
    pmsis_l2_malloc_free(rcv_buff, (uint32_t) BUFFER_SIZE);
    pi_flash_close(&flash);

    printf("\nHyperflash transfer done with %ld error(s) !\n", errors);
    printf("\nTest %s with %ld error(s) !\n", (errors) ? "failed" : "success", errors);

    uart_read();
   
    pmsis_exit(errors);
}

/* Program Entry. */
int main(void)
{
    printf("\n\n\t *** PMSIS Hyperflash Test ***\n\n");
    return pmsis_kickoff((void *) test_hyper_flash);
}

sivasivarajah, 2020-03-03:
Hi Ben Marlin,

On Gap8 chip, there is pad mux bug between UART_RX pad and HYPER_DQ[6] pad. Both are muxed to the same pad, while they should not, this is a HW bug. Sorry for this inconvenience.

When using UART_RX, the pad in HW is set to func0(default) while for Hyperbus, the pad should be set to func3 as alternate function. These settings are expected as normal behaviour, due to the HW bug on the pad mux, the pads must be set depending on use cases. This is the only workaround.
**PI_PAD_46_B7_SPIM0_SCK** should be set to **PI_PAD_FUNC3** when using Hyperbus(flash or ram).
**PI_PAD_46_B7_SPIM0_SCK** should be set to **PI_PAD_FUNC0** when using UART_RX(the bug affects only the RX part).

The best solution is to set the pad **PI_PAD_46_B7_SPIM0_SCK** to **PI_PAD_FUNC0** before using UART_RX and when it is done, switch back to **PI_PAD_FUNC3** for Hyperbus.

In your case, it would be :
`pi_pad_set_function(PI_PAD_46_B7_SPIM0_SCK, PI_PAD_FUNC0); // Switch pad to UART_RX.
uart_read();
pi_pad_set_function(PI_PAD_46_B7_SPIM0_SCK, PI_PAD_FUNC3); // Switch pad to HYPER_DQ[6].`

Best.

benmarlin, 2020-03-08:
Thank you. That fixed the problem.

closed 2020-03-08 by benmarlin