GreenWaves-Technologies/gap_sdk #112: gap8-openocd-tools issue

benmarlin, 2020-03-08:
Hi. I am attempting to enable hyperflash boot on the version 2 Gapuino board following the instructions here: https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/gap8-openocd-tools/README

After compiling openocd in the gap8-openocd directory, I attempted to run the first openocd example from the gap8-openocd-tools directory:

openocd -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/gap8revb.tcl -f ./tcl_scripts/jtag_boot.tcl 

First, the path to the local tcl directory appears to be incorrect in the README. The command looks like it should be as shown below based on the current directory names:

openocd -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/gap8revb.tcl -f ./tcl/jtag_boot.tcl 

With this fix to the command, I get the error below:

Open On-Chip Debugger 0.10.0+dev-00003-g5c5f57b-dirty (2020-02-29-14:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex OpenOCD JTAG ARM-USB-OCD-H', serial '*' at bus location '*'

Next, first part of the process to fuse for hyperflash boot says to run openocd with the arguments: 

interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/gap8revb.tcl -f ./tcl_scripts/jtag_boot.tcl -f ./tcl_scripts/fuser.tcl -f ./tcl_scripts/fuser_flash.tcl

This also looks incorrect. There is a missing leading -f and the tcl directory is again incorrect. It looks like the full command should be:

openocd -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/gap8revb.tcl -f ./tcl/jtag_boot.tcl -f ./tcl/fuser.tcl -f ./tcl/fuser_flash.tcl

I get the same error running this as for the first command:

Open On-Chip Debugger 0.10.0+dev-00003-g5c5f57b-dirty (2020-02-29-14:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex OpenOCD JTAG ARM-USB-OCD-H', serial '*' at bus location '*'

The gapunio is connected and I am able to compile, load and run code examples using "make all run". My lsusb output is below. 

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C Dual USB-UART/FIFO IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Please let me know what the fix is. 

Thank you!

jinhuang01, 2020-03-09:
I think the `olimex-arm-usb-ocd-h.cfg` should be replaced by `sipeed-rv-debugger.cfg`.
Because in the file `sipeed-rv-debugger.cfg`,  there is 
`ftdi_vid_pid 0x0403 0x6010`

And GAP8 ftdi just have that two ids(lsusb -v)

idVendor           0x0403 Future Technology Devices International, Ltd
idProduct          0x6010 FT2232C Dual USB-UART/FIFO IC

But I still could not find the fuser_flash.tcl.

Pafrak, 2020-03-09:
Hi benmarlin,
 here are there script you are looking for:

JTAG_TCL=$GAP_SDK_HOME/tools/gap8-openocd-tools/tcl/jtag_boot.tcl
FLASH_TCL=$GAP_SDK_HOME/tools/gap8-openocd-tools/tcl/flash_image.tcl
FUSER_TCL=$GAP_SDK_HOME/tools/gap8-openocd-tools/tcl/fuser.tcl
FUSER_FLASH_TCL=$GAP_SDK_HOME/tools/gap8-openocd-tools/tcl/fuser_flash.tcl

benmarlin, 2020-03-09:
Hi. The file fuser_flash.tcl is not in the master branch of the sdk in the gap8-openocd-tools/tcl/ folder. This is the master branch content of the gap8-openocd-tools/tcl directory according to [https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/gap8-openocd-tools/tcl] :

jtag_boot.tcl 	[3.1 release] 	
flash_image.tcl    [3.2 dev (#104)] 	
fuser.tcl 	[3.1 release ]	
hyper_boot.tcl 	[3.1 release] 

Also, can you comment on whether olimex-arm-usb-ocd-h.cfg is the correct .cfg file for the gap8? It appears not the be as the vid and pid don't match the gapuino v2 ftdi chip as reported by lsusb (FT2232C, 0403:6010).

Thank you again for the help.

Yaooooo, 2020-03-09:
Hi Ben,

If I understand correctly, you are using a GAPuino v2 board.
In this case, the olimex config file is not the right script file to use.
As on the board, there is already a ftdi chip integrated. You don not need
further debugger, like olimex, to program and use GAP8.
The right config file for GAPuino is called:
gapuino_ftdi.cfg
Normally, you can find it in gap8_openocd repo:
https://github.com/GreenWaves-Technologies/gap8_openocd/blob/gap8/tcl/interface/ftdi/gapuino_ftdi.cfg

Also in the sdk:
gap_sdk/tools/gap8_openocd/tcl/interface/ftdi/gapuino_ftdi.cfg

Once you have configured your environment for gap_sdk by using source
sourceme.sh
The env parameter: OPENOCD_CABLE
will include this information.

Hope this will help you.

Best
Yao

Yao Zhang
Greenwaves Technologies

closed 2020-03-12 by Pafrak