GreenWaves-Technologies/gap_sdk #126: Failed with example/pmsis/helloworld

FrontMage, 2020-05-20:
Hi, I'm a newbie to gap8, I got a gapuino and I was just trying to get it working, but after I followed official sdk instruction, I ran into this issue, looks like target/gap8revc.tcl is missing, is there any extra steps to take before running examples?

Thanks for the help~
```bash
Launching execution with command:
openocd -c 'gdb_port disabled; telnet_port disabled; tcl_port disabled' -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8revc.tcl -f tcl/jtag_boot.tcl -c 'gap8_jtag_load_binary_and_start "/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8_V3/GCC_RISCV/test" elf'
Open On-Chip Debugger 0.10.0+dev-00015-gf04b0ae6 (2020-05-20-10:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:26: Error: Can't find target/gap8revc.tcl
in procedure 'script'
at file "embedded:startup.tcl", line 26
Traceback (most recent call last):
  File "/gap_sdk/tools/gapy/gapy", line 162, in <module>
    main()
  File "/gap_sdk/tools/gapy/gapy", line 152, in main
    operationFunc(args, config)
  File "/gap_sdk/tools/gapy/run.py", line 146, in operationFunc
    raise RuntimeError('Runner has failed with value: %d' % status)
RuntimeError: Runner has failed with value: -1
/gap_sdk/tools/rules/pulp_rules.mk:209: recipe for target 'run' failed
make: *** [run] Error 1
```

sivasivarajah, 2020-05-20:
Hi @FrontMage,

Can you tell which commit you are working on? (command `git rev-parse HEAD`)

And also what board are you using? Because depending on the version you have, you can source the board's source script(cf. [Configure the SDK](https://github.com/GreenWaves-Technologies/gap_sdk#configure-the-sdk)).

Yaooooo, 2020-05-20:
Hi @FrontMage 

I think it's due to a wrong config in your environment. The gap8revc is not available yet.
You probably source the wrong config file in gap_sdk/configs. (I think you have chosen gapuino_v3.sh)
In fact, you should use: configs/gapuino_v2.sh 

Please take a fresh terminal, than:

1. source configs/gapuino_v2.sh 
2. make all

And retry your test with your board.
Enjoy!

Yao

FrontMage, 2020-05-21:
@sivasivarajah I'm on commit f04b0ae6cf2961690e33f8baa0007182221a5943
@Yaooooo Thank you, it seems like I sourced the wrong config then make the wrong version of sdk.

Now I've got another problem, I'm running Manjanro on my machine, but the sdk requires ubuntu, so I used a ubuntu:18.04 docker image to do this, the sdk build is ok now, but it cant find device inside docker container, since I've already mounted /dev into the container, it should worked.

So I wonder if there is a way to check that the device is indeed connected, thanks again for the help!

FrontMage, 2020-05-24:
Sorry for the inconvenience, my boss told me it was the newest gapuino, it turns out that it was a v1 board, problem solved, thx for the help!

closed 2020-05-24 by FrontMage