GreenWaves-Technologies/gap_sdk #127: How to use custom network with nntools [Question]

FrontMage, 2020-05-25:
Hi, I've been  trying to use nntools to make my own network, here is what I did:

1. change the code in `examples/nntools/mnist/model/train.py` to train another network
2. change all the 28x28 to 500x500 in `examples/nntool/mnist/common.mk` and `examples/nntool/mnist/model/run.py`

Then I did `make -f emul.mk all`, it fails at `nntool -g -M BUILD_MODEL_8BIT_EMUL -m mnistModel.c -T BUILD_MODEL_8BIT_EMUL/tensors  BUILD_MODEL_8BIT_EMUL/mnist.json`, since I did not change the name `mnist`.

Ouput as followed
```bash
root@10bd4f9e53f4:/gap_sdk/examples/nntool/mnist# nntool -g -M BUILD_MODEL_8BIT_EMUL -m mnistModel.c -T BUILD_MODEL_8BIT_EMUL/tensors  BUILD_MODEL_8BIT_EMUL/mnist.json
<graph.nngraph.NNGraph object at 0x7fb95a96b2e8>
code_generator - Quantization record not found for node input_1
Traceback (most recent call last):
  File "/gap_sdk/tools/nntool/nntool", line 94, in <module>
    main()
  File "/gap_sdk/tools/nntool/nntool", line 78, in main
    mod.generate_code(args)
  File "/gap_sdk/tools/nntool/interpreter/generator.py", line 69, in generate_code
    model = code_template(G, code_generator=code_gen)
  File "/gap_sdk/tools/nntool/generation/default_template.py", line 138, in default_template
    return execute_template(generator_template_v3, G, naming_convension, code_generator)
  File "/gap_sdk/tools/nntool/generation/default_template.py", line 132, in execute_template
    return template_function(G, code_generator)
  File "/gap_sdk/tools/nntool/generation/default_template.py", line 88, in generator_template_v3
    ${gen.kernel_generator(indent=1)}
  File "/gap_sdk/tools/nntool/generation/code_generator.py", line 463, in kernel_generator
    raise err
  File "/gap_sdk/tools/nntool/generation/code_generator.py", line 460, in kernel_generator
    qrec = self.G.quantization[NodeId(node)]
KeyError: <utils.node_id.NodeId object at 0x7fb95a962e80>
```

Seems it stuck at `It then generates an nntool state file by running an nntool script with commands to adjust the tensor and activation order, fuse certain operations together and automatically quantify the graph` step.

And the reason being the input layer of my network is not named `input_1`, I searched the repo but can't find where to define this input layer name.

Is there a documentation for nntools? cause I can't find it in the official documentation.

Thanks for the help again~
Cheers!

marco-fariselli, 2020-05-28:
Hi @FrontMage, 
for the first issue can you provide us your model so that we can replicate the problem and try to solve it? Even with fake weights is ok. It seems that the quantization for the input layer is not set or has another name.
For the second issue, the error you see appears when you try to feed the network with images of a different size with respect to the network input layer dimensions. Whenever you feed nntool with some images, like when you use the commands aquant or dump, you can specify that you want to resize automatically your images set to the desired size with -W and -H flags.

You can find more information and some documentation about nntool in the README of the nntool repository.

Cheers, Marco

FrontMage, 2020-05-29:
@marco-fariselli Hi, I'm using the network `squeezenet` from this GitHub repo `https://github.com/rcmalli/keras-squeezenet` typically using this file `https://github.com/rcmalli/keras-squeezenet/blob/master/keras_squeezenet/squeezenet.py`.

```python
from squeezenet import SqueezeNet
from keras.preprocessing import image
from keras.models import Model
from keras.layers import Dense, GlobalAveragePooling2D, Input
from keras import backend as K
from keras.optimizers import Adam

out_shape_x = 500
out_shape_y = 500
channel = 3

input_tensor = Input(shape=(out_shape_x, out_shape_y, channel))
base_model = SqueezeNet(input_tensor=input_tensor, weights='imagenet', include_top=False)

x = base_model.output
x = GlobalAveragePooling2D()(x)
x = Dense(1024, activation='relu')(x)
predictions = Dense(1, activation='sigmoid')(x)

model = Model(inputs=input_tensor, outputs=predictions)

for layer in base_model.layers:
    layer.trainable = False

model.compile(optimizer=Adam(lr=1e-4), metrics=['accuracy'], loss='binary_crossentropy')
model.summary()
```

The weights can be found here [https://drive.google.com/file/d/1tC2fq150Iql6rY-bMm1EguraOMhaQCnh/view?usp=sharing](URL)

The weights are saved by using Keras `ModelCheckpoint`.

---

As for the image input size part, I altered the `common.mk` file, 

```makefile
MODEL_PREFIX=mnist
AT_INPUT_WIDTH=500
AT_INPUT_HEIGHT=500
AT_INPUT_COLORS=3
```
But 500x500x3 is not exactly 2500.

---

Thanks again for the help!

FrontMage, 2020-06-01:
@marco-fariselli 
---
Hi,
I managed to locate the image size error comes from `nntool_script_emul8` line 6, which says `aquant -f 8 images/*`.

Apparently `aquant` reads images as flatted tensor instead of matrix.

According to the documentation, I added `-H 500 -W 500` to that line but found no luck.

As a matter of fact, it's the image format that broke this, from the `mnist` example, it uses `.pgm` which should be `.ppm` if I changed the image from grayscale to colored.

But now I get stuck at `BUILD_MODEL_8BIT_EMUL/GenTile -o BUILD_MODEL_8BIT_EMUL -c BUILD_MODEL_8BIT_EMUL -f BUILD_MODEL_8BIT_EMUL --L1 52000 --L2 307200 --L3 8388608` which gives `Error: AddCallToNode: Node S7_Relu argument OutputStep7 is not part of graph node arguments`.

This seems to be from an auto-generated file `BUILD_MODEL_8BIT_EMUL/mnistModel.c`, no sure what went wrong here.

Thanks for the help!

sousoux, 2020-06-02:
Sigmoid is available in the generators. It is fused with other layers and also available standalone (along with swish, relu, relun and leaky). Please try with the new SDK released yesterday.

Yaooooo, 2020-06-02:
Hi @FrontMage

As mentioned in previous reply, we released a new version of SDK (https://github.com/GreenWaves-Technologies/gap_sdk/releases/tag/release-v3.5), which is a major release for NNTool. 
Please feel free to try it, any feedback is more than welcome.

BTW, we would like to talk with our GAP8 users to understand how to better support you, and your project. Are you available to have a talk with us? 

Thank you again for your questions.

Regards,
Yao

FrontMage, 2020-06-03:
> Sigmoid is available in the generators. It is fused with other layers and also available standalone (along with swish, relu, relun and leaky). Please try with the new SDK released yesterday.

@Yaooooo 

The new release posts these errors when I try to `make clean all`

```bash
find: 'include': No such file or directory
find: 'include': No such file or directory
g++ -std=gnu++11 -MMD -MP -O3 -g -I/gap_sdk-release-v3.5/install/workstation/include -fPIC -DUSE_DPI -Iext/sv/include -Iext/nosv -o /gap_sdk-release-v3.5/build/gvsoc/dpi/src/dpi.o -c src/dpi.cpp
g++ -o /gap_sdk-release-v3.5/build/gvsoc/libgvsocdpi.so /gap_sdk-release-v3.5/build/gvsoc/dpi/src/dpi.o -L/gap_sdk-release-v3.5/install/workstation/lib -fPIC -shared -O3 -g -ljson -lpulpvp-sv
install -D /gap_sdk-release-v3.5/build/gvsoc/libgvsocdpi.so /gap_sdk-release-v3.5/install/workstation/lib/libgvsocdpi.so
make[2]: Leaving directory '/gap_sdk-release-v3.5/gvsoc/gvsoc/dpi-wrapper'
make -C models props ARCHI_DIR=
make[2]: Entering directory '/gap_sdk-release-v3.5/gvsoc/gvsoc/models'
plpinfo mkgen --makefile=/gap_sdk-release-v3.5/build/gvsoc/props.mk  --property=udma/version  --property=timer/version  --property=stdout/version  --property=soc_eu/version  --property=fc_itc/version  --property=icache_ctrl/version  --property=xne/version  --property=fll/version  --property=event_unit/version  --property=cluster_ctrl/version  --property=pulp_chip  --property=cluster/version  --property=dma/version  --property=udma/spim/version  --property=udma/cpi/version  --property=padframe/version  --property=pulp_chip_family  --property=pmu/version  --property=udma/hyper/version  --property=efuse/version  --property=udma/i2c/version  --property=rtc/version  --property=gpio/version  --property=fc/iss_class  --property=pe/iss_class  --property=udma/i2s/version  --property=udma/mram/version  --property=hwce/version  --property=udma/tcdm/version  --property=hwacc/base  --property=udma/uart/version
Traceback (most recent call last):
  File "/gap_sdk-release-v3.5/install/workstation/bin/plpinfo", line 140, in <module>
    configs = plpconf.get_configs_from_env()
  File "/gap_sdk-release-v3.5/install/workstation/python/pulp_config.py", line 189, in get_configs_from_env
    configs = get_configs(config_str, interpret=interpret)
  File "/gap_sdk-release-v3.5/install/workstation/python/pulp_config.py", line 170, in get_configs
    config_tree = get_config(file, full_config, ini_configs=ini_configs, ini_configs_dict=ini_configs_dict, interpret=interpret)
  File "/gap_sdk-release-v3.5/install/workstation/python/pulp_config.py", line 116, in get_config
    config = js.import_config_from_file(file, find=True, interpret=interpret)
  File "/gap_sdk-release-v3.5/install/workstation/python/json_tools.py", line 125, in import_config_from_file
    config_dict = get_config_file(file_path, interpret=interpret, find=find, path=path, paths=paths)
  File "/gap_sdk-release-v3.5/install/workstation/python/json_tools.py", line 120, in get_config_file
    config_dict = json.load(fd, object_pairs_hook=OrderedDict)
  File "/usr/lib/python3.6/json/__init__.py", line 299, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.6/json/__init__.py", line 367, in loads
    return cls(**kw).decode(s)
  File "/usr/lib/python3.6/json/decoder.py", line 339, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.6/json/decoder.py", line 357, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
Makefile:33: recipe for target 'props' failed
make[2]: *** [props] Error 1
make[2]: Leaving directory '/gap_sdk-release-v3.5/gvsoc/gvsoc/models'
Makefile:24: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/gap_sdk-release-v3.5/gvsoc/gvsoc'
Makefile:107: recipe for target 'gvsoc' failed
make: *** [gvsoc] Error 2
root@3e45aea871ab:/gap_sdk-release-v3.5#
```

My project is pretty simple, I try to make `squeezenet` work on the board.
I read [https://github.com/GreenWaves-Technologies/AliOS-Things](url) and I thought I could get it working.

FrontMage, 2020-06-04:
@sousoux @marco-fariselli @Yaooooo 
I used the latest release, and it works great now on `sequential model` with `emulator`.
How ever, `squeezenet` still fails, it complaints about one layer can only connect to one other layer, where `squeezenet` has this `fire_module` which was made of 2 layers side by side.

You can see as follow:
```python
def fire_module(x, fire_id, squeeze=16, expand=64):
    s_id = 'fire' + str(fire_id) + '/'

    if K.image_data_format() == 'channels_first':
        channel_axis = 1
    else:
        channel_axis = 3

    x = Convolution2D(squeeze, (1, 1), padding='valid', name=s_id + sq1x1)(x)
    x = Activation('relu', name=s_id + relu + sq1x1)(x)

    left = Convolution2D(expand, (1, 1), padding='valid', name=s_id + exp1x1)(x)
    left = Activation('relu', name=s_id + relu + exp1x1)(left)

    right = Convolution2D(expand, (3, 3), padding='same', name=s_id + exp3x3)(x)
    right = Activation('relu', name=s_id + relu + exp3x3)(right)

    x = concatenate([left, right], axis=channel_axis, name=s_id + 'concat')
    return x
```

On the board, however, I managed to get only `helloworld` running, cause even the official example also stuck at `Graph constructor exited with an error`.

I checked `mnistCNN_Construct()`'s return value, it's `1`, according to `BUILD_MODEL_SQ8BIT/mnistKernels.c`, it returns `1` when `AT_HYPERFLASH_FS_OPEN(&HyperFlash, &HyperFlashConf, "desktopclassify_L3_Flash_Const.dat", &Error);` fails.

Thanks again for the help!

sousoux, 2020-12-06:
This has been resolved some time ago. Squeezenet has been tested with both TFLITE and ONNX input.

closed 2020-12-06 by sousoux