GreenWaves-Technologies/gap_sdk #130: rt_bridge_connect not working on release 3.4

ltancerel, 2020-05-27:
Hello,

some application code does not work on release 3.4 while working on release 3.3.

There is an infinite loop in 
``rt_bridge_connect(1, NULL);
we call at the beginning of our application to store images.
From printf debugging, there is a never ending condition in 
``  __rt_wait_event_check(event, call_event);

We cannot find working examples of rt_bridge_connect on that release.

Thank you for your help

Yaooooo, 2020-05-27:
Hi,

Please tell us which application you are using? 
In fact, the plpbridge is no more used in our SDK, since we support OpenOCD semihosting now. 
Therefore, you can use semihosting to Read/Write to your host. 
For further information, you can take a look of this [camera example](https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/pmsis/test_periph/test_camera_stream)

This example is using semihosting to write an image captured from camera to your host of PC.

You can use "hostfs" as in this example instead of using "rt_bridge_xxxxx".

ltancerel, 2020-05-29:
Hello, we use our own application, but it was based on plpbridge from a camera example as well.
Thank you for your reply.

I try the test_camera_stream example.
Is this test supposed to work with gapuino ?
I indeed get a output.raw, but I am not able to read the camera image.

I will try to get the fs part from the test and check with my application.

Yaooooo, 2020-06-02:
Hi, 

Yes, it's correct, just the image hasn't been written into ppm format. 
You can also check this example: https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/examples/pmsis/test_periph/test_camera_io

It's the same principle, but writting the image into a pgm file which can be read on PC.

Yao

closed 2020-06-02 by Yaooooo