GreenWaves-Technologies/gap_sdk #131: unable to run hello world example! plus another MAJOR request

tp19021, 2020-05-28:
I tried running 'make clean all run' for testing Hello World example but I am getting this error regarding the makefile where it can't recognize the pmsis_rules.mk file.

Makefile:9: /pmsis_rules.mk: No such file or directory
make: *** No rule to make target '/pmsis_rules.mk'. Stop

How do I fix this? 

Also do you know where I can find any multilayer NN examples and how to use nntool to map them to the GAP8? Also to view performance counters for such benchmarks?

tp19021, 2020-05-28:
To add to the hello world issue, the 1st time I run, this is the error:
[image]

Pafrak, 2020-05-28:
Hi tp19021,
this should solve your problem:
`pip3 install pyelftools`

To read performance counter you can look at this example:
[(https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/pmsis/test_periph/perf)
]

For NNtool you can take a look at the readme: [https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/nntool].

closed 2020-05-28 by Pafrak