GreenWaves-Technologies/gap_sdk #132: Error when building mnist using nn tools

tp19021, 2020-05-29:
in the gap_sdk/examples/nntool/examples/mnist, im trying to compile and run it onto the gap8 but when i run 'make clean all run' im getting an error where it says 'no module named argcomplete' as shown below
[image]
I tried to install it using pip3 but it says its already installed but I still get the same error afterwards. how do i fix it?

Pafrak, 2020-05-29:
try pip

tp19021, 2020-05-29:
so ive tried exiting su and ran make again but now it says it does not recognize tensorflow as shown below:
[image]

This is bizarre since ive installed anaconda and the result of make nntool shows that it successfully installed tensorflow 1.14.0 as shown below:
[image]

I tried pip3 install tensorflow anyway but had this error about msgpack and wrapt:
[image]

I then tried checking tensorflow by doing pip freeze but it and this is what I got:
[image]

Im not sure whats the next step to take exactly to allow the make to work

sousoux, 2020-05-29:
I think you have probably not run a make sdk in the gap_sdk directory. Make all does not build nntool / autotiler.

sousoux, 2020-05-29:
Also, check that the anaconda python is actually active. It should be in your bash.rc. which python can help to see what is actually configured.

tp19021, 2020-05-29:
I did run make nntool and make autotiler separately. I can do make sdk still if thats worth doing

Regarding the anaconda, do I need to activate the environemnt cuz i think i did activate the installation when I was following the installation procedure

sousoux, 2020-05-29:
Honestly. I'd delete the whole thing and start again. You shouldn't be running the build as root. When you execute sudo you start a new shell so all the exports done in the sourceme are lost. Please delete the directory and follow the readme to the letter.

closed 2020-05-29 by sousoux

tp19021, 2020-05-29:
So I tried doing everyting in su mode, that is install and activate anaconda and then ran make sdk and got this error to do with not finding the version for tensorflow 1.14.0 (the same error when running pip install -r tools nntool/requirements.txt as per gap8 manual):
[image]