GreenWaves-Technologies/gap_sdk #133: no support for mbed anymore ?

lyris85, 2020-05-31:
Hello,

mbed_rules.mk has been removed from the project in master branch. Any reason for that ?

lyris

Yaooooo, 2020-06-02:
Hi Lyris,

Since we don't have enough energy to maintain mbed os well, and connect it with all our new tools:
- NNTools: a neural network tool set for GAP8
- New version of autotiler: GAP8 code generator

We decide to take it away from our SDK. It is still on Github; https://github.com/GreenWaves-Technologies/mbed-os

If it's a problem for you, we are willing to have a discussion with you, to understand your need and requirement.

You can drop me an email if it's ok for you.

Regards,
Yao

lyris85, 2020-06-03:
Hi @Yaooooo ,

it is fine I don't need mbed. However I want to use gap8 for ml application which requires opencv and a core in c++. I managed to tweek the makefiles. Some hugly warnings remain but it works. That would be cool if you could provide a clean c++ support.

Really good job with gap8, love the simulator.

lyris

Yaooooo, 2020-06-03:
Hi @lyris85,

Thank you for your feedback. We will considerate about your suggestion seriously. 

To better use GAP8 for ML/DSP applications, [NNTool ](https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/nntool) and [autotiler](https://greenwaves-technologies.com/manuals/BUILD/AUTOTILER/html/index.html) are necessary. 

These tools support only pure C for now, but we will seriously consider this.

Thanks again.

Yao

closed 2020-06-03 by Yaooooo

gemenerik, 2021-01-19:
Hi @lyris85,

Can you share your approach to tweaking the Makefiles? Just have it accept ++ files? I am mostly working with CNNs>`NNTool` myself but want to compare with some OpenCV functions.

Kind regards,
Rik

lyris85, 2021-01-19:
@gemenerik I don't have the code anymore. I can only remember that the gap risv toolchain has support for c++. Maybe look at mbed_rules.mk rules to see how they did with c++ files. Good luck

gemenerik, 2021-01-19:
@lyris85 Thanks!