GreenWaves-Technologies/gap_sdk #136: How to parallelize cifar10 and mnist?

tp19021, 2020-06-01:
I want to be able to parallelize the execution of the CNN layers of the cifar10 and mnist examples across 2, 4 and 8 RISC-V cores but I'm not too sure how it can be done? Are there generic functions from a library that we can use the parallelize cifar10 and mnist? Do we use pulp-os? If you can provide or point out a detailed doc or examples on whats the simplest way to do it would be very helpful.

sousoux, 2020-06-02:
All the examples that use the AutoTiler CNN generators directly or indirectly via nntool parallelize execution on all 8 cores of the GAP cluster. Please refer to these examples.

closed 2020-06-02 by sousoux

tp19021, 2020-06-05:
From the autotiler documentation, it says this directory is where the examples that use Autotiler generators for CNN are provided:
$GAP_SDK_HOME/examples/pulp-examples/autotiler_examples/

However, the link doesn't work anymore. Could you point me out where the examples are?