GreenWaves-Technologies/gap_sdk #138: Not able to JTAG the board after openocd-fuser-hyperflash

ltancerel, 2020-06-05:
I am not able to JTAG through the board after the `openocd-fuser-hyperflash`.
My board is no more usable...

Here is the sequence done. I am using openocd JTAG for a while on release 3.4. This is a gapuino v2.
I checked I was able to execute my code and get printf on UART :
```
make clean all run io=uart
```
It worked well and I got the printf in teh /dev/ttyUSB console.

After this I executed the `openocd-fuser-hyperflash`, but it did not work because of missing variables: 
OPENOCD_CABLE, OPENOCD_CHIP_TARGET
These variables are not initialized using `source sourceme.sh` option 2.
After some search, I sourced the following:
```
source configs/gapuino_v2.sh
openocd-fuser-hyperflash
```
and I got following result :

```
BE CAREFUL: This will permanently configure the board to boot from hyperflash and cannot be undo. Do you confirm (y/N) ?
y
Configuring board for hyperflash, please stop it when "fuse_once is done" is displayed.

Open On-Chip Debugger 0.10.0+dev-00015-gf04b0ae6 (2020-05-29-09:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : Listening on port 3333 for gdb connections
fuse_spiflash_boot
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
RESET: jtag boot mode=3
Loading binary through JTAG
Warn : Burst read timed out
/home/ludovic/repo/gap_sdk/tools/gap8-openocd-tools/gap_bins/gap_fuser@gapoc_a.elf
Wait for gap init
fuse boot from flash bit
signal gap that setup is done
Wait for fuser to finish
fuse_once is done
fuse boot from hyper bit
signal gap that setup is done
Wait for fuser to finish
fuse_once is done
signal gap to exit 
fuse done
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

Since then, I cannot flash
I do not get anything on the /dev/ttyUSB console.
My board is bricked 
What should I do ?
If I did anything wrong, you should improve your doc

antoine-faravelon, 2020-06-11:
Hello,

The procedure does not seem to be the issue here. 
Have you re tried accessing the board since then ? In particular after leaving it unplugged for a while ?

ltancerel, 2020-06-17:
I have retried this since today.
I still get the same behavior.
Is there anything else I can try ?

ltancerel, 2020-06-17:
After a few tries, I am able to flash and run some code on my board.
However, the code does not start on reset after the flash (it executes only once like before fusing).
Note that I can see things happen only when using the io=uart console.

antoine-faravelon, 2020-06-17:
Hello,

Indeed, you will only see things happen with io=uart. If you boot from flash, openocd will not/should not initialize, as the board needs to run freestanding, which is impossible with semihosting. This is working as expected.

elietaillardat, 2020-07-29:
Hi @antoine-faravelon 
I'm working within the same company than @ltancerel 
I've also tried to boot from flash my GAPuino board, but still in trouble.
First, I used the command `openocd-fuser-hyperflash` to fuse necessary bits.
Then, I verified with the program `examples/pmsis/test_features/efuse_burner` that I get the right values from the efuse bits concerned, according to the **Application Note AN004** (https://gwt-website-files.s3.eu-central-1.amazonaws.com/AN004_BootFromFlash.pdf) -> OK
Next, I used the `examples/pmsis/test_periph/blink_led` source code to test it, adding the `PLPBRIDGE_FLAGS += -f -hyper` option in the Makfile, and `io=uart` for the `make clean all run` command.
The behavior is difficult to analyse, in the first place I couldn't see any result (just logs in the UART console with cutecom) but no flash boot. I tried a lot, with various commands or step order (make image flash), unplu/plus USB cable, resetting, ... The board doesn't boot, and lines such as `Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff` comes, with no more access to the GAP8, and I have to unplug the board and wait a few minutes before being able to work on it again.
And sometimes (like not a lot at all), like magic, the board boot from flash when I press the rest button...
So is there a way to make it work properly and in a decent way? I noticed the https://github.com/GreenWaves-Technologies/gap_sdk/issues/91 issue, which is still opened, and quite similar to this one.
Thank you for your time

antoine-faravelon, 2020-07-29:
hi @elietaillardat 
So for the first part, procedure seems right. io=uart too.
But: PLPBRIDGE_FLAGS are no longer in use. Remove it.

Instead type "make image" then "make flash" to get the flash image built & flashed on your board.
For the boot from flash magic, this could be L2 keeping some code/data if reset button somehow fail.
So procedure is: "make clean all io=uart" then "make image", "make flash". 
Unplug everything, count a few seconds to be on the safe side and ensure all condensators are empty so reset is really done(gapuino only) and then plug it back.

elietaillardat, 2020-07-29:
Hey, thank you for your reply
Just before your answer I(ve just tried the make image flash and wait few seconds and indeed it works!

closed 2020-08-04 by antoine-faravelon