GreenWaves-Technologies/gap_sdk #139: IMPORTANT ISSUE regarding recent changes to makefile for cifar10 and mnist

tp19021, 2020-06-07:
The ciraf10 and mnist examples from autotiler that i've been working on last week as been working until today where when I run make clean all run I am getting this error relating to nntool_extra_generators.c:

[image]

I urgently request a fix to this error as running these codes is an important starting point for my msc thesis and I suspect it is a result of changes made by the devs to the makefiles.

sousoux, 2020-06-08:
Looks like you have not done a make sdk in the gap_sdk directory since you pulled the update.

tp19021, 2020-06-08:
I've ran make sdk in the gap_sdk as you suggested and tried running make clean all run on these examples but the problem is still persist. I tried it twice as well, the 2nd time after doing a git pull from the gap_sdk dir.

tp19021, 2020-06-09:
Hi Martin,

Could you or anyone else please get back to me soon. My MSC project to contribute for the company's board cannot proceed until this issue is resolved as a result of their updates on the gap_skd. Apologise for the rush.

Mohammed

sivasivarajah, 2020-06-09:
Have you downloaded latest version of autotiler lib?

If so, try as follows :
* rm -rf build install  (those folders should not have any of your source code)
* source sourceme.sh
* make clean all

And try again the example.
I have tried with the latest version(tag release-v3.5), after those commands it worked.

tp19021, 2020-06-09:
how do i check if ive got the latest autotiler lib?

sivasivarajah, 2020-06-09:
It has been updated with the latest release, that is the reason you have those compilation errors(mismatch between your lib and updated code).

tp19021, 2020-06-09:
I see should i run those commands from th4e ga_sdk folder?

sivasivarajah, 2020-06-10:
Yes in gap_sdk you should run thoses commands.

tp19021, 2020-06-10:
Yeah it works now thanks. Its worth mentioning that I also need to run make autotiler as well or run make sdk instead of make all.

closed 2020-06-10 by tp19021