GreenWaves-Technologies/gap_sdk #145: Gap8 configuration doubt [Question]

tp19021, 2020-06-12:
I'm doing some power consumption experiments with the gap8 and I need to change the the input voltage, FC FREQ and CL FREQ. In the benchmarks documentation you provide the max freq for these at the possible input voltages:

Input Voltage | Fabric Controller Max Freq | Cluster Max Freq
-- | -- | --
1.0 V | 150 MhZ | 90 MhZ
1.2 V | 250 MhZ | 75 MhZ

**I was wondering whats the minimum of these two frequencies I can use at these two voltages or what are the possible frequency settings I can use at these voltages?** 

Also the gap8 hw  ref manual's electrical characteristics says a different max CL freq at 1.2 V and 1 V:
Voltage (V) | SOC max clock Freq. (MHz) |  Cluster max clock Freq. (MHz)
-- | -- | --
1.20 | 250 | 170
1 | 150 | 87

What is the reason for this difference in CL Freq between the datasheet and the benchmark doc?

sousoux, 2020-06-12:
Cluster max frequency at 1.2 is 175MHz. Could you provide a pointer to the 75MHz? Must be a typo.

tp19021, 2020-06-12:
https://greenwaves-technologies.com/gap8-cnn-benchmarks/#:~:text=GAP8%20CNN%20Benchmarks,check%20cycle%20and%20energy%20results.

This is the link for the typo

tp19021, 2020-06-12:
But yeah what are the possible settings for the frequencies?

Pafrak, 2020-06-14:
Thanks,
we will correct it.

tp19021, 2020-06-14:
No problem. Do you have any comments regarding my 1st query?

Pafrak, 2020-06-14:
It is an implementation choice.

closed 2020-07-06 by sousoux