GreenWaves-Technologies/gap_sdk #146: Visual Wake Words not running

tp19021, 2020-06-12:
I've tried building and running the VWW example on the GAP8 but im getting this run time error regarding the graph constructor as shown below and not sure what the cause nor the fix for it is :

[image]

sousoux, 2020-06-13:
Looks like a memory allocation issue in the graph constructor. Have you changed the amount of memory allocated or changed compile options so that the binary is not stripped/includes a lot of debug information?

tp19021, 2020-06-13:
I've not made any changes of such. i just simply ran the make clean and then make run as mentioned in the README on VWW on the code as given and this error came up.

tp19021, 2020-06-14:
An update regarding VWW, its seems to be working now after retrying it. Im not sure what caused it to not work initially.

I was wondering how can I remove the print statements of each layer thats emebedded in the generatd kernel as shown below (its because im measuring the power of the NN architecture for VWW and the prints are in the way of the readings):

[cid:fa2b95e1-6564-4edc-b7a5-a160de4f2a6e]
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 13 June 2020 17:20
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] Visual Wake Words not running (#146)

I've not made any changes of such. i just simply ran the make clean and then make run as mentioned in the README on VWW on the code as given and this error came up.

tp19021, 2020-06-16:
An update regarding VWW, its seems to be working now after retrying it. Im not sure what caused it to not work initially.

I was wondering how can I remove the print statements of each layer thats emebedded in the generatd kernel as shown below (its because im measuring the power of the NN architecture for VWW and the prints are in the way of the readings):

[cid:3b6bc2fc-d6c7-4ad4-b288-6f23270cd6a2]

tp19021, 2020-06-17:
A further update. I managed to find the prints statements for the these layers as I was looking at the wrong version of the BUILD_MODEL initially. Theres a very odd new issue i'm facing where I set the gpio and I use the code for asserting the gpio as done in the benchmarks example code but for some reason for the vww code it doesn't work on the board but for all other examples like the ones under autotiler it works on the board just fine. When we do make -n on vww example, it shows D_PULP_, which pulls the rt_api.h which should pull the the peripheral driver for the gpio.  So the build has no issues apparently but despite that the gpio is not asserted at all. Note that I use the exact code from the benchmarks, same gpio pin, same gpio config and set functions. I consulted this issue with an expert university student  and he doesn't know where the issue lies with regards to vww. I was hoping if you could assist with solving this issue?

Mohammed
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 16 June 2020 21:20
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] Visual Wake Words not running (#146)

An update regarding VWW, its seems to be working now after retrying it. Im not sure what caused it to not work initially.

I was wondering how can I remove the print statements of each layer thats emebedded in the generatd kernel as shown below (its because im measuring the power of the NN architecture for VWW and the prints are in the way of the readings):

[cid:3b6bc2fc-d6c7-4ad4-b288-6f23270cd6a2]

tp19021, 2020-06-17:
Also to remind the kws and nmnist from nntools are giving a graph constructor error and this is from the latest code version for these two examples pulled from git.

closed 2020-07-23 by MoIbs-tech