GreenWaves-Technologies/gap_sdk #148: Flashing issue

EyssK, 2020-06-23:
Hello,

I am implementing a CNN on the first GAP8. I ported it and it runs smoothly on emulation.

When I try to flash the board, I often receive an error "flasher script could not connect to board, check your cables". Sometimes, very rarely, it succeed. 

All examples that do not require flashing are running perfectly.
It seems only the flashing step is failing.

 Do you know what can be the problem ?
Thank you

Yaooooo, 2020-06-23:
Hi,

I think we may need more information to find out your problem. Can you please tell us:

- Which board you have? If it's a GAPuino, it's v1 or v2?
- Which version of SDK you are using? 
- Whether this example works for you: https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/pmsis/test_features/filesystem

EyssK, 2020-06-24:
Thanks for your answer.

I have the GAPuino Model 1.0
I use the 3.5.1 release.
The filesystem example does not work, here is its output.

Did you receive information about flash chip failing ?

```
[usr@ubuntu1804 ~/gap_sdk/examples/pmsis/test_features/filesystem]$ make clean all run             
rm -f /home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8/GCC_RISCV/flash.img     
riscv32-unknown-elf-gcc  -DCONFIG_GAPUINO -DRT_FC_STACK_SIZE=2048 -D__RT_IODEV__=2 -fno-jump-tables
 -fno-tree-loop-distribute-patterns -fdata-sections -ffunction-sections -mchip=gap8 -mPE=8 -mFC=1 -
D__riscv__ -DCHIP_VERSION=1 -mnativeomp -D__pulp__ -DCONFIG_GAP -D__PULP_OS__ -MMD -MP -c test_file
system.c  -I/home/usr/gap_sdk/install/GAP8/include  -I/home/usr/gap_sdk/install/GAP8/include/io  -I
/home/usr/gap_sdk/install/workstation/include  -include /home/usr/gap_sdk/install/GAP8/include/rt/c
hips/gap/config.h -MD -MF /home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8/GCC_
RISCV/test_filesystem.d -o /home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8/GCC
_RISCV/test_filesystem.o                                                                           
riscv32-unknown-elf-gcc  -DCONFIG_GAPUINO -DRT_FC_STACK_SIZE=2048 -D__RT_IODEV__=2 -fno-jump-tables
 -fno-tree-loop-distribute-patterns -fdata-sections -ffunction-sections -mchip=gap8 -mPE=8 -mFC=1 -
D__riscv__ -DCHIP_VERSION=1 -mnativeomp -D__pulp__ -DCONFIG_GAP -D__PULP_OS__ -MMD -MP -c /home/usr
/gap_sdk/rtos/pulp/pulp-os/kernel/conf.c  -I/home/usr/gap_sdk/install/GAP8/include  -I/home/usr/gap
_sdk/install/GAP8/include/io  -I/home/usr/gap_sdk/install/workstation/include  -include /home/usr/g
ap_sdk/install/GAP8/include/rt/chips/gap/config.h -MD -MF /home/usr/gap_sdk/examples/pmsis/test_fea
tures/filesystem/BUILD/GAP8/GCC_RISCV/pulp-os/conf.d -o /home/usr/gap_sdk/examples/pmsis/test_featu
res/filesystem/BUILD/GAP8/GCC_RISCV/pulp-os/conf.o                                                 
riscv32-unknown-elf-gcc  -MMD -MP -Wl,--gc-sections   -I/home/usr/gap_sdk/install/GAP8/include  -I/
home/usr/gap_sdk/install/GAP8/include/io  -I/home/usr/gap_sdk/install/workstation/include  -include
 /home/usr/gap_sdk/install/GAP8/include/rt/chips/gap/config.h -o /home/usr/gap_sdk/examples/pmsis/t
est_features/filesystem/BUILD/GAP8/GCC_RISCV/test  /home/usr/gap_sdk/examples/pmsis/test_features/f
ilesystem/BUILD/GAP8/GCC_RISCV/test_filesystem.o /home/usr/gap_sdk/examples/pmsis/test_features/fil
esystem/BUILD/GAP8/GCC_RISCV/pulp-os/conf.o   -L/home/usr/gap_sdk/install/GAP8/lib/gap/gapuino -lpi
bsp -L/home/usr/gap_sdk/install/GAP8/lib/gap -lrt -lrtio -lrt -lgcc -T/home/usr/gap_sdk/install/wor
kstation/ld/link.gap8.ld -T/home/usr/gap_sdk/install/workstation/ld/gapuino.conf.ld  -nostartfiles 
-nostdlib -include /home/usr/gap_sdk/install/GAP8/include/rt/chips/gap/config.h                    
riscv32-unknown-elf-objdump -D /home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8
/GCC_RISCV/test > /home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8/GCC_RISCV/te
st.s                                                                                               
gapy --target=gapuino --platform=board --work-dir=/home/usr/gap_sdk/examples/pmsis/test_features/fi
lesystem/BUILD/GAP8/GCC_RISCV  --config-opt=flash/content/partitions/readfs/files=hello.txt   run -
-image --binary=/home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8/GCC_RISCV/test
                                                                                     
gapy --target=gapuino --platform=board --work-dir=/home/usr/gap_sdk/examples/pmsis/test_features/fi
lesystem/BUILD/GAP8/GCC_RISCV  --config-opt=flash/content/partitions/readfs/files=hello.txt   run -
-flash --binary=/home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8/GCC_RISCV/test
                                                                                                   
Flashing image with command:                                                                       
openocd -c "gdb_port disabled; telnet_port disabled; tcl_port disabled" -c "script interface/ftdi/g
apuino_ftdi.cfg; script target/gap8.tcl; script tcl/flash_image.tcl; script tcl/jtag_boot.tcl; gap_
flash_raw_hyper /home/usr/gap_sdk/examples/pmsis/test_features/filesystem/BUILD/GAP8/GCC_RISCV/targ
et.board.devices.flash.img 262236 /home/usr/gap_sdk/tools/gap8-openocd-tools; exit;"               
Open On-Chip Debugger 0.10.0+dev-00017-g0ab61554-dirty (2020-06-15-12:17)                          
Licensed under GNU GPL v2                                                                          
For bug reports, read                                                                              
        http://openocd.org/doc/doxygen/bugs.html    
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 9 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x495
1, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option ADBG_USE_HISPEED
[... 40 lines cut ...]
Info : gdb port disabled 
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : adv_dbg_unit debug unit selected
[... cut ...]
Traceback (most recent call last):
  File "/home/usr/gap_sdk/tools/gapy/gapy", line 162, in <module>
    main()
  File "/home/usr/gap_sdk/tools/gapy/gapy", line 152, in main
    operationFunc(args, config)
  File "/home/usr/gap_sdk/tools/gapy/run.py", line 146, in operationFunc
    raise RuntimeError('Runner has failed with value: %d' % status)
RuntimeError: Runner has failed with value: -1
/home/usr/gap_sdk/tools/rules/pulp_rules.mk:213: recipe for target 'run' failed
make: *** [run] Error 1
```
Thanks a lot

tp19021, 2020-07-10:
I'm also facing the exact same issue as well with the .dat file not being flashed despite removing the #PLPBRIDGE_FLAGS += -f for my own cnn and the visual_wake. I did try explicitely including the path for the .dat file for my cnn in READFS_FILES and that fixed the isue for visual_wake but not for my own examples strangly (well 2 of my own cnns didn't work but this other one cnn did)

Yaooooo, 2021-05-28:
Sorry for keeping this openn for long time. For your problem, you can send one email to support@greenwaves-technologies.com
We will reach out to you and resolve this.

closed 2021-05-28 by Yaooooo