GreenWaves-Technologies/gap_sdk #149: Running generated nntool code on gap8 issue

tp19021, 2020-06-23:
I've inputted an 8-bit quant tflite file to nntool and successfully generated the kernel.c and .h files and the .dat file for the constant params. I've written the main application to call the constructor, CNN function and its destructor but i'm not sure how to compile and run the main code with the kernels and .dat using the makefile examples. I tried adapting the autotiler make examples like from Mnist but i'm getting an issue with recognizing the Gap.h. 

Could you provide the commands on how to compile and run the main app with the generated kernels and the gap headers?

sousoux, 2020-06-24:
Hi. There are multiple examples of the build process. I'm afraid that as part of your studies you are going to have to learn how to read Makefiles!

closed 2020-06-24 by sousoux

tp19021, 2020-06-24:
Hello Martin,

Thats reasonable. As you suggested I've worked through and managed to debug the errors associated with unrecognized headers. But now I am facing this error where the build pauses when creating the contents of the BUILD folder as shown below:

[cid:7af707f7-64fc-4b16-a1ac-8d1787f5cd19]

It manages to generate this folder with the .d and .o for merged2c.c and merged2ckernel.c and I can't seem to pinpout out the cause to the best of my understanding as I use the same CFLAGS as done for the mnist's Makefile under autotiler (my makefile is mostly adapted from mnist's exepect the build and run commands for ATModel.c is based on nntool quick guide). Its hard to debug this since contents of the makefile for the mnist and cifar10 works seamlessly in fully generating the BUILD. Here it says missing operand which I'm guessing means the home and pulp-os folders but I'm not sure how to pass it from makefile or whether its needed to since it worked for mnist and cifar10.

I've attached the makefile above for reference.

Thank you,
Mohammed

tp19021, 2020-06-25:
Hello Martin,

You can ignore my last email as I've managed to get the makefile to work on my own CNN based on the makefile example for vww. Now that the code can run I am however still facing the same issue in my CNN as with your existing mnist and kws examples under nntool where the Graph constructor throws an error.  After looking into the constructors for my own CNN and mnist, it looks like the error comes from the AT_HYPERFLASH_FS_OPEN() which I think is trying to allocate L3 Flash memory with the contents of the .dat. but fails to do so. Could you kindly address this issue as it concerns the existing examples like mnist and kws.

Cheers,
Mohammed
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 24 June 2020 17:50
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] Running generated nntool code on gap8 issue (#149)

Hello Martin,

Thats reasonable. As you suggested I've worked through and managed to debug the errors associated with unrecognized headers. But now I am facing this error where the build pauses when creating the contents of the BUILD folder as shown below:

[cid:7af707f7-64fc-4b16-a1ac-8d1787f5cd19]

It manages to generate this folder with the .d and .o for merged2c.c and merged2ckernel.c and I can't seem to pinpout out the cause to the best of my understanding as I use the same CFLAGS as done for the mnist's Makefile under autotiler (my makefile is mostly adapted from mnist's exepect the build and run commands for ATModel.c is based on nntool quick guide). Its hard to debug this since contents of the makefile for the mnist and cifar10 works seamlessly in fully generating the BUILD. Here it says missing operand which I'm guessing means the home and pulp-os folders but I'm not sure how to pass it from makefile or whether its needed to since it worked for mnist and cifar10.

I've attached the makefile above for reference.

Thank you,
Mohammed