GreenWaves-Technologies/gap_sdk #150: .dat file memory allocation Failed

tp19021, 2020-06-26:
I'm fishing this error where in my CNN application generated nntool as well as for the existing example applications like mnist and kws under nntool, I'm facing this error where it fails to allocate the contents of the .dat file to L3 Flash for these 3 applications but for the vww example this was oddly not the case. I can't see anything different nor wrong regarding the variables and functions associated with this task in the cnn constructor in the generated kerenels nor in the built in headers like at_api_pmsis.h which contains the function definition for allocating the .dat contents to L3 memory.

I'm afraid its quite hard for me to debug this issue since ideally I shouldn't change any of the generated kernel code nor in the relevant api so I suspect its something to do with the .dat since it wasen't an issue with vww but with mnist, kws and my own application. However the .dat file is geneated from nntool so I'm not sure weather I should change the .dat file either.

sousoux, 2020-06-26:
Please could you provide the error message

tp19021, 2020-06-26:
[cid:fa95ebb9-f125-4826-b285-b35355b1bd7e]

Note that the error 3 is a print I added in at_api_pmisis.h specifically in the at_hyperflash_fs_open function in which this line of code:
file->file = pi_fs_open(&file->fs, filename, is_write ? PI_FS_FLAGS_WRITE : 0);
  if (file->file == NULL)
  {
    printf("error 3\n");
    pi_fs_unmount(&file->fs);
    pi_flash_close(&file->hyperflash);
    *err = -1;
    return;
  }

tp19021, 2020-06-29:
Hi Martin,

Let me know how it goes with this error. I want to give an update that I also mapped the quantized graphs for Inception_v1 and MobileNet_v1 which were from the tensorflow hosted models website and they also produce the same graph constructor error.
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 26 June 2020 16:08
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] .dat file memory allocation Failed (#150)

[cid:fa95ebb9-f125-4826-b285-b35355b1bd7e]

Note that the error 3 is a print I added in at_api_pmisis.h specifically in the at_hyperflash_fs_open function in which this line of code:
file->file = pi_fs_open(&file->fs, filename, is_write ? PI_FS_FLAGS_WRITE : 0);
  if (file->file == NULL)
  {
    printf("error 3\n");
    pi_fs_unmount(&file->fs);
    pi_flash_close(&file->hyperflash);
    *err = -1;
    return;
  }

tp19021, 2020-07-03:
Hi Martin,

Has there been any news regarding this issue yet? I'm afraid its very important for my research on the gap8 to be able to get these benchmarks running adn to the best of my skills I can't debug it since its an issue in the code generated from nntool.

Mohammed

Yaooooo, 2020-07-04:
Hi,

It seems the filesystem cannot find your *.dat file in the flash, when
opening the filesystem.
Can you please confirm, the filesystem has been flashed into the flash
successfully, with this *.dat file included?

To write this file into the filesystem, you need to use like we described
in the doc:
https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html#section21

Yao

tp19021, 2020-07-04:
Hi Yao,

So looking at the mnist example under nntools and running it as provided from git which gave the same graph constructor error, I looked at the make file for the READFS_FILES in its Makefile and its assigned as follows (the same is seen for visual_wake which works...):

READFS_FILES=$(abspath $(MODEL_TENSORS))
PLPBRIDGE_FLAGS += -f

I can't tell if this means the .dat file is included. I also tried make all to build flash image and make run but that did not fix the issue.

https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html#section21

<http://greenwaves-technologies.com/>

tp19021, 2020-07-04:
I've also tried all the make commands in that section you pointed out but still does not fix the issue.

https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html#section21

<http://greenwaves-technologies.com/>

Yaooooo, 2020-07-06:
Hi,

I reproduce the problem and it works fine with this sequence:

   1. make clean all
   2. make image
   3. make flash
   4. make run

It also works like this: remove this line in Makefile:
-PLPBRIDGE_FLAGS += -f

then: make clean all run

Yao

tp19021, 2020-07-06:
Hi Yao,

I'm afraid neither of these solutions work on mnist. Should I send you the mnist folder for you to see if the error persist?

I will try pulling in recent changes in the sdk i.e. release 3.6 and try your solutions again and let you know.

Mohammed

<http://greenwaves-technologies.com/>

tp19021, 2020-07-07:
Hello Yao,

After upgrading to sdk release 3.6 I managed to get the mnist under nntool to work but for the visual_wake example its now giving me the graph constructor error when it didn't before. The error still persisted when trying both of your solutions. I've not changed anything in the source code for this example either. Could you please look into this on your end?

Mohammed

<http://greenwaves-technologies.com/>

Yaooooo, 2020-07-07:
Yes, in fact it's the problem of this PLPBRIDGE_FLAGS, which should be
removed from all the examples.
Since this flag, the examples jump the flash step, therefore, you don't
have the right filesystem for your example/application.
We will push a fix for this, however, it should works for you if you remove
this flag in the makefile

Yao

tp19021, 2020-07-07:
I see however I did try commenting out the PLPBRIDE_FLAGS in visual_wake but as I said it didn't fix the flash error. Could you confirm this on your end incase I have done something wrong, when you can ofcourse?

<http://greenwaves-technologies.com/>

tp19021, 2020-07-07:
I did fully remove PLPBRIDGE_FLAGS fully and ran make clean all run, I can very much confirm it has not fixed the flash error.
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 07 July 2020 13:57
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] .dat file memory allocation Failed (#150)

I see however I did try commenting out the PLPBRIDE_FLAGS in visual_wake but as I said it didn't fix the flash error. Could you confirm this on your end incase I have done something wrong, when you can ofcourse?

<http://greenwaves-technologies.com/>

Yaooooo, 2020-07-07:
Just tested and there's no issue for me on the last release (3.6) by
removing the PLPBRIDGE_FLAGS:

*As you can see my modifies:*

*(base)
yao@ubuntu:~/Project/SDK/gap_sdk_github/examples/nntool/visual_wake$ git
diff .diff --git a/examples/nntool/visual_wake/Makefile
b/examples/nntool/visual_wake/Makefileindex 76abe256..d376486f 100644---
a/examples/nntool/visual_wake/Makefile+++
b/examples/nntool/visual_wake/Makefile@@ -52,14 +52,13 @@
USE_PMSIS_BSP=1 APP = vww APP_SRCS += vww.c $(MODEL_GEN_C)
$(MODEL_COMMON_SRCS) $(CNN_LIB)  APP_CFLAGS += -g -O3 -mno-memcpy
-fno-tree-loop-distribute-patterns  APP_CFLAGS += -I. -I$(MODEL_COMMON_INC)
-I$(TILER_EMU_INC) -I$(TILER_INC) $(CNN_LIB_INCLUDE) -I$(realpath
$(MODEL_BUILD)) APP_CFLAGS += -DPERF -DAT_MODEL_PREFIX=$(MODEL_PREFIX)
$(MODEL_SIZE_CFLAGS) APP_CFLAGS += -DSTACK_SIZE=$(CLUSTER_STACK_SIZE)
-DSLAVE_STACK_SIZE=$(CLUSTER_SLAVE_STACK_SIZE) APP_CFLAGS +=
-DAT_IMAGE=$(IMAGE)  READFS_FILES=$(abspath
$(MODEL_TENSORS))-PLPBRIDGE_FLAGS += -f  # all depends on the model all::
model*

*The output log of this example:*

Entering main controller
Constructor
Reading image
Image
/home/yao/Project/SDK/gap_sdk_github/examples/nntool/visual_wake/images/COCO_val2014_000000174838_1.ppm:
 [W: 238, H: 208] Bytes per pixel 3, HeaderSize: 15
P6
238 208
255

Image
/home/yao/Project/SDK/gap_sdk_github/examples/nntool/visual_wake/images/COCO_val2014_000000174838_1.ppm,
[W: 238, H: 208], Bytes per pixel 3, Size: 148512 bytes, Loaded successfully
Finished reading image
Call cluster
person seen (2710, 30057)

                  S1_Op_input_1_formatterBody: Cycles:      69609,
Operations:     222768, Operations/Cycle: 3.200276
                      S2_Conv2d_8x3x3x3_Relu6: Cycles:    1939589,
Operations:    2772224, Operations/Cycle: 1.429284
                     S3_Conv2d_12x8x1x1_Relu6: Cycles:     410515,
Operations:    1188096, Operations/Cycle: 2.894160
                     S4_Conv2d_12x1x3x3_Relu6: Cycles:     165532,
Operations:     374400, Operations/Cycle: 2.261798
                           S5_Conv2d_8x12x1x1: Cycles:      97725,
Operations:     299520, Operations/Cycle: 3.064927
                     S6_Conv2d_20x8x1x1_Relu6: Cycles:     155279,
Operations:     499200, Operations/Cycle: 3.214859
                     S7_Conv2d_20x1x5x5_Relu6: Cycles:     313022,
Operations:    1622400, Operations/Cycle: 5.183022
                          S8_Conv2d_16x20x1x1: Cycles:     189330,
Operations:     998400, Operations/Cycle: 5.273332
                    S9_Conv2d_36x16x1x1_Relu6: Cycles:     351632,
Operations:    1797120, Operations/Cycle: 5.110798
                    S10_Conv2d_36x1x3x3_Relu6: Cycles:     366576,
Operations:    1123200, Operations/Cycle: 3.064030
                         S11_Conv2d_16x36x1x1: Cycles:     328514,
Operations:    1797120, Operations/Cycle: 5.470452
                          S12_MatAdd_16x52x60: Cycles:      68409,
Operations:      49920, Operations/Cycle: 0.729729
                   S13_Conv2d_32x16x1x1_Relu6: Cycles:     314780,
Operations:    1597440, Operations/Cycle: 5.074782
                    S14_Conv2d_32x1x7x7_Relu6: Cycles:     374032,
Operations:    1248000, Operations/Cycle: 3.336613
                         S15_Conv2d_20x32x1x1: Cycles:      78639,
[... 87 lines cut ...]
                   S59_Conv2d_200x1x7x7_Relu6: Cycles:     306366,
Operations:     560000, Operations/Cycle: 1.827879
                        S60_Conv2d_96x200x1x1: Cycles:     104571,
Operations:    1075200, Operations/Cycle: 10.282009
                  S61_Conv2d_152x96x1x1_Relu6: Cycles:      81371,
Operations:     817152, Operations/Cycle: 10.042300
                   S62_Conv2d_152x1x5x5_Relu6: Cycles:      60630,
Operations:     221312, Operations/Cycle: 3.650206
                        S63_Conv2d_96x152x1x1: Cycles:      82918,
Operations:     817152, Operations/Cycle: 9.854941
                            S64_MatAdd_96x7x8: Cycles:      10119,
Operations:       5376, Operations/Cycle: 0.531278
        S65_Conv2d_640x96x1x1_AveragePool_7x7: Cycles:    1390038,
Operations:    3041920, Operations/Cycle: 2.188372
                         S66_Linear_2x640x1x1: Cycles:       2450,
Operations:       1280, Operations/Cycle: 0.522449
                                  S67_SoftMax: Cycles:       1449,
Operations:          2, Operations/Cycle: 0.001380

                                        Total: Cycles:   11360825,
Operations:   48393362, Operations/Cycle: 4.259669

*ONLY ONE COMMENT: it takes time (around 10s ) to have the trace come out,
need to wait a bit after the info of "reading image....". It's waiting for
the image be loaded into memory. *

tp19021, 2020-07-07:
*(base)
yao@ubuntu:~/Project/SDK/gap_sdk_github/examples/nntool/visual_wake$ git
diff .diff --git a/examples/nntool/visual_wake/Makefile
b/examples/nntool/visual_wake/Makefileindex 76abe256..d376486f 100644---
a/examples/nntool/visual_wake/Makefile+++
b/examples/nntool/visual_wake/Makefile@@ -52,14 +52,13 @@
USE_PMSIS_BSP=1 APP = vww APP_SRCS += vww.c $(MODEL_GEN_C)
$(MODEL_COMMON_SRCS) $(CNN_LIB) APP_CFLAGS += -g -O3 -mno-memcpy
-fno-tree-loop-distribute-patterns APP_CFLAGS += -I. -I$(MODEL_COMMON_INC)
-I$(TILER_EMU_INC) -I$(TILER_INC) $(CNN_LIB_INCLUDE) -I$(realpath
$(MODEL_BUILD)) APP_CFLAGS += -DPERF -DAT_MODEL_PREFIX=$(MODEL_PREFIX)
$(MODEL_SIZE_CFLAGS) APP_CFLAGS += -DSTACK_SIZE=$(CLUSTER_STACK_SIZE)
-DSLAVE_STACK_SIZE=$(CLUSTER_SLAVE_STACK_SIZE) APP_CFLAGS +=
-DAT_IMAGE=$(IMAGE) READFS_FILES=$(abspath
$(MODEL_TENSORS))-PLPBRIDGE_FLAGS += -f # all depends on the model all::
model*

So here I'm a bit confused as I can see the PLPBRIDGE_FLAGS still included? Is it on the same line as READFS_FILES?

<http://greenwaves-technologies.com/>

Yaooooo, 2020-07-07:
This is because you don't know how the git diff works.
In fact, you can see a "-" before the PLPBRIDGE_FLAGS line, which means
according to the remote (github) server version, my local version has
removed this line.

tp19021, 2020-07-07:
<http://greenwaves-technologies.com/>

tp19021, 2020-07-07:
This is the makefile im working with.
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 07 July 2020 14:55
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] .dat file memory allocation Failed (#150)

<http://greenwaves-technologies.com/>

tp19021, 2020-07-07:
I tried comparing mnist's make file and visual wakes and tried including APP_CFLAGS += -DUSE_HYPER which was the only relevant difference I could find but I'm afraid that didn't fix the issue either.
I don't understand why it worked for mnist only now. I dont think its a HW issue since mnist worked. So i'm not quite sure what to do to fix. I'm willing to send any of the code related to visual_wake to you if that helps.
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 07 July 2020 14:55
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] .dat file memory allocation Failed (#150)

This is the makefile im working with.
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 07 July 2020 14:55
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] .dat file memory allocation Failed (#150)

<http://greenwaves-technologies.com/>

tp19021, 2020-07-07:
This is the output I'm getting and I included a print thats pointed out in the vwwkernel.c for when it throws an error if it the FS_OPEN() fails:

[cid:592cd39b-d853-45e7-aea6-3352be486b47]

<http://greenwaves-technologies.com/>

Yaooooo, 2020-07-08:
We cannot see any information you attached, please check!

tp19021, 2020-07-08:
I sent another email directly to your personal email as I think git doesn't let me send attachments larger than 4 MB.

Mohammed

tp19021, 2020-07-10:
Hello Yao,

Has there been any update on the example cnn app I sent you? One of my newest examples I made somehow works but I couldn't get its makefile to work on this emnist example I sent you. So I'm getting inconsistent behavior in terms of the .dat file being flashed or not with the examples I've got.

Mohammed

tp19021, 2020-07-19:
Hello Yao,

Apologize for disturbing you while your busy. I'm afraid I'm not entirety sure why my own example cnn applications like the emnist example I sent you still has the .dat file Flash Memory allocation error despite removing the PLPBRIDGE_FLAG and trying to explicitly include the path to the .dat file to READFS_FILES. Currently the visual wake words example works as mentioned before and the same goes for the mnist_lstm (due to explicitly stating the .dat file in READFS_FILES for both examples) . Its hard to debug this issue especially since its showing inconsistent behavior across different examples.

Would it be alright if you or anyone else get back to me on this issue as its important for my research work and I've tried all options in the makefile associated with the flash step. I can also send you another example of my own to ensure this issue is resolved.

Thank you,
Mohammed
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 10 July 2020 13:33
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] .dat file memory allocation Failed (#150)

Hello Yao,

Has there been any update on the example cnn app I sent you? One of my newest examples I made somehow works but I couldn't get its makefile to work on this emnist example I sent you. So I'm getting inconsistent behavior in terms of the .dat file being flashed or not with the examples I've got.

Mohammed

sousoux, 2022-06-01:
Closed due to inactivity

closed 2022-06-01 by sousoux