GreenWaves-Technologies/gap_sdk #155: GPIO examples not working

elietaillardat, 2020-07-21:
Hi,
The 2 examples I found in the gap_sdk (pmsis and native/pulp-os) for **gpio input testing are not working** with the 3 gapuino boards I have.
My gap_sdk is sourced with **gapuino_v2.sh**, and I'm on **release 3.5** (3.6 have some issues)
I have the jumber **J24 set** up (but I tried both position), nothing appends when pressing the push button or sending voltage to another GPIO pin. I tried different pin macros in the code than the PI_GPIO_A0_PAD_12_A3, but nothing either.
Any ideas? Thanks

sivasivarajah, 2020-07-21:
Hi Elie,

I am guessing you are using PulpOs. The feature to handle input events and callbacks is missing, it is scheduled to for the next release by the end of month.
You can use Freertos which has this feature(make PMSIS_OS=freertos clean all run).

Best,

elietaillardat, 2020-07-22:
Thanks @sivasivarajah I 'm now able to build the pmsis example, but we are developing an application using PULP-OS, nntool etc... and if I try to compile our code with this make option, it fails. Is there a way to make a compromise between both os if I want to use the GPIOs, or do we have to wait the next release? Thanks :)

sivasivarajah, 2020-07-22:
Can you develop the errors you get? We can take a look if you want.

If it is not an easy step to switch OS, I can push an early release for this part yes.

elietaillardat, 2020-07-22:
For instance:

camControl.h:5:10: fatal error: rt/rt_api.h: No such file or directory
 #include "rt/rt_api.h"
          ^~~~~~~~~~~~~
compilation terminated.

sivasivarajah, 2020-07-22:
Where is this file from?
Is there anyway you can use PMSIS API instead of rt_api?

elietaillardat, 2020-07-22:
I suceeded to build my app by changing RT_L2_DATA to GAP_L2_DATA, I'm new to this so I'm not sure what it really involves.

sivasivarajah, 2020-07-23:
Using GAP_L2_DATA is the same as RT_L2_DATA for PulpOs, data will be in L2.
But consider using these : PI_L2, PI_FC_L1 ot PI_CL_L1. Those macros are defined in both PulpOs and Freertos.

Also I believe you can use [PMSIS_API](https://greenwaves-technologies.com/manuals/BUILD/PMSIS_API/html/modules.html) instead of rt_XX or pulp functions.

As for execution, does application start? And what does it do ?
This message is from the bridge, here OpenOCD, trying to read something.
A full log may be helpful.

elietaillardat, 2020-07-23:
Allright, do you know what can replace this function:

rt_time_get_us()

May be it is why it tries to read something

sivasivarajah, 2020-07-23:
> rt_time_get_us()

Are you using timer in your application? And which one exactly?

A patch for the gpio is on branch **3.6.1_dev**,  you can try gpio_input example.

elietaillardat, 2020-07-23:
I checkout this version, first:
- The GPIO input exemple for pmsis is now working without PMSIS_OS=freertos
- When I build my app, I get an error:
_myApp/BUILD_MODEL_SQ8BIT/myAppModel.c:30: undefined reference to `Load_RNN_SQ8_Library'_
I've commented this line in the generated file myAppModel.c, and it works (why is this function not recognized?)
- Then I add some GPIO stuff to my app from the example, and the app is now working again with no bugs

sivasivarajah, 2020-07-24:
Good to hear!
I pushed a fix for gpio mask, driver was checking wrong mask when trying to output values. You might want to pull latest updates from this branch.

For the Load_RNN_SQ8_Library lib, do you have latest version of autotiler?

elietaillardat, 2020-07-24:
Yes, I've cleaned my gap_sdk install and try 'make autotiler' again to be sure.
Then I did 'make clean all' for my app but still the Load_RNN_SQ8_Library error.

elietaillardat, 2020-07-29:
Any solution?

sivasivarajah, 2020-07-29:
Hi @elietaillardat ,

You need to upgrade to release 3.6. Otherwise if you want to stay with your version, you have to checkout updates from latest version to your current version, although this is not best way.

elietaillardat, 2020-07-29:
But I'm on your branch 3.6.1_dev, is it not based on the release 3.6?

sivasivarajah, 2020-07-29:
Ah yes indeed, sorry I forgot you switched to this branch.
Can you share traces of compilation with your Makefile?

elietaillardat, 2020-07-29:
Building GAP8 mode with 8 bit quantization
script model/nntool_script
GEN ... ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Generator_Util.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Generators_SQ8.c ../gap_sdk/tools/nntool/autotiler/generators/nntool_extra_generators.c
APP_SRCS... app.c BUILD_MODEL_SQ8BIT/appKernels.c  ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Activation_SQ8.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Bias_Linear_SQ8.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Conv_SQ8.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Pooling_SQ8.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Conv_DW_SQ8.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_MatAlgebra_SQ8.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_SoftMax_SQ8.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_AT_Misc.c ../gap_sdk/tools/nntool/autotiler/kernels/norm_transpose.c camControl.c imgAlgo.c imgIO.c imgDraw.c nn.c utils.c
APP_CFLAGS... -g -O1 -mno-memcpy -fno-tree-loop-distribute-patterns -I. -I../gap_sdk/libs/gap_lib/include -I../gap_sdk/tools/autotiler_v3/include -I../gap_sdk/tools/autotiler_v3/include -I../gap_sdk/tools/autotiler_v3/generators/CNN -I../gap_sdk/tools/autotiler_v3/generators/CNN -I../gap_sdk/tools/nntool/autotiler/kernels -IBUILD_MODEL_SQ8BIT -DAT_MODEL_PREFIX=app -DAT_INPUT_HEIGHT=28 -DAT_INPUT_WIDTH=28 -DAT_INPUT_COLORS=1 -DSTACK_SIZE=4028 -DSLAVE_STACK_SIZE=1024 -DEYEMB_MODE=0 
mkdir BUILD_MODEL_SQ8BIT
cp model/app.tflite BUILD_MODEL_SQ8BIT/app.tflite
echo "GENERATING NNTOOL STATE FILE"
GENERATING NNTOOL STATE FILE
nntool -s model/nntool_script BUILD_MODEL_SQ8BIT/app.tflite 
-s is not a recognized command, alias, or macro
model/nntool_script is not a recognized command, alias, or macro
BUILD_MODEL_SQ8BIT/app.tflite is not a recognized command, alias, or macro
open - opening graph file BUILD_MODEL_SQ8BIT/app.tflite
new_tflite_graph_all - Importing TFLITE model version 3
nngraph - update graph dimensions
nngraph - calculate liveness
eliminate_transposes - eliminating unnecessary transposes
eliminate_transposes - no further transpose sequences found
nngraph - adjusted order
nngraph - update graph dimensions
nngraph - calculate liveness
move_node_up - Node MAX_POOL_2D_0_1 can be moved
move_node_up - Node MAX_POOL_2D_0_3 can be moved
move_node_up - Moving node MAX_POOL_2D_0_1 between DEPTHWISE_CONV_2D_0_0 and DEPTHWISE_CONV_2D_0_0_activation
move_node_up - Moving node MAX_POOL_2D_0_3 between CONV_2D_0_2 and CONV_2D_0_2_activation

[...]

echo "GENERATING AUTOTILER MODEL"
GENERATING AUTOTILER MODEL
nntool -g -M BUILD_MODEL_SQ8BIT -m appModel.c -T BUILD_MODEL_SQ8BIT/tensors -H appInfo.h  BUILD_MODEL_SQ8BIT/app.json
echo "COMPILING AUTOTILER MODEL"
COMPILING AUTOTILER MODEL
gcc -g -o BUILD_MODEL_SQ8BIT/GenTile -I. -I../gap_sdk/tools/autotiler_v3/include -I../gap_sdk/tools/autotiler_v3/include -I../gap_sdk/tools/autotiler_v3/generators/CNN -I../gap_sdk/tools/autotiler_v3/generators/CNN -I../gap_sdk/tools/nntool/autotiler/generators -I../gap_sdk/tools/autotiler_v3/generators/CNN -I../gap_sdk/tools/autotiler_v3/generators/CNN -I../gap_sdk/tools/nntool/autotiler/kernels ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Generator_Util.c ../gap_sdk/tools/autotiler_v3/generators/CNN/CNN_Generators_SQ8.c ../gap_sdk/tools/nntool/autotiler/generators/nntool_extra_generators.c BUILD_MODEL_SQ8BIT/appModel.c ../gap_sdk/tools/autotiler_v3/lib/libtile.a
/tmp/cc82SNby.o : Dans la fonction « appModel » :
../steropes/BUILD_MODEL_SQ8BIT/appModel.c:30 : référence indéfinie vers « Load_RNN_SQ8_Library »
collect2: error: ld returned 1 exit status
model_rules.mk:78: recipe for target 'BUILD_MODEL_SQ8BIT/GenTile' failed
make: *** [BUILD_MODEL_SQ8BIT/GenTile] Error 1

elietaillardat, 2020-07-29:
I think I know the problem, looking at it.

elietaillardat, 2020-07-29:
I found what it was thanks ;)

sivasivarajah, 2020-08-03:
`PI_L2` are some macros to store variables in L2. In any case global variables are always in L2, default behaviour.

As for `pi_task_t`, those are events tasks used by drivers(cluster, interfaces) and bsp drivers to sync transfers for example. They can be used as simple blocking tasks, where one can use it to start a transfer and then wait on this event task until completion or callbacks tasks, callback is executed when the transfer is done for example. Cf. rtos/pmsis/pmsis_api/include/task.h.

You can take a look at this usage in examples/pmsis/test_features/cluster_callback/test_cluster_callback.c.

Yaooooo, 2020-09-02:
This issue will be closed, please feel free to reopen it if you have further question related. 
Thanks.

closed 2020-09-02 by Yaooooo