GreenWaves-Technologies/gap_sdk #156: CNN flash step not working

MoIbs-tech, 2020-07-23:
Hello,

I've been making my own CNN benchmarks to run on gap8 using nntools. The mapping process has been successful, where I adapt the Makefiles, main application and nntool scripts from the existing examples like visual wake words. 

However, I'm getting this problem where it sometimes is able to write the .dat file to the flash but most of the it doesn't work. I've tried including the .dat file in the READFS variable in the makefile after reading the GAP8 Manual, specifically the section on hyperflash. Unfortunately this has shown not to work either. 

The error says that the flasher script could not connect ot board and that I should check my cables but I think my cables are fine since the examples under autotiler work, the existing examples under nntool works but not initially as they worked after making the following tweaks to their makefiles like removing PLPBRIDGE_FLAG, explicitly including the .dat file as mentioned above.

I've tried virtually all possible fixes but none of them worked. Some assistance is very much appreciated. I can provide my CNN example to see if it works for anyone who is willing to assist.

Note that I'm using the the SDK release 3.6.

Pafrak, 2020-07-27:
Hi Molbs-tech,
 can you please provide your example? since on my side I see no problem with VWW. Are you running on board or gvsoc?

MoIbs-tech, 2020-07-27:
I managed to get VWW to work but I had to explicitly include the .dat file in the READFS. I had to rely on this fix to try and get it to work but it doesn't work all the time. It almost unpredictable when the flash step will work for some benchmarks. However if it works once for a benchmark it will likely work once I unplug and replug the cable and rebuild the code.

MoIbs-tech, 2020-07-27:
Could you let me know if you received it as git won't let me send zip files. If not could you provide your work email so that I can send it that way?

Pafrak, 2020-07-27:
you can check my email in profile

MoIbs-tech, 2020-07-27:
I've sent it to you.

Pafrak, 2020-07-27:
There problem was that in your makefile you are doing this:

clean:: clean_model

but when you do a just a make all, the model does not get regenerated. So you had problem flashing because the file was not there (in your pc). Moreover there was a bit of a mess in inclusion and flows of things. I corrected them. I just sent it to you over email.

closed 2020-07-27 by Pafrak

MoIbs-tech, 2020-07-27:
OK so do I just run make clean all run using the fixed makefile and it should work?

Pafrak, 2020-07-27:
yes

MoIbs-tech, 2020-07-27:
I'm afraid that change didn't work as I tried it for mnist lstm and it didn't work. Ive attached the make file incase im doing something wrong.

I tried make clean all run, didn't work and the same for make all run

MoIbs-tech, 2020-07-27:
I commented out clean :: clean_model as you said in that file.
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 27 July 2020 15:59
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] CNN flash step not working (#156)

I'm afraid that change didn't work as I tried it for mnist lstm and it didn't work. Ive attached the make file incase im doing something wrong.

I tried make clean all run, didn't work and the same for make all run

MoIbs-tech, 2020-07-27:
Part of me is a bit doubtful if what you said is the cause of it since the cable is not detected by the flash script thats responsible for uploading the.dat file to flash.

[cid:c086ac70-b682-4274-876d-b7c6da57b727]

________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 27 July 2020 16:00
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] CNN flash step not working (#156)

I commented out clean :: clean_model as you said in that file.
________________________________
From: Mohammed Ibrahim <tp19021@bristol.ac.uk>
Sent: 27 July 2020 15:59
To: GreenWaves-Technologies/gap_sdk <reply@reply.github.com>
Subject: Re: [GreenWaves-Technologies/gap_sdk] CNN flash step not working (#156)

I'm afraid that change didn't work as I tried it for mnist lstm and it didn't work. Ive attached the make file incase im doing something wrong.

I tried make clean all run, didn't work and the same for make all run