GreenWaves-Technologies/gap_sdk #158: [IMPORTANT Question] How to detect if a Performance Counter has overflowed?

MoIbs-tech, 2020-08-02:
Has the implementation of the performance counters addressed the above question as if your measruing for example execution cycles of a large cnn model running on the gap8, would the api have included code to count the number of overflows of a measured performance counter so that we can account its true value?

Pafrak, 2020-08-03:
This is a 32 bit register, so if you run at 50 MhZ you should be able to measure up to 85 sec, is your application longer? @haugoug can you confirm this?

MoIbs-tech, 2020-08-03:
But how do you know how to set up the register to count it? Well its hard to say I guess for example the visual wake has a lot of layers in its cnn so surely from that it would be concern of whether a PC would overflow? Theres also cnn models with 128 kernel wide layers too

haugoug, 2020-08-03:
Unfortunately, the API does not include anything for the overflow, although the HW has a mechanism to trigger an interrupt when the counter overflows.We'll include that in a future release. For now you are sure there is no overflow if your execution takes less than 85 seconds.

Yaooooo, 2020-09-02:
I will close this issue, please feel free to reopen it if you have further question related. 
Thanks.

closed 2020-09-02 by Yaooooo