GreenWaves-Technologies/gap_sdk #160: Questions: Does the GAP8 have internal SRAM components? What sort of operations does nntool measure?

MoIbs-tech, 2020-08-17:
What is the memory implementation of the L1 and L2 memroy that is shared by the cluster? Is there any SRAM components inside the processor?

Also regarding the nntool, when you enable the performance to be measured, what sort of operations for each layer is it keeping count of?

sousoux, 2020-08-18:
_What is the memory implementation of the L1 and L2 memroy that is shared by the cluster? Is there any SRAM components inside the processor?_

GAP8 has 512KB of L2 and 64KB of Cluster L1 SRAM integrated into the chip.

_Also regarding the nntool, when you enable the performance to be measured, what sort of operations for each layer is it keeping count of?_

Cluster cycles - It's possible to manually change the measured parameter in the generated code

closed 2020-08-18 by sousoux

MoIbs-tech, 2020-08-18:
I see thank you but thats a different parameter as the tool provides ops and and cluster cycles. So what is the ops actually referring to?