GreenWaves-Technologies/gap_sdk #161: Image capture problem

Nuvey81, 2020-08-26:
Hi,
I use the Gap8 with a camera for months and it was working fine. Unfortunately, couple of days ago, I have some problems.
Indeed, I try to process images of 7 segment display, captured with the camera, but now the captured image is "broken". The image is too bright as if the white was off (see the following before/after images, numbers 3, with the same display brightness).
Do you have any idea of the source of the problem ? Have you ever seen something similar ?

Thanks for your help.
[image]
[image]

Yaooooo, 2020-08-26:
Hi,

Can you please tell us which "branch" or which "tag" you are working with?
Besides of potential HW problem, It can be camera tuning issue. 
You can tune the camera by changing its camera registers.

Best,
Yao

Nuvey81, 2020-08-26:
Thanks for your answer,
I use the branch **3.6.1_dev**.
Can you tell me how can i change camera register ? Do you have any documentation ?
I already tried to run the camera test with _examples/pmsis/test_periph/test_camera_io_ but the same problem appears.
My tuning for the camera are based on this example.

Thanks a lot for your help,
N

Yaooooo, 2020-09-02:
Hi

You can tune the camera by:

- changing the init table here: https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/rtos/pmsis/pmsis_bsp/camera/himax/himax.c
- using this api: 
https://greenwaves-technologies.com/manuals/BUILD/PMSIS_BSP/html/group__Camera.html#gaec65a721932f3f7d0ef80549e66a29e7

However, since the camera datasheet is under NDA with HIMAX, if you want to have the datasheet, you have to contact Himax.

closed 2020-09-29 by Yaooooo