GreenWaves-Technologies/gap_sdk #164: RawSerial Mbed callback function never triggered

Gualor, 2020-09-19:
Hi everyone,
I'm working on a project on the **Gapuino development board**, my goal is to implement [rosserial_mbed](http://wiki.ros.org/rosserial_mbed) module on the Gap8 to be able to communicate with a ROS (melodic) node running on Ubuntu 18.04.

Rosserial relies on [BufferedSerial](https://os.mbed.com/users/sam_grove/code/BufferedSerial/) library, which is an implementation of Mbed RawSerial, by exploiting its ability to attach callback functions on the UART rx pin to store data inside a buffer. This is done in order to have non-blocking read/write functionalities. 

I narrowed the problem down to an interrupt handling issue on the Gap8, which seems unable to call callbacks on the triggering of the UART rx pin. To demonstrate what i just explained, I wrote down a simple example program to show such issue:

```C++
#include "mbed.h"

RawSerial pc(USBTX, USBRX, 9600);

void FooCallback(void)
{
    printf("This function never gets called\n");
}

int main(void)
{
    // Attach callback on UART rx
    pc.attach(&FooCallback, RawSerial::RxIrq);

    while(1)
    {
        // Do stuff
        printf("...\n");
    }
}
```

I want to also add a finding (which could be unrelated to the issue) that emerged when reading at the source code for the porting of the mbed library for the Gap8. I came across this line of code inside [serial_api.c](https://github.com/GreenWaves-Technologies/gap_sdk/blob/3.3.3/rtos/mbed-os/targets/TARGET_GWT/api/serial_api.c#L153) which states that uart_irq_rx "/* Do no support in GAP8 */".

My current setup is the following:
- OS: Ubuntu 18.04
- IDE: Visual Studio Code running PlatformIO extension
- Board: GAPuino GAP8
- Platform: RISC-V GAP
- Framework: Mbed

Thank you in advance, any suggestion is welcomed.

Yaooooo, 2020-09-22:
Hi,

For your problem, could you please provide us more information:

1. Which version of GAPuino you are using? If you don't know the version, please put a photo of it, thanks.
2, Which Uart port you are using? (Since the uart is connected to FTDI chip by default, on GAPuino board). 
3, Have you tried with these examples:
- https://github.com/GreenWaves-Technologies/gap_sdk/tree/3.3.3/examples/native/mbed/test_driver_gap8/test_Serial
- https://github.com/GreenWaves-Technologies/gap_sdk/tree/3.3.3/examples/native/mbed/test_driver_c%2B%2B  (There are 4 Uart examples here)

BR,
Yao

Gualor, 2020-09-22:
Hi, thank you @Yaooooo for your quick response.

1. I think I'm using the GAPuino Version 1. but I add you the photos as requested to confirm it:
[image]
[image]

2. I'm interfacing with the USB of my laptop running Ubuntu 18.04, port name /dev/ttyUSB1

3. All of the example work fine without errors of any kind.

Seems like all of the blocking functions of read/write of both RawSerial and Serial work, but when I try using the "attach" method to let the UART rx interrupt call my callback function, it does not work. 

Regards,
Lorenzo

Yaooooo, 2020-09-22:
You have also an Async example here: https://github.com/GreenWaves-Technologies/gap_sdk/blob/3.3.3/examples/native/mbed/test_driver_c%2B%2B/test_Serial_Asynch/test_Serial_Asynch.cpp

Gualor, 2020-09-22:
Yes, you're right, forgot to mention it. The read/write methods works asynchronously as well when passing the callbacks arguments. I think the issue may resides only in the "attach" method, which, unfortunately, is the one I need for using the rosserial_mbed library.

Yaooooo, 2020-09-22:
Ah, understand your issue now.
Since we have this uDMA in GAP8, all the IRQ is generated by this block, which is ralated to each uDMA transfert.
Therefore, we don't have really a HW IRQ related to uart, which means we cannot attach such event to a uart port. 
We can do as the example shows:

- attach the event to each data transfert.

One idea "may" works for you, is to capsule this to a higher level, which include:

- attach the callback to the uart object, like this pc
- bring the pc.callback into each uart transaction. 
- If you really need this irq on each "byte" read/write, then need to split the transaction of uart to byte by byte.

Gualor, 2020-09-22:
Thank you for the suggestion, but I don't think I fully understand where the fix should be implemented in the code. 
At the moment I'm running a separate thread with the [serial read](https://github.com/GreenWaves-Technologies/gap_sdk/blob/3.3.3/examples/native/mbed/test_driver_c%2B%2B/test_Serial_Asynch/test_Serial_Asynch.cpp#L56) function that triggers the callback when the software buffer overflows and the data is copied to the intended destination address.

```C++
#include "mbed.h"

RawSerial pc(UBSTX, USBRX, 9600);
Thread thread;
char buffer_rx[1];

static void ReadBufferCb(int events)
{
    // Get data and copy it in the intended location (one char at a time)
    intended_buffer.put(buffer_rx[0]); 
}

void ReadBufferThread()
{   
    event_callback_t event;
    event = ReadBufferCb;

    while(1)
    {
        pc.read((uint8_t *)buffer_rx, sizeof(buffer_rx), event, SERIAL_EVENT_RX_ALL, 0);
    }
}

void main()
{
    thread.start(&ReadBufferThread);
    while(1)
    {
        // Do other stuff
    }  
}
```

Would it be possible to avoid continuously running a parallel thread and instead have a similar behavior to the "attach" method? For now my fix is working okay, but I fear performance issues when the full application will be implemented.

Gualor, 2020-10-25:
Hi again, just wanted to give an update on my latest fix for this issue in case someone finds himself in this situation.
In the end I managed to avoid continuously running a thread to attach the callback to the read function, instead I have a function callback attached in the constructor of the "BufferedSerial" class, and the callback function when called can attach itself to the next event. I'll provide an example:

1) Define a receiver buffer inside BufferedSerial.h to be size 1 byte (every incoming byte from UART will trigger an interrupt):
```c++
uint8_t _tmp_buf[1];
```

2) Call BufferedSerial::Read inside the the BufferedSerial class constructor and attach a callback function to it:
```c++
BufferedSerial::BufferedSerial(PinName tx, PinName rx, uint32_t buf_size, uint32_t tx_multiple, const char* name)
    : RawSerial(tx, rx) , _rxbuf(buf_size), _txbuf((uint32_t)(tx_multiple*buf_size))
{
    read(_tmp_buf, 1, callback(this, &BufferedSerial::uartIrq), SERIAL_EVENT_RX_ALL, 0);  // <- Change line here
    this->_buf_size = buf_size;
    this->_tx_multiple = tx_multiple;   
    return;
}
```

3) Define a callback function inside the class (in order to be completely hidden from main) that attaches itself to the next event:
```c++
void BufferedSerial::uartIrq(int events)
{
    _rxbuf.put(_tmp_buf[0]);
    read(_tmp_buf, 1, callback(this, &BufferedSerial::uartIrq), SERIAL_EVENT_RX_ALL, 0);  
    
    return;
}
```

This is the cleanest fix I could come up with that substitute a hardware interrupt with a software one. Didn't have any problem with this ever since, hope this help.

Closing the issue, thanks @Yaooooo again for the support provided.

closed 2020-10-25 by Gualor