GreenWaves-Technologies/gap_sdk #167: PulpOS threads_events hangs

dvcirilo, 2020-10-09:
Hello,

I've been trying to run the threads_events example on v3.1 and now on v3.7.1 and none of them works, not even on gvsoc.

Other examples are working.

Am I missing something?

Yaooooo, 2020-10-12:
Hi, 

Could you please tell me what you are looking for?
This is an old example in pure pulp-os, to use the scheduler and event (but not thread, since the pulp-os is an event driven system, it doesn't support multi-thread).
Today, we don't have pure scheduler example in the sdk, however, you can take a look into all the example pmsis which is using "pi_yield()", for example:
example/pmsis/bsp/flash/hyper_flash/test_hyper_flash.c

If this doesn't meet your need, please let me know.

dvcirilo, 2020-10-14:
I am looking for exactly that, using time-sharing threads. I was expecting that the included examples would work.

I'll take a look at the PMSIS examples. Would you recommend it for new projects over pulp-os?

Thank you!

Yaooooo, 2020-10-25:
Sorry for late reply. In fact PMSIS is not at all conflict with pulp-os.
PMSIS is an api level which can be easily plugged into any RTOS like this:
[image]

In this case, we can benefit in using the same API with different RTOS. 
Sorry for our poor documentation, we are trying to improve that.

closed 2020-10-25 by Yaooooo