GreenWaves-Technologies/gap_sdk #168: Burning eFuses to start from HyperFlash

plat-zhaw, 2020-10-12:
Hi,

I am trying to boot the GAP8 on a GAPuino V2 from the HyperFlash. Do do that I need to burn the rights eFuses. According to the documentation of the gap_sdk release 3.7.1 i tried to do that. But i will not work.

Here is the terminal output:

(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1$ openocd-fuser-hyperflash
BE CAREFUL: This will permanently configure the board to boot from hyperflash and cannot be undo. Do you confirm (y/N) ?
y
Configuring board for hyperflash, please stop it when "fuse_once is done" is displayed.

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1500 kHz
Warn : Interface already configured, ignoring
trst_and_srst separate srst_nogate trst_push_pull srst_open_drain connect_deassert_srst
embedded:startup.tcl:21: Error: Unknown target type gap8, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx,  or quark_d20xx
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "/home/immanuel/gap8v1_sdk/install/workstation/share/openocd/scripts/target/gap8.tcl", line 23
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21

Thanks for all the help.

Yaooooo, 2020-10-12:
Hi,

Due to the last release, I think there are 2 parameters are necessary for the fuser haven't been configured by default. 
Could you please do:
export OPENOCD_CABLE=interface/ftdi/gapuino_ftdi.cfg
export OPENOCD_CHIP_TARGET=target/gap8revb.tcl

Then retry the script?

Best

plat-zhaw, 2020-10-12:
Hi,

I very much appreciate the fast answer.

I did as you suggested but unfortunately it did not work.

Here the terminal session:

(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ export OPENOCD_CABLE=interface/ftdi/gapuino_ftdi.cfg
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ export OPENOCD_CHIP_TARGET=target/gap8revb.tcl
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ echo $OPENOCD_CABLE 
interface/ftdi/gapuino_ftdi.cfg
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ echo $OPENOCD_CHIP_TARGET 
target/gap8revb.tcl
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ openocd-fuser-hyperflash 
BE CAREFUL: This will permanently configure the board to boot from hyperflash and cannot be undo. Do you confirm (y/N) ?
y
Configuring board for hyperflash, please stop it when "fuse_once is done" is displayed.

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1500 kHz
Warn : Interface already configured, ignoring
trst_and_srst separate srst_nogate trst_push_pull srst_open_drain connect_deassert_srst
embedded:startup.tcl:21: Error: Unknown target type gap8, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx,  or quark_d20xx
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "/home/immanuel/gap8v2_sdk/install/workstation/share/openocd/scripts/target/gap8revb.tcl", line 25
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21

Thanks for your help.

Best regards
Immanuel

Yaooooo, 2020-10-12:
Em...It seems you (the script) are not using the right openocd.
Could you please try:
which openocd
and
which gap8-openocd

If the one, gap8-openocd is point to: gap_sdk/install/workstation/bin/gap8-openocd
Then please change it in the script, line 9, the openocd -> gap8-openocd

Best,
Yao

plat-zhaw, 2020-10-12:
I did as you suggested:

(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ which openocd
/usr/bin/openocd
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ which gap8-openocd 
/home/immanuel/gap8v2_sdk/install/workstation/bin/gap8-openocd
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ which openocd-fuser-hyperflash 
/home/immanuel/gap8v2_sdk/install/workstation/bin/openocd-fuser-hyperflash

Now I do not know, if that is correct or not. And to which script are you refering to? To the this one /home/immanuel/gap8v2_sdk/install/workstation/share/openocd/scripts/target/gap8revb.tcl?

Yaooooo, 2020-10-12:
Sorry for my un-clear explanation. 
It's the /home/immanuel/gap8v2_sdk/install/workstation/bin/openocd-fuser-hyperflash

But just fyi, this path is the script where has been "installed", if you want to change from the source, you can find it here:
gap_sdk/tools/bin

If it works, please let me know, I will tag this as "bug" and push a fix in the next release.

plat-zhaw, 2020-10-12:
Now I understand it. So I changed line 9 in the script openocd-fuser-hyperflash from
`openocd -f $OPENOCD_CABLE -f $OPENOCD_CHIP_TARGET -f tcl/jtag_boot.tcl -f tcl/fuser.tcl -c 'fuse_hyperflash_boot '${GAP_OPENOCD_TOOLS}''`
to
`gap8-openocd -f $OPENOCD_CABLE -f $OPENOCD_CHIP_TARGET -f tcl/jtag_boot.tcl -f tcl/fuser.tcl -c 'fuse_hyperflash_boot '${GAP_OPENOCD_TOOLS}''`

It did seem to help some. Here is what I get now if I execute it again:
```
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2$ openocd-fuser-hyperflash 
BE CAREFUL: This will permanently configure the board to boot from hyperflash and cannot be undo. Do you confirm (y/N) ?
y
Configuring board for hyperflash, please stop it when "fuse_once is done" is displayed.

Open On-Chip Debugger 0.10.0+dev-00031-g6ec2d931 (2020-10-02-13:21)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
```
So now it finds the chip and tries to do the burning. But I do not get the "fuce_once is done" message. Instead it just repeats the message "Warn : Burst read timed out" over and over. So I assume that the fuse buring did not work. But I will try to load a simple LED toggling into the hyperflash and see if the GAP boots from the hyperflash.

plat-zhaw, 2020-10-12:
So I built the exampled 'examples/pmsis/bsp/blink_led':
```
(base) immanuel@ubuntu1-VirtualBox:~/gap8v2_sdk/examples/pmsis/bsp/blink_led$ make PMSIS_OS=freertos platform=board io=uart all -j
    CC blink_led.c
    ASM  gap8_iet.S
    ASM  startup_gap8.S
    ASM  cluster_core.S
    ASM  port_asm.S
    CC  system_gap8.c
    CC  gap8_it.c
    CC  gap_io.c
    CC  string.c
    CC  stdlib.c
    CC  errno.c
    CC  cl_malloc.c
    CC  cl_to_fc_delegate.c
    CC  fc_to_cl_delegate.c
    CC  cl_team.c
    CC  hyperbus_cl_internal.c
    CC  uart_cl_internal.c
    CC  cl_dma_irq.c
    CC  fc_event.c
    CC  fll.c
    CC  gpio.c
    CC  pad.c
    CC  pmu.c
    CC  pmu_internal.c
    CC  pwm.c
    CC  pwm_internal.c
    CC  rtc.c
    CC  rtc_internal.c
    CC  timer.c
    CC  hyperbus.c
    CC  hyperbus_internal.c
    CC  cpi.c
    CC  cpi_internal.c
    CC  dmacpy.c
    CC  dmacpy_internal.c
    CC  i2c.c
    CC  i2c_internal.c
    CC  i2s.c
    CC  i2s_internal.c
    CC  spi.c
    CC  spi_internal.c
    CC  uart.c
    CC  uart_internal.c
    CC  perf_internal.c
    CC  flash.c
    CC  partition.c
    CC  flash_partition.c
    CC  md5.c
    CC  fs.c
    CC  lfs.c
    CC  lfs_util.c
    CC  pi_lfs.c
    CC  read_fs.c
    CC  semihost.c
    CC  host_fs.c
    CC  ota.c
    CC  ota_utility.c
    CC  updater.c
    CC  bootloader_utility.c
    CC  gapuino.c
    CC  camera.c
    CC  himax.c
    CC  ov7670.c
    CC  gc0308.c
    CC  ov5640.c
    CC  pixart.c
    CC  display.c
[... 204 lines cut ...]
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: gap8.cpu: IR capture error; saw 0x0f not 0x01
Warn : Bypassing JTAG setup events due to errors
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
Warn : CRC ERROR! Computed 0x00000000, read CRC 0xffffffff
```
As you can see I get a warning that the CRC is wrong. And the LED is not blinking so it is not working. I have no idea what is going on. I just hope I did not destroy my GAPuino board.

plat-zhaw, 2020-10-12:
Since I have two GAPuino V2 boards I loaded the blink_led example onto the second board on which I did not try to burn the fuses to boot from the HyperFlash. On that board it is working just fine:
```
(base) immanuel@ubuntu1-VirtualBox:~/gap8v2_sdk/examples/pmsis/bsp/blink_led$ make platform=board io=host run
gapy --target=gapuino_v2 --platform=board --work-dir=/home/immanuel/gap8v2_sdk/examples/pmsis/bsp/blink_led/BUILD/GAP8_V2/GCC_RISCV    run --exec-prepare --exec --binary=/home/immanuel/gap8v2_sdk/examples/pmsis/bsp/blink_led/BUILD/GAP8_V2/GCC_RISCV/test 
Launching execution with command:
gap8-openocd -c 'gdb_port disabled; telnet_port disabled; tcl_port disabled' -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8revb.tcl -f tcl/jtag_boot.tcl -c 'gap8_jtag_load_binary_and_start "/home/immanuel/gap8v2_sdk/examples/pmsis/bsp/blink_led/BUILD/GAP8_V2/GCC_RISCV/test" elf'
Open On-Chip Debugger 0.10.0+dev-00031-g6ec2d931 (2020-10-02-13:21)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled

	 *** PMSIS Blink LED ***

Entering main controller
Blinking USER LED
```
The user LED is blinking as it should.

It is my understanding, that after burning the fuses to boot from the HyperFlash it should still be possible to boot from the JTAG interface and load a program from there. Am I wrong? Did I destroy my other GAPuino V2 board because I tried to burn the fuses to boot from the HyperFlash?

plat-zhaw, 2020-10-12:
Soooooo, after some trial and error I got it all working.

1. If you terminate a process in the bash using Ctrl+Z, because it does not react to Ctr+C, it does not kill the currently in the bash terminal running foreground process. All Ctr+Z does is giving control back to the terminal. So after you did that you need to kill the still running process(es) using kill -9 <pid>. I know this does out me as a Linux noob ;-) But there might be other who do not know that. So I thought it worth mentioning.

2. When I tried to burn the eFuses to boot from the HyperFlash I had the Himax camera connected to the GAPuino V2 board. Apparently that is not a good idea. I disconnected the camera and I could load the blink_led over JTAG again. Furthermore, I get a lot less warnings "Warn : Burst read timed out". See my console output for loading blink_led over JTAG:
```
base) immanuel@ubuntu1-VirtualBox:~/gap8v2_sdk/examples/pmsis/bsp/blink_led$ make platform=board io=host run
gapy --target=gapuino_v2 --platform=board --work-dir=/home/immanuel/gap8v2_sdk/examples/pmsis/bsp/blink_led/BUILD/GAP8_V2/GCC_RISCV    run --exec-prepare --exec --binary=/home/immanuel/gap8v2_sdk/examples/pmsis/bsp/blink_led/BUILD/GAP8_V2/GCC_RISCV/test 
Launching execution with command:
gap8-openocd -c 'gdb_port disabled; telnet_port disabled; tcl_port disabled' -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8revb.tcl -f tcl/jtag_boot.tcl -c 'gap8_jtag_load_binary_and_start "/home/immanuel/gap8v2_sdk/examples/pmsis/bsp/blink_led/BUILD/GAP8_V2/GCC_RISCV/test" elf'
Open On-Chip Debugger 0.10.0+dev-00031-g6ec2d931 (2020-10-02-13:21)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled

	 *** PMSIS Blink LED ***

Entering main controller
Blinking USER LED
^Z
[3]+  Stopped                 make platform=board io=host run
```

3. After that I tried to burn the eFuses again. This time I left the camera disconnected and low an behold the burning worked:
```
(base) immanuel@ubuntu1-VirtualBox:~/gap8v2_sdk/examples/pmsis/bsp/blink_led$ openocd-fuser-hyperflash 
BE CAREFUL: This will permanently configure the board to boot from hyperflash and cannot be undo. Do you confirm (y/N) ?
y
Configuring board for hyperflash, please stop it when "fuse_once is done" is displayed.

Open On-Chip Debugger 0.10.0+dev-00031-g6ec2d931 (2020-10-02-13:21)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
[... 172 lines cut ...]
Instruct flasher to begin flash per se
device struct address is 470093656
going to wait on addr GAP_RDY
wait on gap_rdy done witg buff ptr 0 469828864
loading image with addr 469828864 addr_min 469828864 and size 31776
load image done
flasher is done, exiting
--------------------------
flasher is done!
--------------------------
```
It worked! After the flashing of the HyperFlash the user LED started blinking. The I pressed the reset button on the board and the LED started blinking again, without having to load the program over JTAG again. After that I disconnected the GAPuino V2 board from the USB (and therefore removed the power supply). After plugging it in again the GAP8 booted nicely from the HyperFlash and the user LED began to blink.

**Conclusion:**

There is a bug in the openocd-fuser-hyperflash script on line 9. There, openocd has to be changed to gap8-openocd. The same bug is in the scripts openocd-fuser-fll-revb (same line) and openocd-flasher (line 10).

To burn eFuses you should not connect anything to the GAPuino board and it should work.

There still does appear to be a problem which prompts the display of the warning "Warn : Burst read timed out". The people from GreenWaves might want to look into that.

Thanks a lot to Yaoooo for the help.

Regards,
Immanuel

Yaooooo, 2020-10-12:
Glad you find out the problem!
I reopen it for us to track there is a bug fix need to be done.

For what you mentioned disconnect the camera, I found it's very strange. I don't think this will cause the fuse failed. Will double check it. Thanks

plat-zhaw, 2020-10-13:
Of course I am OK with you reopening the issue. I just thought I was supposed to close it, since I was able to get it running.

About the camera: I am not sure the Himax camera was really the problem while burning the eFuses. What I can tell you is that after I disconnected the camera it worked. But that might also be, because that time around the GAPuino was disconnected from the power supply longer than before and that might have given all the capacitors more time to discharge and therefore it was a "more complete" reset of the whole board.
What I can tell you with certainty is, that if the camera is connected, I get a lot more warnings "Warn : Burst read timed out" than when the camera is not connected. That is for just loading some program over JTAG unto the GAP8. I think that is also very odd and I have no idea, why that is the case. As said in my (not anymore) closing post you (or some other engineer from GreenWaves) might want to investigate this warning message. For now they just might confuse people, which are new to the GAP8 universe and do not know that this is kind of normal.

Please do not hesitate to contact me, if you have further questions or you need me to try something out for you. I am happy to help in any way possible (as far as it is in my not so considerable power).

Manucar, 2021-04-27:
Hi @Yaooooo, in the current SDK version this bug is still present, are you planning to include a fix in the next release?

Best,

Manuel

Yaooooo, 2021-07-02:
Efuse tool has been updated. 
In the latest release, you can find this tool: https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/gap_fuser
It should be much faster and easy to be used.

closed 2021-07-02 by Yaooooo