GreenWaves-Technologies/gap_sdk #169: Autotiler

gemenerik, 2020-10-14:
Running the latest gap-

sousoux, 2020-10-14:
Hi. Sorry for the problem. The default nntool template tries to load the RNN kernels as well as the CNN ones. It looks like you are missing a source file in your compile line ${TILER_CNN_GENERATOR_PATH_SQ8}/RNN_Generators_SQ8.c. Actually you could use a custom nntool model template and remove the call to Load_RNN_SQ8_Library() from it since the example you are trying is CNN based but adding the file above should solve the issue for you.

gemenerik, 2020-10-14:
Thanks for the quick reply. That actually makes a lot of sense. I removed the call to load it from the AutoTiler model, and the one-liner ran without error.

closed 2020-10-14 by gemenerik